betaflight/betaflight #8283: RunCam Split camera control doesn't work

TheRealOrange, 2019-05-16:
This issue was reported some time ago for firmware version 3.5.0, reportedly fixed in 3.5.2, but I cannot get the RunCam Split 2 with 3.1.0 firmware to work on Betaflight 4.0. I have confirmed that by removing initialisation logic the camera control works.

McGiverGim, 2019-05-16:
@azolyoung can you look at this?

mmabln, 2019-09-08:
Have the same issue on MatekSys F405-CTR, RunCam Split 2 Firmware 3.1.0, connected to TX3/RX3

I checked the serial TX/RX with oscilloscope, and there is no data transfer taking place when I am toggling the camera control modes 1-3 (although I see modes becoming active in configurator at the same time).

mikeller, 2019-10-01:
@azolyoung: Can you help here please?

qratz, 2019-10-01:
I am running 4.1RC5 on a CLRACINGF7 target (Heli-Nation Talon F7). I have configured my Taranis to start recording on the Split mini 2 (latest FW) prior to arming and stop recording after disarming. For this i use Runcam Device commands (Camera Power Button) and it works reliably. Here's the relevant part of my config:

```
rcdevice_init_dev_attempts = 8
rcdevice_init_dev_attempt_interval = 1000
rcdevice_protocol_version = 0
rcdevice_feature = 0
```
Split Mini 2 is connected to UART6. So to the extent of using the Camera Power Button mode i can say that it works in 4.1RC5.

bri3d, 2019-12-02:
Using the settings from `qratz` (especially the init retry), my Power Button, Mode, and WiFi Switch interface works, but as soon as I attempt to enter the OSD (by pushing yaw-right at mid-throttle), the entire RunCam interface quits working (i.e. once I have tried to enter the OSD, none of the buttons work any longer).

I do not have my logic analyzer handy but can look tonight to see what is being sent.

rolial, 2020-03-28:
Have the same issue on Holybro Kakute F7, RunCam Split 2s Firmware 4.1.5, connected to TX2/RX2

Do we have any palliative solution?

rolial, 2020-03-30:
@mikeller  @azolyoung 
I just used these commands below that were passed by RunCam support in the past and worked with version 4.1.5.
Do you see any problems using these commands until an official fix is released?

After flash, please type the following cli commands, and try again.
> set rcdevice_protocol_version = 1
> set rcdevice_feature = 7
> save

TheRealOrange, 2020-04-01:
that didnt work for me, I just edited the firmware and compiled a version that didn't check camera connection and forced it to use the Runcam Split camera protocol

Telekatz, 2020-06-11:
> 
> 
> Using the settings from `qratz` (especially the init retry), my Power Button, Mode, and WiFi Switch interface works, but as soon as I attempt to enter the OSD (by pushing yaw-right at mid-throttle), the entire RunCam interface quits working (i.e. once I have tried to enter the OSD, none of the buttons work any longer). I am on CLRacing_F4S, BF 4.1.1 release (using board config, not legacy), with a RunCam Split 2 wired into UART6.
> 
> I do not have my logic analyzer handy but can look tonight to see what is being sent.

I have the same behavior with my RunCam Split 3 Nano. I checked the RunCam's response to the GetDeviceInfo command. The result was that the RunCam Split does not support the 5-key OSD remote control used by the stick commands. If these commands are used anyway, the RunCam interface will fail.
Access to the OSD only works with the CAMERA WI-FI, CAMERA POWER and CAMERA CHANGE modes as described here: https://goo.gl/tm8CPS

@azolyoung
I checked previous versions of src/main/io/rcdevice_cam.c. There was a check whether RCDEVICE_PROTOCOL_FEATURE_SIMULATE_5_KEY_OSD_CABLE is supported by the rcdevice. This check was removed with #7497. Was that intentional or a mistake?

tavdog, 2023-03-24:
Time to close ?

closed 2023-03-24 by haslinghuis