betaflight/betaflight #8285: Pilot Logo in OSD

skyfpv, 2019-05-16:
**Feature Request**
I would like a new pilot_logo OSD element to be added. It would allow users to define a pilot logo that would be visible during flight. Current methods for doing this are hacky from a user perspective like creating a custom font and editing editing the craft_name. This is especially true for 2 or 3 row logos.

**Possible Solution**
One way we could remedy the problem is by giving users the ability to upload their own image similar to how the Betaflight logo is uploaded. Since we are low on character space, we could allow the user to utilize a portion of the Betaflight logo image as their pilot logo (for example replacing the lovely bug portion with a prop or something).

**Possible Alternative**
Alternatively, we could downsize the Betaflight logo space to make room for a dedicated pilot logo image. 

**Side Note**
I will likely be using this pilot logo feature for automated craft identification. With that in mind, it would be great if we could allocate somewhere in the region of 12 character spaces. If not, I think this feature will still make a lot of BF users happy!

Docteh, 2019-05-16:
automated craft identification? whats that look like?

Take a look at #3461 they've discussed shuffling the font around for more symbols

skyfpv, 2019-05-16:
@Docteh Craft Identification looks like [this](https://youtu.be/SEc01tGImvo) for example. 

I think even if we only used 1/4th of the characters that #3461 would free up, we'd have plenty of space for the purposes of pilot logos

Docteh, 2019-05-17:
Huh. How complicated is the workflow getting a qr code into a font? Is the logo supposed to be transmitted the whole flight? I'm Trying to think of how to compare qr code with ocr of a code.

skyfpv, 2019-05-17:
The QR code doesn't really have much to do with the feature request. 
I just think it would be nice for pilots to be able to easily place a logo on their OSD. If done via custom fonts, the user still has to find some hack for displaying the characters if he wants a logo taller than 18px. Ideally this would work identically to the user logo seen in the BrainFPV graphical OSD, but with a more simple, character based implementation

jflyper, 2019-05-17:
If a pilot logo is expressed using a custom font glyph, then it's not a character based implementation at all.

skyfpv, 2019-05-17:
@jflyper maybe that's the wrong term. Hope you understood my meaning anyway

pkruger, 2019-09-09:
Is this still something that people want?  I have a version working by replacing character 0xA3 to 0xB7 (spaces in the Betaflight Logo) with the contents of a 48x72 pixel bitmap (via Python script).  A small change in the Betaflight firmware replaces these characters with spaces when the logo is displayed.
Some questions:
1. Is 16x16 characters (48x72 pixels) enough?
2. Would it be better to have 3x2 character (36x36 pixels)?  (i.e. smaller logos, but 2 or 3 of them)
3. Would people want to fill up OSD space with a pilot logo anyway?

Docteh, 2019-09-09:
pilot logo might be nice for DVR footage.

skyfpv, 2019-09-10:
I think a lot of pilots would enjoy putting their logo on the screen since many put their names there anyway.

mikeller, 2019-09-10:
@pkruger: I think there is still interest in getting more out of the OSD. Some more effort has been made to make the extended character set available on most MAX7456 clones available (#8310), and a pilot logo would be good use for this.

pkruger, 2019-09-10:
@mikeller Interesting, thanks for mentioning.  #8310 looks a little more comprehensive than the hacky replacing character method I used, good job @Docteh.  We may be able to use my (to be improved) Python script as a starting point for the work in the Configurator to add a Bitmap into the font file (extended characters maybe), i.e. strip the bitmap and generate the font characters for it.  (works similar to Betaflight logo replacers.)  We just need to decide on what the size restrictions will be.  Otherwise we'll use whatever the Configurator adds, no worries.

closed 2026-09-26 by blckmn