betaflight/betaflight #8286: Race Timer - feature request

BludzUK, 2019-05-16:
I was hoping it might be possible to do some of the following ideas. This is intended for racers running timed heats and wanting a visual display of how long is left, either in actual races or just practice without a lap timer.

Could we have a countdown timer that starts counting down after you take off? Maybe linked with launch mode throttle percentage.

I see a request to add miliseconds to the timer. I think this would be great. Just 0.1 increments would be nice for people who time laps while reviewing dvr. 1 second is quite long on a short track.

One step further to launch mode could be a 3 second countdown before starting the timer and telling you to start flying.

Keep up the great work guys.

JyeSmith, 2019-05-16:
> One step further to launch mode could be a 3 second countdown before starting the timer and telling you to start flying.

If you are running opentx try this http://rcdiy.ca/quad-race-start-sequence-gorace/  (my slightly modified version because it didn't work out of the box for me https://gist.github.com/JyeSmith/c502cdf4caf1e1a3cd59c42041c50c00)

I have it on my launch control switch and it announces arming and then has a random delay to a start tone.

etracer65, 2019-05-16:
Having milliseconds on the timer display would be useless. The OSD elements only update 12 times a second, so minimum resolution is 83ms.

BludzUK, 2019-05-17:
Good suggestion @JyeSmith However,  I run a Nirvana TX. Also the idea of having the timer and/or start sequence on screen is so I can put headphones on while racing or at least practice.

@etracer65 Sorry to completely disagree with you but we only need it to update 10 times a second to add tenths of a second to the timer. Totally agree that anything else is useless due to frame rate limitations.

etracer65, 2019-05-17:
> I see a request to add miliseconds to the timer. I think this would be great.

> Sorry to completely disagree with you but we only need it to update 10 times a second to add tenths of a second to the timer.

What? The elements already update at 12hz like I said. The timer elements already have the ability to display in hundredths of a second so I'm  not sure what you're saying. Even updating in hundredths is pretty useless actually as it'll update in 0.08 steps (or even more depending on interactions with the 25/30hz video frame rate in your DVR). But it's already there so not sure what the problem is. I was responding to your desire to have milliseconds which as I said is useless.

BludzUK, 2019-05-17:
@etracer65 Sorry for not knowing it already does hundreths. I couldn't see the option to change that before requesting it. No point doing tenths, if it already does hundreths other than a more consistant time being displayed as the refresh rate can handle it better and there being 1 less digit to display making it look cleaner. So I'm not that bothered about this issue now, I will find the setting, thanks for pointing it out.

So just a countdown timer that starts on take off instead of arming is what I would like to see added :)

etracer65, 2019-05-17:
The definition of "take off" is a bit difficult. When Launch Control is used it's easy. But otherwise it's a bit difficult to define the "take off" point. We'd probably need to have (yet another) throttle based threshold that the user would need to configure for the non-Launch Control case.

BludzUK, 2019-05-17:
I was worried that might be the case. Perhaps this could just be an additional feature/option of launch control. A timer that is just linked to launch control. Does narrow the use of it slightly as not everyone uses/likes launch control. If it was just for me I would say it's perfect as I will be using launch control when racing with the countdown timer. Would also want the timer to be adjustable through osd not just configurator.

Thanks

DooDoo512, 2019-05-24:
Hi @BludzUK ,I have worked out this issue with an extra plug equiment .I named it OSD laptimer. It's tiny and very useful for solo race training . I will sell it next month.
My email:chenjing512@139.com ,contact me ,I'll send you some pictures or DVR

closed 2026-09-26 by blckmn