betaflight/betaflight #8288: betaflight 4.0.X doesnt work in omnibus F4-F4sd(wizzard x220s)

ElUtku, 2019-05-17:
I´ve just install the last version of betaflight(4.0.X) and i´m reporting that it doesn´t work. 
I can install and configure the firmware, but when i arm the dron and i move the accelator, the drne start to jump and crush with the floor, it can´t land at all.
I think that the problem is caused by a fake read of the sensors, beacuse on betaflight 3.X the drone works perfectly.

Asizon, 2019-05-17:
Running 4.0.2 OMNIBUSF4SD in my OmnibusF4 V5 without problems,working fine. Please, for configuration issues you can use Slack platform.

watageek, 2019-05-17:
Have you provide a diff.

Have you read other issue with serial RX issue on uart.

You certainly have a clone FC. 

I bet the code is superb ( as usual) and you have not check if you have signal in the RX tab in the configurator.

Hence the " not arming" condition...

😐

aowi7280, 2019-05-17:
I had the same issue. You need to set the defaults after flashing because the accelerometer is rotated 270 degrees. When I pasted in my diff, it over-wrote the default.

ElUtku, 2019-05-18:
I forgot to said that i have a omnibus F4 V3.
I read lot of reporte and i din't find anything.

closed 2019-05-18 by mikeller