betaflight/betaflight #8291: HD CAMERA FRAME for OSD

cpnmercury, 2019-05-17:
If possiable would you add lite 4 moveable corners to the OSD so you could frame what the gopro is seeing? BrainFPV has it on their OSD, wouldn´t be hard to impliment.

Thx for a great job!

etracer65, 2019-05-17:
BrainFPV boards have a graphics based OSD. The OSD on normal flight controllers is character based and limited resolution (30x13 for NTSC and 30x16 for PAL). So the only way we could do it and allow you to customize would be to have 4 separate elements that each draw a single "+" (or some other character). Would be a bit of overkill for the value gained.

cpnmercury, 2019-05-17:
yeah, that´s what I had in mind 4 separate elements, line a L for each corner. I think the value is there, now that mini kwads is getting noticed for the commercial use. 
Just a suggestion, think people would like it.

mikeller, 2019-05-18:
Part of the problem that I think @etracer65 wanted to highlight is that with a resolution of 30x13 or 30x16, the precision for adjusting this 'viewport frame' would be relatively poor. So not sure if there's much value in doing this.

watageek, 2019-05-18:
And since most of the fpv camera have tilt angle adjust on the go and de-adjusted every little time it bump, crash, wear out, it will really be inconsistent...

cpnmercury, 2019-05-18:
Yeah, had the ichabod jr in mind, but your right most frames has the fpv camera that will change after a crash. But i dont crash when i do filming jobs. Seems like the brainFPV is the chocie. Or i can just place like timers etc in the corners instead. I might be early with this request, but i think people would like this option in the future

mikeller, 2019-05-19:
@cpnmercury:

> Or i can just place like timers etc in the corners instead. 

Good idea. Try it out, and see if the position accuracy you can get from that is good enough for what you want to do, and report back. If it works we can always add it, as you say it won't be too hard.

ghostface, 2020-06-18:
FYI: This was implemented via https://github.com/betaflight/betaflight/pull/9261 and can be closed imho

closed 2020-06-18 by etracer65