betaflight/betaflight #8292: Using MSP_SET_RAW_RC together with another serial RX

ricardodeazambuja, 2019-05-17:
First, I'm would like to apologise if this is not the right place to ask this question (just close/delete/kill it and I will understand). I've searched through the old issues, but I couldn't find anything like what I'm looking for.

Currently, I'm able to send commands using MSP (I have an OMNIBUS F4 clone using the latest betaflight firmware), but I really would like to be able to make it work as it worked in the [past](http://www.multiwii.com/wiki/index.php?title=Multiwii_Serial_Protocol) where the values sent through MSP_SET_RAW_RC would override the current RX. Right now, I can only enable RX_MSP or RX_SERIAL, not both. I'm aiming at having both working together because it would be safer to have a transmitter that has higher priority. I know there's the possibility of a trainer cable, but I want to have a single board computer running on the drone (and I like Betaflight :1st_place_medal:, so I don't want to use other firmware).

I'm planning to modify the firmware. Any suggestions which firmware version had that 'feature' I've mentioned? Additionally, I've read through the docs, I'm able to compile my own firmware, but I can't find anything explaining better how the code is structured :confused:.

Thanks in advance for any suggestion!

mikeller, 2019-05-18:
What you describe was not a feature, it was a bug. :-) `RX_MSP` is designed only for testing, and not for flying, as it is lacking any of the safeguards that are built into the other RC protocol drivers, like failsafe detection.

ricardodeazambuja, 2019-05-18:
Thanks for your ultra-fast reply, @mikeller! What do you mean by not having failsafe detection? When I tried it this week, it would enter failsafe if I was not sending MSP_SET_RAW_RC messages fast enough. In fact, it would only accept MSP_SET_RAW_RC to arm the drone if I reached the minimum frequency before I sent the arm command. I haven't tested if the minimum throttle values would trigger failsafe, though.

Anyway, MSP is not a big deal because I can use IBUS (or other protocol) instead, still I'm struggling to enable two RXs. Ideally, I would like to have something like the trainer situation, but using two receivers on the drone - UART connected - giving higher priority to one of the receivers. This could open some cool possibilities for manual flight and reliability in general where you could bind two receivers to the same transmitter ;)

Would you have any suggestions about where should I start looking at the firmware or where I could find more details about the firmware implementation (the source code doesn't have a lot of comments)?

Thanks again!

etracer65, 2019-05-18:
Trying to add support for dual receivers would be a really big task and require completely refactoring the entire receiver stack. I certainly wouldn't want to even consider taking this on. Way too much effort for limited returns. With Frsky equipment (like the R-XSR for example) you can already use two receivers so hardware solutions already exist. The trainer use-case isn't really a good solution as how would you trigger the "takeover". Both receivers will always be outputting signal. Also the possibility of having the master in the trainer have a failsafe and the slave not is a problem. Trainer should be done locally in the TX with a physical cable.

mikeller, 2019-05-18:
@ricardodeazambuja: Looks like refactoring of the RX code got `RX_MSP` covered by failsafe then as well - but this is more coincidental than planned (and tested).

And yes, what @etracer65 says is definitely true. We looked into dual RX support at some point in the past, but we never managed to get it to work in a way that made it usable - there just is not enough information on the 'remote' end of the connection to be able to have a master / slave control that is controllable in all scenarios if you take into account that one or both of the RX links may be failing intermittently.

ricardodeazambuja, 2019-05-20:
@etracer65 and @mikeller, thanks for your comments! You made me think about my plans, but I still would like to get into shape in relation to the firmware.
Would you mind pointing me the directions for a better documented firmware source code? The code itself is very hard to read / follow (at least at a first glance) I may have missed a doc somewhere and I hope the core developers have some kind of diagram, etc. that they wouldn't mind sharing :)

Thanks again for your patience!

mikeller, 2019-05-20:
@ricardodeazambuja:

> The code itself is very hard to read / follow (at least at a first glance) 

Try a bit harder then :grin: - compared to other open source project our code quality is relatively good.

Apart from that, the documents in here are worth a read:

https://github.com/betaflight/betaflight/tree/master/docs/development

ricardodeazambuja, 2019-06-04:
I'm trying to implement my idea, but I'm still far from getting it right. The way I'm trying now is by creating an extra variable for the rxRuntimeConfig. I changed the firmware to allow both features: MSP_RX and SERIAL_RX. In rxInit, I moved the MSP part to the end of the sequence (so it takes over by default) and changed the variables for the SERIAL_RX to rxRuntimeConfig_bckup. I also created backup copies for the rxConfig and the rxChannelCount I solved with an extra variable and a pointer.
So far so good, it compiles, I can flash it and only MSP commands are accepted. Now I'm trying to modify readRxChannelsApplyRanges to read the AUX1 from SERIAL_RX and switch (from MSP_RX to SERIAL_RX) when it goes to a value bigger than 1800. When that happens, I'm trying to copy the important variables only once by setting a flag. The system should be stuck into SERIAL_RX until the next boot.
Just a tiny little problem... it compiles, but it does not work the way I was expecting :D

```
static void readRxChannelsApplyRanges(void)
{

    // Verify if the SERIAL_RX has its channel AUX1 set to HIGH (>1800)
    // and if it is, takeover from MSP_RX and only returns after reboot
    #if defined(USE_SERIAL_RX) && defined(USE_RX_MSP)
        if (rcSwitch == 0) {
            const uint16_t serial_aux1 = rxRuntimeConfig_bckup.rcReadRawFn(&rxRuntimeConfig_bckup, rxConfig_bckup()->rcmap[AUX1]);

            if (serial_aux1 < MIN_VALUE_SWITCH){
                rcSwitch = 1;
            }
        }else if (rcSwitch == 1){
            rxRuntimeConfig = rxRuntimeConfig_bckup;
            rxConfig_Copy = rxConfig_bckup_Copy;
            rxConfig_System = rxConfig_bckup_System;
            // pgResetFn_rxConfig = pgResetFn_rxConfig_bckup;
            rxChannelCount_ptr = &rxChannelCount_bckup;

            rcSwitch = 2; // it will ignore MSP_RX until reboot
        }
    #endif

    for (int channel = 0; channel < *rxChannelCount_ptr; channel++) {

        const uint8_t rawChannel = channel < RX_MAPPABLE_CHANNEL_COUNT ? rxConfig()->rcmap[channel] : channel;

        // sample the channel
        uint16_t sample = rxRuntimeConfig.rcReadRawFn(&rxRuntimeConfig, rawChannel);

        // apply the rx calibration
        if (channel < NON_AUX_CHANNEL_COUNT) {
            sample = applyRxChannelRangeConfiguraton(sample, rxChannelRangeConfigs(channel));
        }

        rcRaw[channel] = sample;
    }
}

```
I have to admit, I'm confused with the some of the PG_* macros...

My logic is that I could have two controllers, arm with one and takeover with another by simply switching its arm switch. No extra channels needed.

Any suggestion will be much appreciated!

mikeller, 2019-06-05:
All of the `PG_` and `pgResetFn_` stuff is referring to parameter groups, which is Betaflight's way of dealing with persistent configuration. Apart from giving the user a way to persistently enable or disable the functionality you want to add, your proposed solution should definitely not involve any modifications at this level, or else it will have undesirable side effects.

Also, hard coding this to any protocol, AUX channel, or channel value to trigger a switch to the alternate RC data source is definitely not an acceptable solution.

ricardodeazambuja, 2019-06-05:
Thanks for the feedback @mikeller and don't worry, I will not hard code stuff. I doing that as a first proof-of-concept version since hard coding eliminates a lot of degrees of freedom.

Woga65, 2019-07-01:
Since the requested feature is already added to iNav you might want to have a look at it:
https://github.com/iNavFlight/inav/pull/4704

closed 2019-07-01 by ricardodeazambuja

birktj, 2020-06-13:
I made a fork with a hardcoded multiple receiver setup (MSP + SBUS) over at <https://github.com/AscendNTNU/betaflight/>. I am wondering if I should create a bit more refined patch and upstream it? The more refined patch would probably look a lot like the iNav implementation with a MSP override flight mode and override channels. My question is if such a patch would be accepted upstream?

We used this system for autonomous flight with an onboard computer. The reason we ended up using Betaflight instead of another flight controller software more tuned for that use case is that we are flying indoors with our own positioning system. We have found other flight controllers to be quite buggy when flying without GPS and our system with running control algorithms on a separate onboard computer and sending RC commands to Betaflight over MSP has been very stable. This is simply to show that there are people who would like and use such a feature.

Also relevant is #2545

mikeller, 2020-06-14:
@birktj: Yes, I think what you describe can be considered for upstream - while MSP in itself lacks some of the safety features that other protocols have, like failsafe detection, having this set up as an override, and relying on the safety features in the primary protocol to kick in when required seems to be a reasonable approach.

birktj, 2020-06-14:
@mikeller that sounds great! Could you reopen this issue and I will get to work.

ricardodeazambuja, 2020-06-14:
@mikeller Would you mind explaning what you meant by not having failsafe detection when using MSP? I've been using Betaflight (ok, I'm still on 4.1) with my own library [YAMSPy](https://github.com/thecognifly/YAMSPy) and when the flight controller doesn't receive MSP packets for a certain period (I think the minimal frequency by default is 10Hz) it will enter failsafe.

Thanks!

mikeller, 2020-06-14:
@ricardodeazambuja: Ok, if this is the case then this is a (positive) side effect of how RC input processing now works - MSP is not expressly designed to do this, and did not do this in earlier versions of the firmware. If it is now doing this it is probably just a matter of documenting how the failsafe behaviour of MSP is working exactly.

pieniacy, 2020-07-28:
Just want to give a +1 and say that this feature is very useful for more people. Flight dynamics of betaflight is quite cool and we like interfacing our companion computers with it. Making it more friendly on this front can secure betaflight more appearance for example in autonomous drone racing.

On the technical note: I see PR merged yesterday, do you maybe expect 4.2.2 anytime soon @mikeller? Also, I see some clever masks on channels, so channel which decides if override is active will always be taken from standard rx, right? @birktj what do you think about creating a little wiki page? I can offer a bit of my time if you would like some help.

birktj, 2020-07-28:
Yes autonomous drone racing is my angle and the reason I created the patch. Betaflight is just so much more suited to this than all of the GPS based FCs like PX4 and Ardupilot.

The way it works is by creating a new mode MSP override (it is mode 50) which can be mapped to any aux channel like other modes. When the mode is active all the channels described by the bitmask `msp_override_channels`, in most cases you probably want to set this to 15 (0b1111), this will override the roll, pitch, yaw and thrust channels. And yes you can only enable the mode on a channel that will not be overridden so in theory you should never get into a situation where you cannot disable override mode (if you lose normal RX the drone should go into failsafe mode). 

Creating a wiki page is probably a good idea, I am slightly afraid that this feature will be silently forgotten. Adding the ability to configure this from the Betaflight Configurator would also be cool, but I don't think it is very likely that I will get around to it.

@pieniacy you may also be interested in a small ros node I made to comunicate from ROS to MSP, it is available over at: https://github.com/AscendNTNU/msp_flightcontroller_interface/

haslinghuis, 2020-07-28:
@birktj not sure about the status but if you give me the raw info I can put it into the wiki. Would be nice to have some additional info or examples using ROS(2) to MSP to put into the wiki. The PR has to merged before it can be documented.

birktj, 2020-07-28:
@haslinghuis give me a few days and I can try to put together a suitable text. I will give you a ping when it is ready. Info about ROS integration would be nice and there is some stuff out there to do it, unfortunately most of it is quite hacky (like my ROS node). The PR is merged, but it probably doesn't make much sense to publish docs before a new Betaflight version is published?

haslinghuis, 2020-07-28:
Sure, merged and being released are two different things 😅 
Nothing wrong with documenting this in advance, as long as we mention the release info

ricardodeazambuja, 2020-07-28:
@birktj Would you be able to help with this question? I'm imagining the situation where the FC is currently using msp override and it loses the main receiver, but it is receiving valid msp messages from the serial port. Will it enter failsafe? 
Thanks!

birktj, 2020-07-28:
In this case it should enter failsafe.

ricardodeazambuja, 2020-07-28:
In this situation the main rx is the only one capable of triggering the failsafe mode. It would be interesting if telemetry could send some info about the msp messages to the transmitter so the pilot would know when to take over if the msp gets stuck / lost.

pieniacy, 2020-07-28:
> In this case it should enter failsafe.

@birktj which _should_ do you mean? :smile: That it is behaviour we want? Or that you think this is what it will do?

Our use case in autonomous racing would be that computer is flying over msp while human pilot (with radio in hands and with finger on the switch) is observing to take over whenever AI is doing something strange. I presume @birktj you have a similar case? So this is always line-of-sight, pilot can take over whenever the drone does something dangerous or AI operator calls for takeover.

In our case main radio signal should never be lost, so when it is lost failsafe is definitely what we want.
We don't want any rouge AIs, right? :stuck_out_tongue_winking_eye:

mikeller, 2020-07-29:
@pieniacy: #9915 will go into 4.3 - no planned release date for this yet. If you want to test it you can always do so with a development version of the firmware.

@birktj, @haslinghuis: For a feature like this (implemented in firmware, fixed functionality), an entry in `/docs` in the repository is preferrable over the wiki for documentation.

birktj, 2020-07-29:
@pieniacy by _should_ I mean that it will go into failsafe as long as there are no bugs. Yeah, we use the exact same system with a backup pilot in line-of-sight.

pieniacy, 2020-07-29:
I have tested a bit, everything seems fine. Many thanks to everyone involved!

My setup: developement build #2250, KakuteF4V2, sbus from XSR on one uart, msp from companion computer on another uart (not usb msp).

Failsafe conclusions: everything **works normally** on sbus receiver and **failsafe is unaffected by the msp** at all. Regardless of what msp was doing (not sending commands, sending commands disarmed, sending commands armed) when sbus rx failsafes the quad always enters failsafe state. Enabling / disabling mode 50 with aux channel also works as intended and does not influence failsafe.

Possible improvement is detecting when msp stops sending frames and signalling (and only signalling) it somehow to the pilot, for example with rc radio telemetry as @ricardodeazambuja suggested.

haslinghuis, 2020-08-02:
I also read a similar feature request on https://github.com/iNavFlight/inav/issues/1411 for those interested.
For the application I was thinking off (BVLOS), the companion computer should take over in autonomous mode while the RC Link is only for backup and for close proximity (LOS), So to avoid the failsafe issue we have to use Ardupilot/Pixhawk?

birktj, 2020-08-03:
BVLOS sounds a lot more complicated. Current failsafe behavior is by design. It could probably be modified to accommodate your usecase, but the design becomes a lot more complicated and in general it seems quite dangerous. There are a lot of questions to design the failsafe system to be usable for all usecases.

gorghino, 2022-10-13:
Hello @birktj and @haslinghuis, did you eventually write something about MSP_OVERRIDE (and its configure) on the Wifi? Tnks

ricardodeazambuja, 2022-10-14:
Oops, @gorghino, you are right, it's already there. The step you may be missing is to set the mask (`msp_override_channels_mask`) using the CLI to make the `MSP OVERRIDE` mode appear in the Modes tab. The mask will tell which channels will be overridden when you activate the `MSP OVERRIDE` mode.

gorghino, 2022-10-17:
That's right @ricardodeazambuja, I couldn't see the mode because `msp_override_channels_mask` was 0 by default. 
Plus, I was mislead by the missing `RX_MSP_OVERRIDE` entry in `feature` even if it was enabled. Wasn't supposed to be listed along with `OSD, TELEMETRY,` etc.?
Thank you!