betaflight/betaflight #8294: BF not responding to throttle under high voltage sag condition

Jetrell, 2019-05-18:
I'll cut to the chase with an explanation of the issue. Then explain the road I took to get to this conclusion.

_The condition occurs when a full throttle load causes the battery voltage to sag into the voltage range of the next lowest series battery count. 
i.e. 4S pack voltage drops below 3S max.(12.6v) 
When this occurs, the throttle becomes unresponsive and will hold full throttle until the 4S battery voltage rises to around 11.1volts.
Keep in mind, that the voltage keeps sagging farther under this condition because the FC will not drop the throttle back, until the potential rises._ 

I have include the log. But they only show the effects of the issue. And not the response of the FC... Because it is not responding to throttle commands. This made it very hard to chase down.
So the `RC throttle command` and `set point throttle ` only respond once the pack voltage starts to rise back towards the top end of the 3S range.

To start with.. I did not rule out the possibility of the Crossfire RX not responding to the throttle command.
So I used Logic switch's and Special functions it ping the CRSF telemetry. To give a buzzer sound on my TX, when the CRSF received a throttle value below 50% stick command. To make sure that I wasn't imaging pulling the throttle back :-) lol
This responded instantly, proving that it was not the Radio.

I then used another logic and special function on the radio to change the OSD screen, under these two different conditions.
1) The first was to change to OSD screen when Roll stick input was given above 10%.  
    This react instantly, on the OSD. And the quad also react instantly.
2) On the second, I did the same as above. But to change the OSD screen when the throttle was
    dropped below 50%.
    Under this condition. The throttle value did not drop, nor did the OSD screen change. Until the
    pack voltage rose back near the high end of the 3S voltage range. 

At time index 00:40 is when the throttle stick command was given to drop the throttle. 
The noise around that area on the log, was me, turning the quad around at full throttle to slow it down and wait for the throttle to respond again.
At time index 00:44.160 is when the voltage starts to rises and the throttle output then responds to the stick input is again.

[Log.zip](https://github.com/betaflight/betaflight/files/3193687/Log.zip)

The DVR images show the battery voltage verses throttle value. How the throttle did not respond to the input command from the radio, until the battery voltage started to rises again.
I had the quad in a roll at this stage, trying to drop the load on the pack.

[image]
[image]
[image]

I have no idea how old this issue is, or in what firmware it may have turned up. I only noticed it a few days ago, after a hard climb out.

Thanks

McGiverGim, 2019-05-18:
Have you checked the Setpoint of throttle on the log vs rc command of throttle? I can't open it now to see how it behaves.

mikeller, 2019-05-18:
@Jetrell: Please provide the information you were asked to provide when opening the issue.

Jetrell, 2019-05-18:
Yes. The RC throttle command at the above mention time index, is 99% and the set point is 88.5%

Jetrell, 2019-05-18:
Sorry.. I meant to include the setup. But forgot.
[diff all.txt](https://github.com/betaflight/betaflight/files/3193754/diff.all.txt)

Jetrell, 2019-05-18:
Forgot to mention as a side note. I tried and tried to get this issue to show up with 3S. But the cell voltage would never drop into the 2S range. So this issue never showed up on 3S, only on 4S.

McGiverGim, 2019-05-18:
It usually don't climbs from 88-90 because the air mode keeps it low to not sure the motor, but this is different from the 24 from your captures. I will try to look at the logs later.

Jetrell, 2019-05-18:
> 
> 
> It usually don't climbs from 88-90 because the air mode keeps it low to not saturate the motor, but this is different from the 24 from your captures. 

Sorry @McGiverGim  To clarify... The 24% is the point on the last image that throttle control came back again, at 11.4v.  This was the stick position that I dropped the throttle back too, 4 seconds earlier.

McGiverGim, 2019-05-18:
The log is very strange. Is clear something is wrong, but I'm not sure what is. The RC Command changes from 100% to 25% in a very linear way. It is clear that they are not your movement. But to me is not that when the voltage recovers, the throttle recovers. It seems the inverse: the throttle recovers and then the battery voltage grows.

[image]

Are you sure is not a problem with the TX? I don't know which debug mode serves to record the input received before any smoothing. @etracer, can you confirm which debug is?

Jetrell, 2019-05-18:
@McGiverGim There are no mixers in my TX that are throttle related. 
From my first post. The first thing I did was to rule out any RC link issues with the crossfire. 
 
Plus... I have few different DVR footage's from the time I discovered this. It is only after the recovery of the throttle output... that the input value I pulled the gimbal back too, then appears in the log and on the DVR.
It is varied. 20%, 24%, 32% etc.  

I even disabled `battery cell count based automatic PID profile switching`. And just use Profile 1, which is my 4S PID profile. But nothing changed with this.

It doesn't do it every time though. As can be seen from this particular log, it only did it one time in the 4 times that I tried to induce it. That was my whole purpose of flying today. And this log example was the best to see the effect from. 
Its really baffling to me...

McGiverGim, 2019-05-18:
What it seems is that something "locked" (logically, no physically) the throttle up about the second 00:39.196 (a little before your roll move) because since here there is not any "imperfection" in the throttle RC Command line, it is fixed at 1997us. And at 00:44.153 it recovers to the real value.
If it is an FC problem, it must be a very recent commit, because we don't have any other complains. But I will not discard a hardware (TX or RX maybe) problem.

Jetrell, 2019-05-18:
It does seem like a lock up alright. 
The odd thing.  Was how all axis's still responded to TX/RX commands. I was still in complete control of the quad, except for the throttle.
And I tried and tried to get it to do the same on 3S. But it would not do it once.

Here is another log of it a few days ago. It might reveal something else.

[BTFL_BLACKBOX_LOG_.zip](https://github.com/betaflight/betaflight/files/3194027/BTFL_BLACKBOX_LOG_.zip)

etracer65, 2019-05-18:
Most likely causing a brown-out in the receiver because of the low voltage. You're killing that battery with the cell voltage dropping all the way to 2.7V! If the battery is sagging that much with only ~50A then it's already junk.

The flight controller can only respond to the inputs it receives from the receiver and it's getting what you're seeing in the log for throttle.

McGiverGim, 2019-05-18:
Exactly the same, from 00:05.720 to 00:12.103 the throttle is locked at 1997us.

What is strange to me too is that the Yaw is almost locked too (not at all, but near perfect) at 1500us. Maybe simply you are very precise at the movement ;-)

Jetrell, 2019-05-18:
@etracer65 that was something that did cross my mind as well.
Both the 4S batteries that I use on this quad are 700maH 80C batteries. And this 3" does have 1408 motors.
So it does pull a lot of current. 
I just find it odd that I can run 3S 850maH 75C (of the same brand or battery) with a current drain of 52amps, and the voltage sagging to 9.6v. Without this problem taking place on 3S.

etracer65, 2019-05-18:
Sagging to 9.6v on 3S is only 3.2v per cell. That's a big difference from the 2.7v you're getting on 4S. When the voltages get that low the cell behavior will be unpredictable. The voltage reported in Betaflight is filtered and smoothed so you can't see the actual severe fluctuations that are likely taking place when the cells are under that much duress. This is not a Betaflight issue. Just get better batteries or don't use 4S on that quad.

Jetrell, 2019-05-18:
> 
> 
> When the voltages get that low the cell behavior will be unpredictable. The voltage reported in Betaflight is filtered and smoothed so you can't see the actual severe fluctuations that are likely taking place when the cells are under that much duress. 

That makes sense. I'll get some higher C rated batteries and do some more testing.
I'm not blaming betaflight. I just wanted to run it by you guys, to cover every possibility.

@McGiverGim @mikeller @etracer65 
Thank you for your time.

Jetrell, 2019-06-30:
@etracer65 @McGiverGim @mikeller 
I thought i'd report back on the cause of this issue. 
In a nut shell... It was the TBS nano RX holding its output positions, when the UART TX line experiences specific external EMI noise. 

I was lucky in the case of the quad. It only affected the throttle channel. 
And not just at the high end throttle ..... as I spoke of earlier in this post. 
This picture, shows how it affected low throttle as well.  See the `current value` vs the `throttle value`.  I actually had the throttle at 84%.  Trying to miss the ground.

[image]

But wasn't so lucky with a fixed wing using iNav. That had a rather noisy PAN servo, that was using the same supply as the Nano RX.  
(new TBS install) Under fast servo motion, it too put the Nano RX into an output hold state. Which left the flight controller unaware of any changing input channel states. 
And because the RSSI and LQ channel also held their last position. RTH recovery would not react either, even if the TX was turned off.  

Rather annoyed with that... it wrote the plane off.
That plane had worked fine with the same gear and wiring for over 12 months. It was just because of the TBS nano's lack of filtering, that this happened.

I wonder if others have experienced this... But have been unaware of what the cause was?   As you guys know. It was very illusive. And took a lot of chasing down.
I might re open this post. To give others a chance to reply if they have experienced a similar issue with there Nano RX.

mikeller, 2019-06-30:
Ouch. That's ugly.

closed 2019-08-06 by Jetrell