betaflight/betaflight #8298: discalibrating acc during flight after switch from acro to angle mode

smarq8, 2019-05-18:
I currently try to learn Line of Sight freestyle in ACRO mode and as a safe switch I use ANGLE mode. Today I alredy twice noticed that after switch from ACRO to ANGLE then my quad get weird angle as neutral position, somthing about 20-30 deg in random direction (once it was dangerous for me).
The only thing for fix that angle was acc calibration again. Landing, reboot, power off and off didn;t work.
Currenlty I am not sure what is certain reason why it happen, it might be self recalibration acc during flight or something happen during switchng mode

My safe switch is 3 pos switch, and it is setup in this way: ANGLE | HORIZON | ACRO
My FC: Flip32 F4

**Flight controller configuration**
```
# diff

# version
# Betaflight / OMNIBUSF4 (OBF4) 4.0.0 Apr 11 2019 / 07:30:28 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name REVO
manufacturer_id 

# name

# resources

# timer

# dma

# mixer
mixer CUSTOM
mmix 0  1.000 -1.000  0.772 -1.000
mmix 1  1.000 -1.000 -0.772  1.000
mmix 2  1.000  1.000  0.772  1.000
mmix 3  1.000  1.000 -0.772 -1.000

# servo

# servo mix

# feature
feature -OSD
feature -AIRMODE
feature -RX_SPI
feature RX_SERIAL
feature TELEMETRY

# beeper
beeper -ON_USB

# beacon

# map
map AETR1234

# serial
serial 0 4096 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 900 1300 0 0
aux 2 2 2 1300 1700 0 0
aux 3 13 4 1700 2100 0 0
aux 4 28 2 1700 2100 0 0
aux 5 47 3 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set acc_calibration = -87,63,-128
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = IBUS
set motor_pwm_protocol = DSHOT600
set vbat_min_cell_voltage = 340
set vbat_warning_cell_voltage = 355
set current_meter = NONE
set vbat_scale = 111
set yaw_motors_reversed = ON
set small_angle = 180
set report_cell_voltage = ON
set ibus_sensor = 3,0,0,0,0,0,0,0,0,0,0,0,0,0,0

# profile
profile 0

set vbat_pid_gain = ON
set level_limit = 80

# rateprofile
rateprofile 0

# end the command batch
batch end
```

etracer65, 2019-05-18:
This is normal and unfortunately not much we can do about it. During flight the attitude estimate (the internal view of the quad's orientation) is maintained by estimating the orientation based on gyro movement fused with converging data from the accelerometer. During typical acro flight the accelerometer is not useful for orientation because the "accelerations" in flight can't be distinguished from gravity. So during flight with lots of quick movements (rolls, flips, etc.) the attitude estimate can drift from reality due to the accuracy limits of the gyro data, and the inability to use the accelerometer data. So if you immediately switch back to a self-level mode after this the attitude estimate can be significantly incorrect. Flying more "calmly" for a few seconds before engaging a self-level mode can allow the attitude estimate to converge to a more accurate representation because the accelerometer cam be "trusted" more.

You could experiment with increasing `imu_dcm_kp` as this is the "strength" of the convergence based on accelerometer data. This would let it correct the orientation quicker when in stable flight, but at the same time possibly cause attitude estimate errors when inmate dynamic flight.

smarq8, 2019-05-18:
I exactly understand how FC estimate its orientation and I'm aware why I can not trust acc while drone is moving etc (I alredy writed own FC), but still I do not understand why it's require to recalibrate acc. I would understand if it would be 2-3 deg (measurement error) but 20-30 deg is somthing I can not imagine. Its also understandable for couple of a seconds after switch but not permanently. I always expected that once I calibrate acc then FC save current attitude as reference and then in theory in ANGLE mode FC should try to repeat that reference in neutral sticks position. So in this case it look like my "reference" is changed with estimated "reference" inflight after switch from ACRO to ANGLE? Please explain that.

Docteh, 2019-05-18:
If you've got an OSD you should turn on the artificial horizon for a flight or two. Comparing its horizon to real horizon should be consistent before/after your acro flight. But then maybe you don't have FPV setup if you're flying LOS. What about telemetry? 

I know when I smack my 75mm drone onto the floor the ACC gets a bad idea of which way is down. Maybe you are similarly having that happen? If I feel like it I sit on the ground for a few seconds, while the artificial horizon corrects then take off. But when I'm impatient I've noticed that restarting betaflight fixes the ACC. 

If my ACC got confused without hitting the ground I'd be looking at swapping the flight controller out. 

I'm assuming you're describing a scenario where you put the drone back into STAB/HOR mode and its happily listing to the side after its had time to go level (second or so?) instead of being perfectly level.

etracer65, 2019-05-18:
First of all to correct some incorrect assumptions, the ACC doesn't lose or change it's calibration. As I said, the attitude estimate uses the gyro almost exclusively when in acro flight. Due to the limitations in the gyro's resolution, gyro noise from vibration, etc. this estimate can drift. The longer the flight the more potential drift (particularly with flight using high deg/sec activities). The gyro-based attitude estimate is a "dead-reckoning" calculation. The convergence to actual attitude uses the ACC but only in times where the total acceleration is within the range of about 0.8 to 1.2G (if I recall correctly). And even then the convergence is quite slow as the ACC data isn't that trusted in flight (the `imu_dcm_kp` controls this). If you were to hold the quad steady after switching to self-level you would see the attitude estimate return to level (slowly).

If you want to improve this then by all means have a look at https://github.com/betaflight/betaflight/blob/master/src/main/flight/imu.c

mikeller, 2019-05-19:
@smarq8:

> The only thing for fix that angle was acc calibration again. Landing, reboot, power off and off didn;t work.

If this is the case, then your accelerometer is definitely broken. Replace the flight controller.

closed 2019-05-19 by mikeller