betaflight/betaflight #8304: SD card: Startup failed on MATEKF722SE

smaugfm, 2019-05-19:
The card is SanDisk Ultra 16GB, I am successfully using it on another board Matek F722-STD, so the card itself is fine. 
I tried clean flashing 4.0.2, 4.0.0, the earliest one available 3.5.3, 4.1.0 #1647 from development, and also tried 4.0.0 from matek [website](http://www.mateksys.com/Downloads/FW/betaflight_4.0.0_MATEKF722SE.hex) - same results.
Tried turning sdcard_dma = ON - no luck, and tested with Lipo connected - same results.

smaugfm, 2019-05-19:
Oh, I see you transefered the issue to betaflight repo. Then I could use some help. Maybe there is a way to figure out what is causing it not see sd card?

etracer65, 2019-05-19:
Try reformatting the SD card with your computer. But likely there's a problem with the board if the same card works in another board. I also have a couple of the MATEKF722-SE and they work fine.

smaugfm, 2019-05-19:
@etracer65 I tried that too, doesn't work. If I could compile a firmware with some verbose debug info to look into the internals.

etracer65, 2019-05-19:
That's not really possible without programming skills and debugging tools/equipment. Your best bet is to check for warranty replacement.

closed 2019-06-25 by stale[bot]