betaflight/betaflight #8311: Re arm motors after signal loss as an option

nigelsheffield, 2019-05-20:
Please can we consider having the handling of Rx signal loss and recovered to allow for the temporary loss of signal without just allowing my heavy quad to just fall out of the sky ...

Again today I lost signal for long enough to failsafe but signal recovered again (  I think ) once the quad dropped down somewhat and could have been saved if it had just re armed itself like my silverware quads do when flying in sketchy areas or too high for signal ..

I know the re arm thing is for safety when a quad has crashed and so that it doesn't just start motors up in the ground by accident if the user forgets disarm but I feel the option could be included to allow it to be changed for those of us who don't forget to do things like that but really want to be able to get control back instantly in the sky after a failsafe if signal comes back.

Having my quad fall to the ground whilst seeing perfectly out of the camera and having rx signal is frustrating.

mikeller, 2019-05-20:
@nigelsheffield: You answered your own question there:

> I know the re arm thing is for safety when a quad has crashed

> the option could be included to allow it to be changed for those of us who don't forget

This option is already included. Just disarm on the TX as soon as RX loss is detected, and re-arm again once the connection is recovered, and you will be fine.

nigelsheffield, 2019-05-20:
No, it didn't arm again, I tried , I had about a second at most to  think about in arming and re arming again, I hit the switch but didn't work ...

Sorry but I think this should be thought about regarding safety ...
Likely hood of me crashing and leaving the quad armed whilst I look for it=zero.
Likely hood of me crashing into someone or something and causing injuries/damage because my quad falls out of the sky when I could have saved it when control signal came back = pretty high....

If we are really worried about the re arming by mistake thing how about a extra failsafe level maybe on timer or something, if signal comes back after 1 minute don't automatically re arm.

Like I said, I have never had a problem re arming by mistake with my silverware quads but have failsafed loads of times but recovered once signal comes back...

Think about a simple punch out, at present if we go out of range we have to mess about with arming and hope it works before it hits the ground, how dangerous is that for by standers or property nearby or even ourselves?

I just don't understand the reluctance to implement this even if it's as a advanced option and not turned on by default.

mikeller, 2019-05-20:
@nigelsheffield:

> No, it didn't arm again, I tried , I had about a second at most to think about in arming and re arming again, I hit the switch but didn't work ...

Could it be that you forgot to set the throttle to zero before arming?

> If we are really worried about the re arming by mistake thing how about a extra failsafe level maybe on timer or something, if signal comes back after 1 minute don't automatically re arm.

The problem is not users re-arming, the problem is craft falling out of the sky with props twitching erratically, as it will happen with what you are proposing if the RC connection is intermittent (and the pilot is trying to 'save the quad', just like you describe what you are doing). This is a way worse scenario than a craft falling out of the sky like a brick.

Let's face it, if you take the risk of flying over people you already assume responsibility for people potentially being hit by your craft as a worst case - it might not be a failsafe, but a broken battery connection (or any other of a number of other catastrophic failures), so this might always happen. If you do not want for people to be potentially hit by your craft, then don't fly over them in the first place.

> Like I said, I have never had a problem re arming by mistake with my silverware quads but have failsafed loads of times but recovered once signal comes back...

Just for the record, how many watts of power output have your silverware craft, and how many watts of power output have your craft using Betaflight? See a difference there?

nigelsheffield, 2019-05-20:
Throttle was at zero before trying to arm again, just got nothing or at least nothing IN TIME anyway..
Of course eratuc props spinning during the crash is worse than a dead drop, I do see that, buying my experience the connection comes back as soon as I drop back from the excessive height or clear the object blocking the signal ( in this case was a tree and I had cleared the tree just as it failsafed so could of carried on flying ...

My silverware quads are various type ranging from whoops to a 150 size brushless with tons of power and speed, I don't just fly brushed with it like most people .. so no difference between betaflight and silverware in this case ....

Flying over people and property is not something I do intentionally but I'm sure you know how it is sometimes people appear out of no where lol.
I minimize the risk as much as possible and that is what I'm trying to do here, if there is and hands of saving a free fall quickly and easily I'll take it, if the signal is so poor and unreliable that it's just twitching and not responding properly  I WILL ditch it myself without hesitation..

I've flown for over 30 years and never hit anyone , if my gliders stopped responding to my controls I would expect to regain control over them IF they happened to come back into range and not just dive themselves down into whatever was bellow...
I know this is a bit different and a total shut down is the safest bet often but there are many times when when we could save a temporary Rx loss becoming a crash....
Would be nice to have the choice in any case.

It's why I suggested having it as a advanced option or something so that into those that understand the dangers could activate it..

bizmar, 2019-05-20:
I have re-armed mid fall a few times*, it does work. if you have accelerometer set up, there's a safety feature that prevents it from arming when not level. 

You can increase the stage 1 failsafe window, try the very risky "land" failsafe option, or add GPS and use GPS rescue. A have my throttle set to hold, for stage one, I tried setting a channel to flip it into autolever when in stage 1, but it was too disorientating, so I abandoned that idea. 

*Have to set double arming on the radio, and in a way that allows for quick re-arming.

nigelsheffield, 2019-05-20:
I'm not saying that re arm doesn't work, I'm sure it does, just in my cases so far I haven't had much (any) success doing it.
I don't have accelerometer turned on.
Increasing stage 1 failsafe might have saved today's crash after flying behind a tree..
Wouldn't help with a punch out though because it would just carry on going up and further away from the signal?

bizmar, 2019-05-20:
If you are below 30m going up will almost always increase signal strength. As the line of sight to your receiver will be better, less signal will be lost due to ground reflection and absorption. If you are having failsafes this often I would really look at your equipment and more importantly antenna placement, both quad and radio. If you loose signal, you should raise up your radio above your head if you can :) Failsafes do happen "randomly" I usually get 3-5 unexplained ones a year.

Anyway, this is not a support page.

nigelsheffield, 2019-05-20:
I'm not asking fkr support for my failsafes lol.
I'm not getting loads of failsafes with betaflight , I just got this one today first one in ages and I know what caused it ( flying behind myself and behind 2 trees , rssi on a new quad was not calibrated properly so no warnings ), I'm asking for a rethink of the failsafe options to cope with temporary loss of signal , 
I've heard of several people who simply lost control after a punch out over head and then crashed because it didn't arm in time to save it when coming back down through a perfectly good signal.

If it's just me that wants this then it's ok, forget it.

closed 2019-05-20 by nigelsheffield