betaflight/betaflight #8313: Resolve "Two system_stm32yxxx.c" problem

jflyper, 2019-05-20:
There are two `system_stm32yxxx.c` files for each platform, one in `startup/` and another in `drivers/` directory. There are at least two confusions caused by these twins; (1) things going into these files are not well separated, and (2) the same file name.

The problem should be solved, starting with given them different names.

hydra, 2022-08-26:
I agree this should be resolved, I'm always loading up the wrong one.

The simplest solution that addresses point 2 is to rename the non-HAL files them from `system*.c/h` to `mcu*.c/h`.

I agree that the content of the non-HAL `system.*` should likely be extracted so it's not so much of a garbage bin.  perhaps by creating an `mcu` folder in `drivers` and splitting up the content into probably at least `memory.c`, `reset.c`, `boot.c`.

Prajwal-Koirala, 2023-12-27:
**Here's a summary of the proposed solution and its key points:**

**Problem:**

- The current naming of files as "system*.c/h" is causing confusion and accidental loading of incorrect files.
- The non-HAL "system.*" files contain a mix of functionalities, making it a "garbage bin" and less organized.

**Proposed Solution:**

1. **Rename non-HAL files:**
   - Rename files from "system*.c/h" to "mcu*.c/h" to clearly distinguish them from HAL-related files.

2. **Restructure non-HAL content:**
   - Create a new folder called "mcu" within the "drivers" folder.
   - Move the non-HAL "system.*" files into the "mcu" folder.
   - Split the content of the non-HAL files into more specific files, such as:
     - memory.c
     - reset.c
     - boot.c

**Benefits:**

- **Clarity:** Clearer distinction between HAL-related and MCU-specific files, reducing confusion.
- **Organization:** Improved organization by separating non-HAL content into a dedicated folder and more specific files.
- **Maintainability:** Easier maintenance and understanding of code due to better structure.

**Additional Considerations:**

- **Thorough testing:** Conduct thorough testing after making these changes to ensure everything functions as expected.
- **Update documentation:** Update any relevant documentation to reflect the new file names and structure.
- **Communication:** Communicate the changes to other developers working on the project.

Prajwal-Koirala, 2023-12-27:
##  How to Rename and Restructure Files:

**Step 1: Renaming Files:**

1. **Backup**: Before making any changes, back up your project directories in case of any unforeseen issues.
2. **Identify Files**: Look for all files named "system*.c" and "system*.h" outside the HAL directory.
3. **Rename Individual Files**: Use your preferred file explorer or IDE to rename each file individually. Change "system" to "mcu" while keeping the rest of the filename intact. (e.g., `system_clock.c` becomes `mcu_clock.c`).
4. **Bulk Rename**: Consider using a tool like `sed` or a dedicated file renamer if you have a large number of files to rename. Make sure to test the script on a small set of files before applying it to everything.

**Step 2: Restructuring Content:**

1. **Create Folder**: In the "drivers" directory, create a new folder called "mcu".
2. **Move Files**: Cut or copy the renamed files from their previous locations and paste them into the "mcu" folder.
3. **Extract Content**: Analyze the content of the "mcu*.c" files. Identify functionalities that can be separated into independent files with clearer names like "memory.c", "reset.c", and "boot.c".
4. **Refactor Code**: Refactor the code within the files to isolate these functionalities into separate files. Make sure to properly update header file references and function calls to reflect the new locations.
5. **Clean Up**: Once the refactoring is complete, you can consider removing any redundant code or empty folders left behind in the original locations.

**Additional Tips:**

* **Version Control**: If you are using a version control system like Git, commit your changes in logical steps with clear and descriptive commit messages.
* **Testing**: Ensure you thoroughly test your code after each step to verify that everything functions correctly.
* **Documentation**: Update any relevant documentation, such as makefiles or build scripts, to reflect the new file names and structure.
* **Communication**: Inform other developers working on the project about the changes you made and the reasons behind them.

By following these steps, you can efficiently rename and restructure the non-HAL "system.*" files, improving project clarity, organization, and maintainability.