betaflight/betaflight #8314: Airbot F7 Current Sensor doesn't work right

MudFPV, 2019-05-20:
**Describe the bug**
If I set up the current sensor in Betaflight for the new Airbot F7, the current sensor shows a permant current and didn't change. If I plug in the Batterie, the current sensor shows a wrong current instantly. It doesn't matter, If I use the bf 3.5.5/3.5.6 or 4.0.2 for the Airbot F7.

Video of the Problem (yes, it is german, but you can see the Problem I mean):
https://www.youtube.com/watch?v=FE10cHSkd8s

**Expected behavior**
That the current sensor work right :)

**Flight controller configuration**
# name
name Chameleon TI

# resources
resource BEEPER 1 C05
resource MOTOR 1 C08
resource MOTOR 2 B06
resource MOTOR 3 C09
resource MOTOR 4 B07
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A15
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource I2C_SCL 1 B08
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 B09
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
[... 928 lines cut ...]
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650
set throttle_limit_type = OFF
set throttle_limit_percent = 100

# restore original rateprofile selection
rateprofile 0

# save configuration
save
**Setup / Versions**
 - Flight Controller: Airbot F7
 - Other components: ESC is a Furling32 4 in 1
 - how are the different components wired up: with a 4 in 1 cable

Betaflight Versions can be downloaded here:
https://drive.google.com/drive/folders/1sO50hiysjleC3RZCUnIVzwOTzYQskZYi?usp=sharing

mikeller, 2019-05-21:
@MudFPV: Can you please refrain from opening duplicate issues?

Duplicate of #8315.

MudFPV, 2019-05-21:
@mikeller
Yes sure, I'm sorry. I used a different browser and couldn't edit this one here.

Here is the actual issue, which is already solved.
https://github.com/betaflight/betaflight/issues/8315#issuecomment-494032911

closed 2019-05-21 by mikeller