betaflight/betaflight #8315: Airbot F7 Current Sensor doesn't work right

MudFPV, 2019-05-20:
**Describe the bug**
If I set up the current sensor in Betaflight for the new Airbot F7, the current sensor shows a permant current and didn't change. If I plug in the Batterie, the current sensor shows a wrong current instantly. It doesn't matter, If I use the bf 3.5.5/3.5.6 or 4.0.2 for the Airbot F7.

Video of the Problem (yes, it is german, but you can see the Problem I mean):
https://www.youtube.com/watch?v=FE10cHSkd8s

**Expected behavior**
That the current sensor work right :)

**Flight controller configuration**
Can be donwloaded in the link below

Betaflight Versions can be downloaded here:
https://drive.google.com/drive/folders/1sO50hiysjleC3RZCUnIVzwOTzYQskZYi?usp=sharing

mikeller, 2019-05-20:
@MudFPV: None of these are official Betaflight targets, as the AIRBOTF7 is not yet supported by Betaflight (it will be in the upcoming 4.0.3). As such, we can not provide you with any help for them (and they were provided by AIRBOT in breach of Betaflight's license).

Can you try a development build of the official AIRBOTF7 target, and see if the problem persists there?

Also, please refrain from opening multiple issues - you can always go back to an existing issue and edit the text you entered.

MudFPV, 2019-05-20:
@mikeller
I only have this Targets and the Problem exists in the official airbotf7 target too.

By the way, the Airbot F7 3.5.5 target in the Google drive link, is the target which airbot has send me.

Asizon, 2019-05-20:
There is not oficial and fi al target,please try with Betaflight oficial dev 4.1 version first

Asizon, 2019-05-20:
Also,you need to calibrate current sensor,it have defaults values

MudFPV, 2019-05-20:
@Asizon 
Yes okay. Can you say me an approximately time when the official target will be released?

AirbotFPV, 2019-05-20:
@MudFPV  

The Current ADC pin in the target defined wrong, it should be C01(in the target it is c02), will get it fixed now. 

If you enter CLI, resource the CURRENT_METER_ADC_PIN to C01, then it should works as it should be.

MudFPV, 2019-05-20:
@AirbotFPV  Can you write the exact command which I should type into the cli? Thanks!

Asizon, 2019-05-20:
Resource CURRENT_METER_ADC_PIN= C01`` and save

jflyper, 2019-05-20:
Nope.

```
resource adc_curr c1
```

AirbotFPV, 2019-05-20:
By the way, when you use Furling32 4in1, then try my scale first: 

Scale = 68

MudFPV, 2019-05-20:
@AirbotFPV 
With this command:
resource adc_curr c1

the problem isn't fixed in bf 3.5.5

Asizon, 2019-05-20:
> @AirbotFPV
> With this command:
> resource adc_curr c1
> 
> the problem isn't fixed in bf 3.5.5

Also need to resource rssi pin to another pin

AirbotFPV, 2019-05-20:
Yes, the RSSI is not defined, should be removed from the target too(will do that). 

C02 will be used for board which has an onboard current sensor(not over ESC).

jflyper, 2019-05-20:
And make sure pin assignments are correct by typing `resource list` (3.x) or `resource show` (4.x).

MudFPV, 2019-05-20:
Yeah so I should use C01 for adc_curr

And which pin should I use for rssi?

AirbotFPV, 2019-05-20:
What kind of RSSI output you have, on your receiver?

MudFPV, 2019-05-20:
@AirbotFPV 

It is a Graupner sumd receiver, has a rssi channel output on aux8

AirbotFPV, 2019-05-20:
No pin needed, it is not ADC RSSI, it combines in the SUMD data. 
You can map the channel to RSSI input, 
but you need to read the Graupner manual, to find out which channel it should be.

RSSI CH
Die Signalstärke des Empfängers wird anstelle der Servofunktion
auf dem ausgewählten Kanal bei SUMD, SBUS oder SP2048 ausgegeben. Gleichzeitig wird das digitale Summensignal auf die Anzahl
der Kanäle eingegestellt, so dass das RSSI Signal auf dem höchsten
Kanal anliegt. Dieses ermöglich mit Hilfe einer Flugkontrolle mit
OSD, wie z. B. 48377 die Anzeige der Signalstärke des Empfängers
in der Videobrille bzw. auf dem Bildschirm.
Einstellungen: OFF, CH8, CH12 und CH16

MudFPV, 2019-05-20:
Yeah it is on ch8 now.

But the current sensor doesn't work yet.
My resource list is here:
`A00: FREE
A01: FREE
A02: FREE
A03: FLASH_CS
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: CAMERA_CONTROL
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: LED 1
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: ADC_CURR
C02: FREE
C03: FREE
C04: FREE
C05: BEEPER
C06: FREE
C07: FREE
C08: MOTOR 1
C09: MOTOR 3
C10: FREE
C11: SERIAL_RX 3
C12: SERIAL_TX 5
C13: FREE
C14: FREE
C15: OSD_CS
D00: FREE
D01: FREE
D02: MPU_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE`

AirbotFPV, 2019-05-20:
you resourced the C01 as ADC_BATT, 

Please do what @jflyper posted.

MudFPV, 2019-05-20:
`A00: FREE
A01: FREE
A02: FREE
A03: FLASH_CS
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: CAMERA_CONTROL
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: LED 1
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: ADC_CURR
C02: FREE
C03: FREE
C04: FREE
C05: BEEPER
C06: FREE
C07: FREE
C08: MOTOR 1
C09: MOTOR 3
C10: FREE
C11: SERIAL_RX 3
C12: SERIAL_TX 5
C13: FREE
C14: FREE
C15: OSD_CS
D00: FREE
D01: FREE
D02: MPU_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE`

this is the actual resource list.

AirbotFPV, 2019-05-20:
I may need to see your connection between 4in1 and FC too. 

The huge reading means the ADC caught the wrong input.

MudFPV, 2019-05-20:
Yes on moment please

AirbotFPV, 2019-05-20:
Which ESC it is? 

If the connection is correct, then the result is weird, please wait, let me try the setup tomorrow(will test at the first time I can get the boards).

MudFPV, 2019-05-20:
It is a furling32 4 in 1 v1.2

AirbotFPV, 2019-05-20:
by the way, one more question, do you power the ESC when you check the reading?

MudFPV, 2019-05-20:
> by the way, one more question, do you power the ESC when you check the reading?

Yes

MudFPV, 2019-05-20:
Hmm weird.
No I don't have any spare parts here. I will test the other betaflight firmware (4.0.2)

AirbotFPV, 2019-05-20:
I don't think it is software/firmware related. 
can you take the parts down and send back to our Germany warehouse? 
I will ask my friend to check what is going on with your setup and send back to you(or replace for you if there is a hardware issue)

MudFPV, 2019-05-20:
I have a Dys F4 Pro v2 here.

If I know how to wire him up with the furling, I can test it

MudFPV, 2019-05-20:
[image]
Here can you see that there is the issue still

AirbotFPV, 2019-05-20:
if you like please send to my friend in Germany, that is the easiest way I think. 

If you want to try the furling 32 4in1 with the DYS FC. 

please connect the pwm 1234 to DYS motor output

VBAT and GND to where the DYS FC taking power. 

and the TLM pin, please just leave it open(no need for this test) 

the current output wire, to DYS FC's Current ADC pad. 

Before you mount the 4in1 to another FC, it will be great, that you can show me a better photo that I can see how everything connected.

MudFPV, 2019-05-20:
I will make you a video. Just one moment please

AirbotFPV, 2019-05-20:
Can you connect only the FC and 4in1, like this:
[image]

AirbotFPV, 2019-05-20:
Can you measure the voltage reading:

on the 4in1 current wire, when the ESC is powered on. (but FC is not connected )

Asizon, 2019-05-20:
Wait,telemetry output pins are mixed in the manual

MudFPV, 2019-05-20:
> Wait,telemetry output pins are mixed in the manual

Yes I now. tlm and current adc are changed

AirbotFPV, 2019-05-20:
@Asizon  do you have any problem with the Current reading?

MudFPV, 2019-05-20:
@AirbotFPV Please say me which cable is used for the current sensor? tlm or current adc?

Cause I got a different version of the airbot f7 than you

Asizon, 2019-05-20:
> @Asizon do you have any problem with the Current reading?

I can test it to the night,im working now

AirbotFPV, 2019-05-20:
On my board, when the Current pin set correctly,  the reading is correct:
[image]

When the current pin set wrong, and lipo plugged in, the reading is like what you got.

MudFPV, 2019-05-20:
Yeah I make the video of the cables now

AirbotFPV, 2019-05-20:
I see, lol,  that is the mistake @Asizon  Mentioned， 
the TLM and Current swapped in the manual, you should not swap the wires. 

12345678 to 12345678 will work please try.

MudFPV, 2019-05-20:
@AirbotFPV  nNo! in the manual you can see, that tlm and current adc must be swapped!

AirbotFPV, 2019-05-20:
That is the mistake @Asizon  mentioned, sorry for the mistake. 

Here is the correct pinmap:
[image]

Asizon, 2019-05-20:
Nice!

MudFPV, 2019-05-20:
But why do I have a wrong pinmap? Airbots mistake?

As my manual show, Airbot F7 uses gnd, vbat, tlm, current, m1-m4 and the Furling32 4 in 1 esc uses gnd, vbat, current, tlm, m1-m4 (from left to right)

MudFPV, 2019-05-20:
[image]

AirbotFPV, 2019-05-20:
Yes, one version of the manual was wrong with the first batch.

MudFPV, 2019-05-20:
Haha funny :D 

I will change it now and test it. See ya back in 5 minutes

AirbotFPV, 2019-05-20:
OK, good luck:)

MudFPV, 2019-05-20:
@AirbotFPV Hahahah thanks a lot! Works now

AirbotFPV, 2019-05-20:
have fun:)

closed 2019-05-21 by MudFPV

mikeller, 2019-05-21:
@AirbotFPV: Are you planning on submitting a pull request fixing the defective parts of the target definition for the AIRBOTF7 that were identified in this issue any time soon?

AirbotFPV, 2019-05-21:
yes, my friend already did that I think, let me ask him tonight.