betaflight/betaflight #8317: Talon F7 V2 no bidirectional DSHOT possible

MartinFPV, 2019-05-20:
**Describe the bug**
When trying to activate the bidirectional DSHOT under Betaflight. Turn on 2 motors only briefly, then nothing happens anymore. Before activating the Bidirectional DSHOT, all 4 engines were unremarkable.

**To Reproduce**
https://youtu.be/rpyD-B7AGoQ

**Expected behavior**
All 4 motors should behave normally with activated bidirectional DSHOT.

**Flight controller configuration**
``` 
# diff
# version
# Betaflight / CLRACINGF7 (CLR7) 4.0.2 May 5 2019 / 11:41:04 (56bdc8d26) MSP API: 1.41
# start the command batch
batch start
board_name CLRACINGF7
manufacturer_id
# name
# resources
# timer
# dma
# mixer
# servo
# servo mix

# feature
# beeper
# beacon
# map
# serial
# led
# color
# mode_color
# aux
# adjrange
# rxrange
# vtx
# rxfail
# display_name
# master
set gyro_sync_denom = 2
set acc_hardware = NONE
set mag_hardware = NONE
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set vbat_scale = 81
set pid_process_denom = 1
set scheduler_optimize_rate = ON
# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end
#
```
**Setup / Versions**
 Talon F7 BB V2 20x20  @BF 4.0.2 (Target CLRACINGF7)
 Furling32 4in1 mini ESC - F3 32bit BLHELI ESC  @ BLHeli_32 Test code Rev32.6.2 (Target Furling32_4in1_Multi_32_62.Hex)
[image]

 

**Additional context**
Add any other context about the problem here.

RipperDrone, 2019-05-20:
inverted DShot needs 32.63, doesn't it?

RipperDrone, 2019-05-20:
https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/BLHeli_32%20Test%20code%20Rev32.6.2%20hex%20files/Rev32.6.3%20for%20inverted%20bidirectional%20dshot%20only

MartinFPV, 2019-05-20:
Rev32.6.3 for inverted bidirectional dshot is only needed from Betaflight 4.1, right? At 4.0.2 it is still the uninverted 32.6.2.

RipperDrone, 2019-05-20:
of course, thought you were trying inverted. My bad, sry...

etracer65, 2019-05-20:
Since that board uses the CLRACINGF7 target, it would be save to assume the same restrictions apply.

[image]