betaflight/betaflight #8327: BF 4.0.2 LED strip GPS overlay off by one

grizewald, 2019-05-21:
If you configure an RGB LED on a strip with the GPS overlay, it flashes one more time than it should before pausing and beginning the flash count again.

For example, where the OSD shows 7 satellites detected, the LED(s) blinks 8 times.

This is on a DYSF4PRO running 4.0.2 firmware. The LED strip config was pasted in from a 3.4 dump.

mikeller, 2019-05-22:
@grizewald: Are you sure that this wasn't the case in older versions as well? Looking at the code generating the blink pattern (https://github.com/betaflight/betaflight/blob/1541466dac570f54d9eb24cd47d439b41ef8b41e/src/main/io/ledstrip.c#L781), it seems to be designed to blink once for 0 satellites, and then once for every satellite found, and it has not been changed in years.

grizewald, 2019-05-22:
I guess it may well have been the case in older versions. I only noticed this because my GPS no longer seems to pick up more than 7 satellites since I upgraded the machine it is on to the 4.0.2 release.

I'm not sure it's very logical the way it works just now...

mikeller, 2019-05-22:
@grizewald: I didn't say it was logical, it's just the way it was implemented originally. And I guess a case can be made for having a distinct 'system working but 0 satellites found' pattern.

mikeller, 2019-05-25:
@grizewald: Potentially. Feel free to open a pull request to change it.

closed 2019-07-01 by stale[bot]