betaflight/betaflight #8330: Target SPRacingF3NEO not building

mikeller, 2019-05-22:
From https://github.com/betaflight/betaflight/pull/8321#issue-280867950 reported by @hydra:

> when compiling master a linker error occurs due some baro code referenced somewhere (baroConfig_System undefined) but the defines and src definitions were missing.

hydra, 2019-05-22:
The actual error I get is:

```
Linking SPRACINGF3NEO 
Memory region         Used Size  Region Size  %age Used
           FLASH:      246156 B       252 KB     95.39%
    FLASH_CONFIG:          0 GB         4 KB      0.00%
C:\Users\Hydra\AppData\Local\Temp\cculaMJF.ltrans0.ltrans.o: In function `findSerialPortIndexByIdentifier':
             RAM:       35800 B        40 KB     87.40%
             CCM:          2 KB         8 KB     25.00%
D:\My Documents\dev\embedded\betaflight/./src/main/io/serial.c:168: undefined reference to `barometerConfig_System'
       MEMORY_B1:          0 GB         0 GB     -1.#J%
collect2.exe: error: ld returned 1 exit status
make[1]: *** [Makefile:311: obj/main/betaflight_SPRACINGF3NEO.elf] Error 1
make[1]: Leaving directory '/cygdrive/d/My Documents/dev/embedded/betaflight'
make: *** [Makefile:472: hex] Error 2
"make TARGET=SPRACINGF3NEO -j12 DEBUG=INFO" terminated with exit code 2. Build might be incomplete.
```

Confirmed that it's not a build artifact as it still does it after issuing `rm -rf obj/main/SPRACINGF3NEO`

hydra, 2019-05-22:
it appears the issue comes from the `src/main/SPRACINGF3NEO/config.c/targetConfiguration()`

```
    barometerConfigMutable()->baro_hardware = BARO_DEFAULT;
```

the linker error above is a bit of a red-herring.

as noted in `https://github.com/betaflight/betaflight/pull/8321#issue-280867950`

I'll delete a couple of lines from that method as the defaults in `pgResetFn_barometerConfig` are already correct.

hydra, 2019-05-22:
https://github.com/betaflight/betaflight/pull/8332/files created to address the underlying issue.

PR #8332 (2019-06-04): SPRACINGF3NEO - Remove some unneeded target config.
The defaults in pgResetFn_barometerConfig and pgResetFn_compassConfig
are fine.

fixes #8330
closed 2019-06-04 by mikeller