betaflight/betaflight #8334: Manufacturer ID for EXF RC.

Linjieqiang, 2019-05-23:
As per https://github.com/betaflight/betaflight/blob/master/unified_targets/docs/CreatingAUnifiedTarget.md, please assign manufacturer ID for EXF RC.

- Requested Id: EXF
- Name: EXF RC
- Facebook: https://www.facebook.com/EXUAV/

mikeller, 2019-05-25:
@Linjieqiang: Do you have anything more permanent than a Facebook page where users can get information on your products, or contact you? http://http://www.exuav.com/ that you are listing on the Facebook page seems to be inactive.

Linjieqiang, 2019-05-27:
Hello @mikeller .The website is under construction. But you can contact with us using facebook.(https://www.facebook.com/profile.php?id=100013651722728)

mikeller, 2019-05-27:
@Linjieqiang: Ok. But your website will be at http://www.exuav.com/ when it is finished?

Linjieqiang, 2019-06-19:
@mikeller Forgive me for forgetting to reply to you. 
We may have suspended the construction of the website. 
Later, our company will focus on the OEM business of hardware, but I want to retain a common brand called "EXF" as a follow-up brand.
Thanks.

mikeller, 2019-06-20:
@Linjieqiang: If it's going to be an OEM, then this probably means that marketing hardware support will be done by the companies selling your products under their label?

Linjieqiang, 2019-06-24:
@mikeller Yes, they will be responsible for product technical support and after-sales work.

mikeller, 2019-06-24:
Ok. In this case these companies should be entered as the 'manufacturer' - no point asking users of these products to contact the OEM manufacturer for after sales support.

Linjieqiang, 2019-06-25:
Ok, just follow your advice. I closed this issue.

closed 2019-06-25 by Linjieqiang