betaflight/betaflight #8336: BF 3.5.4 - MATEKF722SE (MF7S) kiss ESC flashing does not work

dk7xe, 2019-05-23:
**Describe the bug**
KISS ESC32A flashing does not work on MATEKF722SE (MF7S) with kiss flashloader in Octo setup.

**To Reproduce**
follow steps on https://github.com/betaflight/betaflight/wiki/Betaflight-ESC-pass-through
Kiss flashloader send data but ESC don't get flashed. LED's do not blink at all.

**Expected behavior**
After issuing 'escprog ki 255' and disconnecting betaflight-configurator the kiss flashloader should be able to flash the KISS ESC via any BF FC.

**Flight controller configuration**
`diff

- version
- Betaflight / MATEKF722SE (MF7S) 3.5.4 Dec 21 2018 / 12:00:31 () MSP API: 1.40

board_name MATEKF722SE
manufacturer_id 

- name
name TheBeast

- resources

- mixer
mixer OCTOFLATX

- servo

- servo mix

- feature
feature GPS
feature AIRMODE
feature ESC_SENSOR

- beeper

- beacon

- map

- serial
serial 0 2 115200 57600 0 115200
serial 2 16384 115200 57600 0 115200
serial 3 32 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

- led

- color

- mode_color

- aux
aux 0 0 0 1900 2100 0 0
aux 1 13 1 1900 2100 0 0

- adjrange

- rxrange

- vtx

- rxfail

- master
set motor_pwm_protocol = DSHOT150
set current_meter = ESC
set battery_meter = ESC
set gps_provider = UBLOX
set osd_vbat_pos = 2486
set osd_throttle_pos = 2146
set osd_crosshairs_pos = 2318
set osd_current_pos = 2465
set osd_craft_name_pos = 2475
set osd_gps_speed_pos = 2115
set osd_gps_sats_pos = 2136
set osd_warnings_pos = 2409
set osd_stat_min_rssi = OFF
set osd_stat_used_mah = OFF
set osd_stat_max_alt = ON

- profile
profile 0

- rateprofile
rateprofile 0 `

**Setup / Versions**
 - Matek BetaFlight FC F722-Mini
https://fpv-racing-forum.de/thread/10739-thebeast-oder-196kmh-sind-nicht-genug-für-einen-octo/

it seems Octo setup is hardly used ;)
https://github.com/betaflight/betaflight/issues/2973

dk7xe, 2019-05-23:
@ronlix ..under Ubuntu at least the flashloader was working. It's time that you do a FC that supports Octo ;)

etracer65, 2019-05-23:
ESC serial passthrough doesn't work on F7 - never has. Needs to be developed and there's little incentive to do so as most people use the BLHeli 4way interface. The only use-case is with KISS ESCs as you've found.

dk7xe, 2019-05-23:
Noted, thanx

closed 2019-06-29 by stale[bot]