betaflight/betaflight #8337: CLRacing F4 Mini - SS broken in 4.0.2

SlipstreamFPV, 2019-05-23:
**Describe the bug**
Softserial mapped to SAT pin (resource SERIAL_RX 6 NONE, resource SERIAL_TX 11 C07) with smartport assigned to serial TX 11 works as expected in BF 4.0.0, in BF 4.0.2 it is broken.

**To Reproduce**
Use 4.0.0 with the diff below - works. Update to 4.0.2 with the same diff - doesn't work.

**Expected behavior**
Smartport working via softserial assigned to C07.

**Flight controller configuration**
```
# diff all

# version
# Betaflight / CLRACINGF4 (CLR4) 4.0.0 Apr 11 2019 / 07:01:44 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CLRACINGF4
manufacturer_id 
mcu_id 002e00353037470436353634
signature 

# name

# resources
resource SERIAL_TX 1 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 11 C07
resource SERIAL_RX 4 NONE
resource SERIAL_RX 6 NONE

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature TELEMETRY
feature ESC_SENSOR

# beeper

# beacon
beacon RX_SET

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1800 2100 0 0
aux 1 13 2 1800 2100 0 0
aux 2 35 1 1400 1600 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 0
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 840
set acc_hardware = NONE
set acc_calibration = 26,26,-91
set rssi_channel = 16
set serialrx_provider = SBUS
set blackbox_device = NONE
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT600
set bat_capacity = 520
set battery_meter = ESC
set ibata_scale = 170
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1
set frsky_unit = METRIC
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 50
set osd_cap_alarm = 470
set osd_tim1 = 769
set osd_vbat_pos = 2497
set osd_rssi_pos = 2510
set osd_tim_1_pos = 2519
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs_pos = 200
set osd_ah_sbar_pos = 200
set osd_ah_pos = 200
set osd_current_pos = 2503
set osd_mah_drawn_pos = 2514
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 441
set osd_gps_lon_pos = 33
set osd_gps_lat_pos = 49
set osd_gps_sats_pos = 417
set osd_home_dir_pos = 65
set osd_home_dist_pos = 66
set osd_compass_bar_pos = 266
set osd_altitude_pos = 86
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 33
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2441
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 391
set osd_disarmed_pos = 2154
set osd_nheading_pos = 311
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_max_fft = ON
set debug_mode = D_MIN
set vtx_band = 3
set vtx_channel = 4
set vtx_power = 2
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5645
set vcd_video_system = PAL

# profile
profile 0

set dyn_lpf_dterm_min_hz = 60
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set p_pitch = 32
set i_pitch = 60
set d_pitch = 24
set f_pitch = 0
set p_roll = 28
set i_roll = 50
set d_roll = 20
set f_roll = 0
set d_min_roll = 0
set d_min_pitch = 0

# profile
profile 1

set dyn_lpf_dterm_min_hz = 60
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set p_pitch = 32
set i_pitch = 60
set d_pitch = 24
set f_pitch = 0
set p_roll = 28
set i_roll = 50
set d_roll = 20
set f_roll = 0
set d_min_roll = 0
set d_min_pitch = 0

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_srate = 79
set pitch_srate = 79
set tpa_rate = 75
set tpa_breakpoint = 1600

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

**Setup / Versions**
 - CLRacing F4 Mini
 - 4.0.0, 4.0.2
 - R9 Mini Smartport wired to SAT pad (C07)

**Additional context**
Smartport via softserial also doesn't work at all if softserial is mapped to A09, C06 or B10 (In either 4.0.0, 4.0.2 as reported in #8026)

This bug may or may not be related to that.

SlipstreamFPV, 2019-06-02:
Anyone?

mikeller, 2019-06-03:
I can't see anything that was changed to do with either softserial or the KAKUTEF4 target between 4.0.0 and 4.0.2, so I am a bit stunned. @jflyper?

jflyper, 2019-06-03:
Lemme give a quick test with Nuc405.

jflyper, 2019-06-03:
Okay ...

If this is about `CLRACINGF4` target, the `PC7` pin under question has never been given a valid timer entry since it was introduced back in June 2017, so it should never have worked as a pin for soft serial.

The same applies for `PC6`.

However, before June 2017, when this target was called `CL_RACINGF4` which was a sub target of `OMNIBUSF4`, there was a valid timer entry for `PC6` and `PC7`, so the soft serial should have worked over these pins back then.

jflyper, 2019-06-03:
I found an interesting additional information.

The PC7 pin under 4.0.0 *MAY* work by _accident_.

This is probably due to a bug in timer selection code and was never intended.

jflyper, 2019-06-03:
@SlipstreamFPV 

In 4.0.0, you can add a timer entry for TIM8_CH2 on PC7 by the following CLI command.
After this, you should be able to use PC7 for softserial.

```
timer c7 1
```

mikeller, 2019-06-03:
@jflyper:

Good find - I did not even bother to check that this pin actually has a timer, as it was reported as working in 4.0.0 by the user.

> The PC7 pin under 4.0.0 may have worked by accident.

What on earth made this work?

jflyper, 2019-06-03:
> What on earth made this work?

I actually didn't bother to dig deeper, but here's how it looked:

```
# resource show
...
C07: SERIAL_TX 11
...

# timer
timer B09 1
timer B00 0
timer B01 2
timer A03 0
timer A02 0
timer B04 0
timer B08 0

# timer c7
timer C07 NONE
```

`openSoftSerial` never registers a pin without a valid return from `timerGetByTag`, so the later must have returned an entry for PC8 (It might have been TIM3_CH2 in which case collides with PWM beeper.)

mikeller, 2019-06-03:
Duplicate of #8026.

closed 2019-06-03 by mikeller