betaflight/betaflight #8344: Barometer on Sparky 2.0 isn't recognized

Kraligor, 2019-05-25:
**Describe the bug**
Barometer on Sparky 2.0 isn't recognized.

**To Reproduce**
Flash Sparky 2.0 FC with Betaflight. Barometer is grayed out in the top status symbols bar and in the sensor graph.

**Expected behavior**
Sparky 2.0 contains an MS5611 barometer and should be enabled by default.

**Flight controller configuration**
Default configuration, directly after flashing with full erase. Will enter diff later.

**Setup / Versions**
 - Taulabs Sparky 2.0, purchased from Hobbyking
 - RX8R Pro (LBT FW), Jumper T12

**Additional context**
The barometer works out of the box in Ardupilot.

etracer65, 2019-05-25:
As requested when you opened the issue, provide a `diff` of your settings.

Kraligor, 2019-05-25:
Done.

mikeller, 2019-05-25:
@Kraligor: This looks like it should work - I don't have this hardware, so it is a bit hard to diagnose. Can you please also provide the output of `get baro` and `get i2c` in CLI?

Kraligor, 2019-05-25:
Sure!

get baro:
```
# get baro
baro_bustype = I2C
Allowed values: NONE, I2C, SPI, SLAVE

baro_spi_device = 3
Allowed range: 0 - 5
Default value: 0

baro_i2c_device = 1
Allowed range: 0 - 5

baro_i2c_address = 0
Allowed range: 0 - 119

baro_hardware = AUTO
Allowed values: AUTO, NONE, BMP085, MS5611, BMP280, LPS, QMP6988

baro_tab_size = 21
Allowed range: 0 - 48

baro_noise_lpf = 600
Allowed range: 0 - 1000

baro_cf_vel = 985
Allowed range: 0 - 1000
```

get i2c:
```
# get i2c
mag_i2c_device = 0
Allowed range: 0 - 3

mag_i2c_address = 0
Allowed range: 0 - 119

baro_i2c_device = 1
Allowed range: 0 - 5

baro_i2c_address = 0
Allowed range: 0 - 119

dashboard_i2c_bus = 1
Allowed range: 0 - 3

dashboard_i2c_addr = 60
Allowed range: 8 - 119

gyro_1_i2cBus = 0
Allowed range: 0 - 3

gyro_1_i2c_address = 0
Allowed range: 0 - 119

gyro_2_i2cBus = 0
Allowed range: 0 - 3

gyro_2_i2c_address = 0
Allowed range: 0 - 119
```

mikeller, 2019-05-26:
:confused: 

This again looks all good. It could be that the target definition for this board was always wrong, and the barometer is on a different I2C bus. You could try testing with values from 1 to 5 for `baro_i2c_device` and see if one of them works. But that is really just clutching at straws.

jflyper, 2019-05-26:
Here's the schematic for Sparky 2.
https://github.com/TauLabs/TauLabs/blob/next/flight/targets/sparky2/hw/Sparky%20v2.0.pdf

The sole I2C is I2C1, which is where MS5611 is connected, so changing `baro_i2c_device` would not work.

The only thing still unknown is if PB8 and PB9 are actually working as I2C1_SCL and I2C1_SDA. To check this out, we need to see what CLI `resource show` say.

jflyper, 2019-05-26:
Okay, I think `target.h` for `SPARKY2` target is missing defs for `I2C1_SCL` and `I2C1_SDA`, making these to fall into defaults, which are PB6 and PB7.

@Kraligor 
You need to issue following CLI commands to assign correct pins for I2C1.
```
resource i2c_scl 1 b8
resource i2c_sda 1 b9
```

Kraligor, 2019-05-26:
Wow, thanks so much, guys. I think I've never experienced a faster and more engaged response to a bug report in any other project!

The fix works.

jflyper, 2019-05-26:
@Kraligor The fix is scheduled to be a part of 4.0.3 maintenance release, in which you don't have to do the `resource` fiddling anymore.

mikeller, 2019-05-26:
@Kraligor: No problem. What surprises me is that this has been broken for so long (probably ever since this target was added), considering that the schematics for this board are openly available.

PR #8346 (2019-05-27): SPARKY2 Assign correct pins for I2C1
Fixes #8344
closed 2019-05-27 by mikeller