betaflight/betaflight #8352: Quad is  bounce back and crash 4.0.2

trunglp, 2019-05-29:
Hi Betaflight,

I used firmware betaflight version 3.5.7 is fly normal. when i update to version 4.0.2 (i checked Full chip erase,.....) => i fly hover my quad about 1m and after that i landing it  => But i don't know my quad is bounce back, i can not control throttle and it is crazy

My quad is 2.5 inch
FC MAMBA F405 F4 Mini

Everyone can check my log for me. i said thank you,

This is my log 
https://drive.google.com/file/d/1R7yGHZvFC5pZ4KJp7ILD9dCzd02pkdTo/view?usp=sharing

[image]

```wrap

# diff

# version
# Betaflight / FURYF4OSD (FY4O) 4.0.2 May  5 2019 / 11:55:04 (56bdc8d26) MSP API: 1.41

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id 

# name
name F2.5

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux
aux 0 0 0 1375 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1700 2100 0 0
aux 4 35 3 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = -61,26,-68
set blackbox_p_ratio = 128
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set deadband = 3
set yaw_deadband = 10
set pid_process_denom = 1
set debug_mode = GYRO_SCALED

# profile
profile 0

set vbat_pid_gain = ON
set iterm_relax_cutoff = 10

# rateprofile
rateprofile 0

set tpa_rate = 80
set tpa_breakpoint = 1750

# end the command batch
batch end

```

bizmar, 2019-05-29:
https://www.facebook.com/groups/betaflightgroup/ is the official place for support, this is for bug reports. 

The default pids are not good for your quad, unfortunately the defaults can't work for every setup, and with each version of BF a different set of quads don't work well. 

There appears to be a lot very low frequency (60hz) vibration on pitch. Probably from the frame. I would start by decreasing RP P-term by 10, if that fixes it, you can bring them back as far as they can go. If not set your dynamic D-term lowpass filter to 80-180. If that doesn't help post in the facebook group for more help.

jflyper, 2019-05-29:
Something is swinging loose on the craft. A huge antenna sticking out?

watageek, 2019-05-29:
Not sure, I think he is saying :
"he is landing it and it bounce back"
I feel he is not disarming and the kwad goes cazy... wich is a normal behavior with airmode and light kwad.

trunglp, 2019-05-30:
@bizmar i said thank you, i will post this error in the group FB https://www.facebook.com/groups/betaflightgroup/ 
I try tuning PID   https://github.com/betaflight/betaflight/wiki/Community-Presets but it would still have been error
I don't know why i fly it on FW 3.5.7 this is normal.

etracer65, 2019-05-31:
Airmode is on by default on BF4.0 where it wasn't on 3.5.7. So you could always turn airmode off on the configuration tab and it'll probably help you land "better". But overall the quad will fly better with airmode on. You should probably just learn how to land like everybody else. Just hover a few inches above the ground and disarm.

trunglp, 2019-06-03:
Today i reflash to 4.0.3 and turn off airmode, everything is very well, after that i decrease low P and tuning PiD from  https://github.com/betaflight/betaflight/wiki/Community-Presets my quad is pretty
Thank you @etracer65 @bizmar @jflyper @watageek  support for me!

antigravity82, 2019-06-04:
put airmode on a switch and turn it off before landing.

closed 2019-07-11 by stale[bot]