betaflight/betaflight #8354: BF 4.1 (#1666): Pasting diff back into CLI makes all USB connection modes (DFU/serial) fail

RipperDrone, 2019-05-29:
Aikon F4 Mini, BF 4.1 #1666, BFC 10.6.0 #203. PC Win10 x64.

Creating a diff, re-pasted it after reflashing FW #1666 --> doesn't connect to FC anymore, neither in DFU mode nor serial. Have to cold restart Win10 computer and remove 'unknown USB device, no device information retrieved' before. Only then it reconnects to FC over DFU - still NOT in serial mode. Have to reflash FW again, then paste everything **but the dma/timer section** in the beginning of the diff --> reads ok. After having restored everything else, I can then paste the dma/timer lines and it's ok.

Still seems some messed up sequence of having dma/timer/pin assignments in the wrong order when pasting from diff...

RipperDrone, 2019-05-29:
Acc @fujin - referencing this to #8303 and #8019

RipperDrone, 2019-05-29:
diff:

# version
# Betaflight / AIKONF4 (AIK4) 4.1.0 May 19 2019 / 01:54:43 (b1d1fba77) MSP API: 1.42
# start the command batch
batch start
# reset configuration to default settings
defaults nosave
board_name AIKONF4
mcu_id 00290027354b500320353146
signature 
# name: Floss 3
# timer
timer C06 0
timer C08 0
timer C09 0
timer B00 NONE
timer B01 NONE
# dma
dma pin B00 NONE
dma pin B01 NONE
# feature
feature ESC_SENSOR
# beacon
beacon RX_LOST
beacon RX_SET
# map
map TAER1234
# serial
serial 0 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200
serial 31 1024 115200 57600 0 115200
# aux
aux 0 0 0 1300 1700 0 0
aux 1 2 1 1700 2100 0 0
aux 2 27 0 1700 2100 0 0
aux 3 13 0 1700 2100 0 0
aux 4 35 2 900 1300 0 0
aux 5 49 0 1300 1700 0 0
# adjrange
adjrange 0 0 3 900 1025 0 2 0 0
adjrange 1 0 3 1025 1075 6 2 0 0
adjrange 2 0 3 1075 1125 7 2 0 0
adjrange 3 0 3 1125 1175 8 2 0 0
adjrange 4 0 3 1175 1225 9 2 0 0
adjrange 5 0 3 1225 1300 10 2 0 0
adjrange 6 0 3 1300 1375 11 2 0 0
adjrange 7 0 3 1375 1425 4 2 0 0
adjrange 8 0 3 1425 1475 5 2 0 0
adjrange 9 0 3 1475 1525 1 2 0 0
# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 350
set gyro_lowpass2_hz = 0
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 300
set dyn_notch_min_hz = 100
set dyn_lpf_gyro_min_hz = 0
set acc_calibration = -18,-3,-70
set baro_hardware = NONE
set max_aux_channels = 4
set serialrx_provider = FPORT
set blackbox_p_ratio = 64
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_off_delay = 25
set failsafe_throttle = 1070
set failsafe_switch_mode = KILL
set failsafe_throttle_low_delay = 50
set failsafe_procedure = AUTO-LAND
set ibata_scale = 170
set ibata_offset = -3800
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2081
set osd_rssi_pos = 40
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2113
set osd_flymode_pos = 461
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 2102
set osd_mah_drawn_pos = 2136
set osd_craft_name_pos = 2093
set osd_gps_speed_pos = 34
set osd_gps_lon_pos = 498
set osd_gps_lat_pos = 466
set osd_gps_sats_pos = 1
set osd_home_dir_pos = 100
set osd_home_dist_pos = 1
set osd_compass_bar_pos = 266
set osd_altitude_pos = 33
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 391
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2378
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 20
set osd_disarmed_pos = 2411
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set vtx_band = 3
set vtx_channel = 2
set vtx_power = 2
set vtx_freq = 5685
set name = Floss 3
profile 0
# profile 0
set dyn_lpf_dterm_max_hz = 175
set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 12000
set p_pitch = 44
set i_pitch = 80
set d_pitch = 33
set p_roll = 37
set i_roll = 70
set d_roll = 30
set f_yaw = 30
set d_min_roll = 18
set d_min_boost_gain = 25
profile 1
profile 2
# restore original profile selection
profile 0
rateprofile 0
# rateprofile 0
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120
rateprofile 1
rateprofile 2
rateprofile 3
rateprofile 4
rateprofile 5
# restore original rateprofile selection
rateprofile 0
# save configuration
save

RipperDrone, 2019-05-29:
diff: 
`# version
# Betaflight / AIKONF4 (AIK4) 4.1.0 May 19 2019 / 01:54:43 (b1d1fba77) MSP API: 1.42
# start the command batch
batch start
# reset configuration to default settings
defaults nosave
board_name AIKONF4
mcu_id 00290027354b500320353146
signature 
# name: Floss 3
# timer
timer C06 0
timer C08 0
timer C09 0
timer B00 NONE
timer B01 NONE
# dma
dma pin B00 NONE
dma pin B01 NONE
# feature
feature ESC_SENSOR
# beacon
beacon RX_LOST
beacon RX_SET
# map
map TAER1234
# serial
serial 0 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200
serial 31 1024 115200 57600 0 115200
# aux
aux 0 0 0 1300 1700 0 0
aux 1 2 1 1700 2100 0 0
aux 2 27 0 1700 2100 0 0
aux 3 13 0 1700 2100 0 0
aux 4 35 2 900 1300 0 0
aux 5 49 0 1300 1700 0 0
# adjrange
adjrange 0 0 3 900 1025 0 2 0 0
adjrange 1 0 3 1025 1075 6 2 0 0
adjrange 2 0 3 1075 1125 7 2 0 0
adjrange 3 0 3 1125 1175 8 2 0 0
adjrange 4 0 3 1175 1225 9 2 0 0
adjrange 5 0 3 1225 1300 10 2 0 0
adjrange 6 0 3 1300 1375 11 2 0 0
adjrange 7 0 3 1375 1425 4 2 0 0
adjrange 8 0 3 1425 1475 5 2 0 0
adjrange 9 0 3 1475 1525 1 2 0 0
# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 350
set gyro_lowpass2_hz = 0
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 300
set dyn_notch_min_hz = 100
set dyn_lpf_gyro_min_hz = 0
set acc_calibration = -18,-3,-70
set baro_hardware = NONE
set max_aux_channels = 4
set serialrx_provider = FPORT
set blackbox_p_ratio = 64
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_off_delay = 25
set failsafe_throttle = 1070
set failsafe_switch_mode = KILL
set failsafe_throttle_low_delay = 50
set failsafe_procedure = AUTO-LAND
set ibata_scale = 170
set ibata_offset = -3800
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2081
set osd_rssi_pos = 40
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2113
set osd_flymode_pos = 461
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 2102
set osd_mah_drawn_pos = 2136
set osd_craft_name_pos = 2093
set osd_gps_speed_pos = 34
set osd_gps_lon_pos = 498
set osd_gps_lat_pos = 466
set osd_gps_sats_pos = 1
set osd_home_dir_pos = 100
set osd_home_dist_pos = 1
set osd_compass_bar_pos = 266
set osd_altitude_pos = 33
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 391
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2378
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 20
set osd_disarmed_pos = 2411
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set vtx_band = 3
set vtx_channel = 2
set vtx_power = 2
set vtx_freq = 5685
set name = Floss 3
profile 0
# profile 0
set dyn_lpf_dterm_max_hz = 175
set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 12000
set p_pitch = 44
set i_pitch = 80
set d_pitch = 33
set p_roll = 37
set i_roll = 70
set d_roll = 30
set f_yaw = 30
set d_min_roll = 18
set d_min_boost_gain = 25
profile 1
profile 2
# restore original profile selection
profile 0
rateprofile 0
# rateprofile 0
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120
rateprofile 1
rateprofile 2
rateprofile 3
rateprofile 4
rateprofile 5
# restore original rateprofile selection
rateprofile 0
# save configuration
save`

RipperDrone, 2019-05-29:
``# version
# Betaflight / AIKONF4 (AIK4) 4.1.0 May 19 2019 / 01:54:43 (b1d1fba77) MSP API: 1.42
# start the command batch
batch start
# reset configuration to default settings
defaults nosave
board_name AIKONF4
mcu_id 00290027354b500320353146
signature 
# name: Floss 3
# timer
timer C06 0
timer C08 0
timer C09 0
timer B00 NONE
timer B01 NONE
# dma
dma pin B00 NONE
dma pin B01 NONE
# feature
feature ESC_SENSOR
# beacon
beacon RX_LOST
beacon RX_SET
# map
map TAER1234
# serial
serial 0 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200
serial 31 1024 115200 57600 0 115200
# aux
aux 0 0 0 1300 1700 0 0
aux 1 2 1 1700 2100 0 0
aux 2 27 0 1700 2100 0 0
aux 3 13 0 1700 2100 0 0
aux 4 35 2 900 1300 0 0
aux 5 49 0 1300 1700 0 0
# adjrange
adjrange 0 0 3 900 1025 0 2 0 0
adjrange 1 0 3 1025 1075 6 2 0 0
adjrange 2 0 3 1075 1125 7 2 0 0
adjrange 3 0 3 1125 1175 8 2 0 0
adjrange 4 0 3 1175 1225 9 2 0 0
adjrange 5 0 3 1225 1300 10 2 0 0
adjrange 6 0 3 1300 1375 11 2 0 0
adjrange 7 0 3 1375 1425 4 2 0 0
adjrange 8 0 3 1425 1475 5 2 0 0
adjrange 9 0 3 1475 1525 1 2 0 0
# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 350
set gyro_lowpass2_hz = 0
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 300
set dyn_notch_min_hz = 100
set dyn_lpf_gyro_min_hz = 0
set acc_calibration = -18,-3,-70
set baro_hardware = NONE
set max_aux_channels = 4
set serialrx_provider = FPORT
set blackbox_p_ratio = 64
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_off_delay = 25
set failsafe_throttle = 1070
set failsafe_switch_mode = KILL
set failsafe_throttle_low_delay = 50
set failsafe_procedure = AUTO-LAND
set ibata_scale = 170
set ibata_offset = -3800
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2081
set osd_rssi_pos = 40
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2113
set osd_flymode_pos = 461
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 2102
set osd_mah_drawn_pos = 2136
set osd_craft_name_pos = 2093
set osd_gps_speed_pos = 34
set osd_gps_lon_pos = 498
set osd_gps_lat_pos = 466
set osd_gps_sats_pos = 1
set osd_home_dir_pos = 100
set osd_home_dist_pos = 1
set osd_compass_bar_pos = 266
set osd_altitude_pos = 33
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 391
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2378
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 20
set osd_disarmed_pos = 2411
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set vtx_band = 3
set vtx_channel = 2
set vtx_power = 2
set vtx_freq = 5685
set name = Floss 3
profile 0
# profile 0
set dyn_lpf_dterm_max_hz = 175
set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 12000
set p_pitch = 44
set i_pitch = 80
set d_pitch = 33
set p_roll = 37
set i_roll = 70
set d_roll = 30
set f_yaw = 30
set d_min_roll = 18
set d_min_boost_gain = 25
profile 1
profile 2
# restore original profile selection
profile 0
rateprofile 0
# rateprofile 0
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120
rateprofile 1
rateprofile 2
rateprofile 3
rateprofile 4
rateprofile 5
# restore original rateprofile selection
rateprofile 0
# save configuration
save''

RipperDrone, 2019-05-29:
https://betaflightgroup.slack.com/files/U3GQQRZAA/FJSAH1SJE/aikon_f4_killer_diff.txt

mikeller, 2019-05-29:
@RipperDrone: Can you please confirm that #8303 fixes this?

RipperDrone, 2019-05-29:
Wasn't #8303 already merged into master, so I actually tested it with Nightly #1666? Result = it does NOT fix it...

mikeller, 2019-05-29:
@RipperDrone: The `diff` you reference is from before #8303. Can you please supply a `diff` from when you actually verified that #8303 did not fix this? Supplying information that does not pertain to the issue is a waste of time.

RipperDrone, 2019-05-30:
argh - u - r - so - right! :-) I picked the wrong file. Repeated the test procedure, working good now!

Here's the correct diff for reference:

``# diff all

# version
# Betaflight / AIKONF4 (AIK4) 4.1.0 May 28 2019 / 10:09:48 (2f3288502) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name AIKONF4
mcu_id 00290027354b500320353146
signature 

# name: Floss 3

# timer
timer C06 AF2
# pin C06: TIM3 CH1 (AF2)
timer C08 AF2
# pin C08: TIM3 CH3 (AF2)
timer C09 AF2
# pin C09: TIM3 CH4 (AF2)
timer B00 NONE
timer B01 NONE

# dma
dma pin C06 0
# pin C06: DMA1 Stream 4 Channel 5
dma pin C08 0
# pin C08: DMA1 Stream 7 Channel 5
dma pin C09 0
# pin C09: DMA1 Stream 2 Channel 5
dma pin B00 NONE
dma pin B01 NONE

# feature
feature ESC_SENSOR

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200
serial 31 1024 115200 57600 0 115200

# aux
aux 0 0 0 1300 1700 0 0
aux 1 2 1 1700 2100 0 0
aux 2 27 0 1700 2100 0 0
aux 3 13 0 1700 2100 0 0
aux 4 35 2 900 1300 0 0
aux 5 49 0 1300 1700 0 0

# adjrange
adjrange 0 0 3 900 1025 0 2 0 0
adjrange 1 0 3 1025 1075 6 2 0 0
adjrange 2 0 3 1075 1125 7 2 0 0
adjrange 3 0 3 1125 1175 8 2 0 0
adjrange 4 0 3 1175 1225 9 2 0 0
adjrange 5 0 3 1225 1300 10 2 0 0
adjrange 6 0 3 1300 1375 11 2 0 0
adjrange 7 0 3 1375 1425 4 2 0 0
adjrange 8 0 3 1425 1475 5 2 0 0
adjrange 9 0 3 1475 1525 1 2 0 0

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 350
set gyro_lowpass2_hz = 0
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 300
set dyn_notch_min_hz = 100
set dyn_lpf_gyro_min_hz = 0
set acc_calibration = -28,6,-60
set baro_hardware = NONE
set max_aux_channels = 4
set serialrx_provider = FPORT
set blackbox_p_ratio = 64
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_off_delay = 25
set failsafe_throttle = 1070
set failsafe_switch_mode = KILL
set failsafe_throttle_low_delay = 50
set failsafe_procedure = AUTO-LAND
set ibata_scale = 170
set ibata_offset = -3800
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2081
set osd_rssi_pos = 40
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2113
set osd_flymode_pos = 461
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 2102
set osd_mah_drawn_pos = 2136
set osd_craft_name_pos = 2093
set osd_gps_speed_pos = 34
set osd_gps_lon_pos = 498
set osd_gps_lat_pos = 466
set osd_gps_sats_pos = 1
set osd_home_dir_pos = 100
set osd_home_dist_pos = 1
set osd_compass_bar_pos = 266
set osd_altitude_pos = 33
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 391
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2378
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 20
set osd_disarmed_pos = 2411
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set vtx_band = 3
set vtx_channel = 2
set vtx_power = 2
set vtx_freq = 0
set name = Floss 3

profile 0

# profile 0
set profile_name = PID 1
set dyn_lpf_dterm_max_hz = 175
set dterm_lowpass_hz = 160
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 12000
set p_pitch = 38
set d_pitch = 27
set p_roll = 35
set d_roll = 25
set f_yaw = 30

profile 1

# profile 1
set profile_name = PID 2

profile 2

# profile 2
set profile_name = PID 3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set rateprofile_name = RATE 1
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120

rateprofile 1

# rateprofile 1
set rateprofile_name = RATE 2

rateprofile 2

# rateprofile 2
set rateprofile_name = RATE 3

rateprofile 3

# rateprofile 3
set rateprofile_name = RATE 4

rateprofile 4

# rateprofile 4
set rateprofile_name = RATE 5

rateprofile 5

# rateprofile 5
set rateprofile_name = RATE 6

# restore original rateprofile selection
rateprofile 0

# save configuration
save

closed 2019-05-30 by RipperDrone