betaflight/betaflight #8355: gps_rescue_min_sats

oliverherdener, 2019-05-30:
**Use case:**
If flying indoors or testing on the bench or waiting for a GPS fix, I want to fly the quad even if failsafe is set to "GPS Rescue" (while being aware of the consequences)
Before BF 4.x, this worked
`set gps_rescue_min_sats = 0`

**Bug**
In BF Configurator (10.5.1), it's possible to set `Minumum satellites` to "0".

Diff in CLI:
`set gps_rescue_min_sats = 0`
`###ERROR: CORRUPTED CONFIG: gps_rescue_min_sats = 0###`

**Behaviour**
- The quad won't arm, beep code 18
- Set the failsafe to "Drop": The quad still won't arm, beep code 18
- Disable GPS in "Configuration": Quad arms

**Expected behavior**
As before BF4.x, setting to zero will not prevent arming

**Flight controller configuration**
```# version
# Betaflight / OMNIBUSF4SD (OBSD) 4.0.2 May  5 2019 / 12:08:37 (56bdc8d26) MSP API: 1.41

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id 

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST

# map
map AERT1234

# serial
serial 2 2 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 900 1300 0 0
aux 2 2 2 1300 1700 0 0
aux 3 13 1 1700 2100 0 0
aux 4 35 4 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set acc_calibration = 4,-23,-49
set rssi_channel = 12
set fpv_mix_degrees = 30
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT300
set failsafe_switch_mode = STAGE2
set failsafe_procedure = GPS-RESCUE
set align_board_yaw = 180
set vbat_min_cell_voltage = 290
set vbat_warning_cell_voltage = 310
set gps_provider = UBLOX
set gps_rescue_sanity_checks = RESCUE_SANITY_OFF
set gps_rescue_min_sats = 0
###ERROR: CORRUPTED CONFIG: gps_rescue_min_sats = 0###
set osd_alt_alarm = 9999
set osd_vbat_pos = 2048
set osd_rssi_pos = 2490
set osd_tim_1_pos = 2465
set osd_tim_2_pos = 2433
set osd_flymode_pos = 2457
set osd_craft_name_pos = 2478
set osd_gps_speed_pos = 2073
set osd_gps_lon_pos = 2497
set osd_gps_lat_pos = 2513
set osd_gps_sats_pos = 2105
set osd_home_dir_pos = 2069
set osd_home_dist_pos = 2066
set osd_altitude_pos = 2422
set osd_avg_cell_voltage_pos = 2054

# profile
profile 0

set dterm_lowpass2_hz = 200
set feedforward_transition = 30
set iterm_relax_type = GYRO
set i_pitch = 85
set i_roll = 80
set d_roll = 30
set f_yaw = 35
set d_min_roll = 25
set d_min_pitch = 28
set d_min_boost_gain = 30
set d_min_advance = 50

# rateprofile
rateprofile 0

set roll_rc_rate = 106
set pitch_rc_rate = 106
set yaw_rc_rate = 106
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 74
set pitch_srate = 74
set yaw_srate = 74

# end the command batch
batch end
```

**Setup / Versions**
 - Flight Controllers: OMNIBUSF4 && OMNIBUSF4SD
- GPS: BN 180

McGiverGim, 2019-05-30:
From the wiki:

## Arm without a GPS Fix
By default, Betaflight will not let arm without a GPS Fix if you have GPS Rescue configured in a switch or as a failsafe procedure. Sometimes you might want to fly without a GPS Fix (maybe you're in a zone with poor coverage, or want to do a quick reconnaissance flight meanwhile sats are acquired) at the expense of GPS Rescue being deactivated. You can achieve this by setting:

`set gps_rescue_allow_arming_without_fix = ON`

With this value, you can take off without a GPS Fix, but the GPS Rescue will not be available during the flight. A warning (RESCUE OFF) will appear in the OSD to make you aware of that. If a proper number of satellites are acquired while flying, to enable GPS Rescue you must land, disarm and arm again.

oliverherdener, 2019-05-30:
Wow thanks, I wasn't aware of that.

So this bug report then comes down to the issue that the Configurator allows users to set `gps_rescue_min_sats` to less than 5, while this seems to be an invalid value in BF.

McGiverGim, 2019-05-30:
I suppose that after saving, if refreshed the tab, it will appear as 5. But yes, it will be better to fix it for the next version. Can you open an issue in the Configurator Github for that? Thanks!