betaflight/betaflight #8360: esc pass-through and general esc problems with latest 4.1

robert-hoffmann, 2019-05-31:
**Describe the bug**
Not sure if this is just because of refactorings in the code, but last night i flashed build 1666 on my Matek F405, with tekko 32/F3 (furling esc's), and it seems like there are initialization problems on the esc side (Furling32_Multi_32_62.Hex and Furling32_Multi_32_61.Hex)

The esc's will initialize randomly, and multiple times. Also BLHELI will fail to systematically read the esc's via the pass-through, and when it works will display input frame errors in the results.

Going back to build 1569, fixes the issues.

I haven't tried any other builds to try to narrow down the problem, because i was actually only interested in getting #7907 into my current config

**Flight controller configuration**
```
# Betaflight / MATEKF405 (MKF4) 4.1.0 Apr 14 2019 / 14:06:05 (eb7e6c0b3) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF405
manufacturer_id 
mcu_id 001b00323436510639333434
signature 

# name
name Chameleon

# resources
resource MOTOR 5 NONE
resource SERIAL_TX 11 A15

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 4 2048 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1950 2075 0 0
aux 1 13 0 1675 1775 0 0
aux 2 19 2 1950 2075 0 0
aux 3 35 0 1900 2000 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 0
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 170
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 0
set acc_hardware = NONE
set acc_calibration = 18,30,-23
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_src_frame_errors = ON
set airmode_start_throttle_percent = 20
set blackbox_device = NONE
set dshot_idle_value = 400
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 10
set align_board_yaw = 180
set ibata_scale = 167
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set pid_process_denom = 1
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_pos = 2137
set osd_tim_2_pos = 2081
set osd_throttle_pos = 2487
set osd_current_pos = 2496
set osd_mah_drawn_pos = 2520
set osd_warnings_pos = 2505
set osd_avg_cell_voltage_pos = 2103
set osd_stat_max_spd = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 2
set vtx_freq = 5880
set vcd_video_system = PAL

# profile
profile 0

set dyn_lpf_dterm_max_hz = 300
set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 4000
set feedforward_transition = 40
set iterm_relax = RPY
set iterm_relax_cutoff = 5
set f_pitch = 70
set abs_control_gain = 5
set d_min_boost_gain = 30
set d_min_advance = 60

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set thr_mid = 27
set thr_expo = 60
set roll_rc_rate = 90
set pitch_rc_rate = 90
set yaw_rc_rate = 90
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 15
set roll_srate = 85
set pitch_srate = 85
set tpa_rate = 70
set tpa_breakpoint = 1750

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

eumelOne, 2019-05-31:
I think you have to flash 32.63 as Dshot has been inverted in meanwhile.

robert-hoffmann, 2019-05-31:
Just did a quick reflash of 4.1 with 32.63
No more weird esc errors, and a quick hover seemed ok.

Will give this a dozen packs this afternoon and see how things go ;-)

Thanks !

closed 2019-05-31 by robert-hoffmann

robert-hoffmann, 2019-05-31:
Works like a charm, even better than build 1569
...however seems like some stuff on vtx table has changed, and lua script cannot update the channel anymore via my xlite ^^