betaflight/betaflight #8361: BF#1666, MatekSE722, AikonF4, ...: CLI diff creates UAV name scattered over the diff file

RipperDrone, 2019-05-31:
Quad name occurs in 2 separate places in the 'diff all' dump:

1. top section: # name: <quad name>
2. Much further down, right after the vtx settings: set name = <QUAD NAME>

I feel that set name = ... should go right under the # name section, shouldn't it?

etracer65, 2019-05-31:
The name at the top is just a comment. People like seeing the make at the top to help identify which quad the config is for. The `set name = ` is the actual command to set the name. It's working as designed.

RipperDrone, 2019-05-31:
@etracer65 Agreed it is working, anyway why not keeping information together which belongs to the same topic? Most other information is grouped / clustered in diff, e.g.: 

- #resources group comment is followed by resource settings grouped
- #timer comment followed by timer settings grouped
- #feature group followed by feature settings grouped
- etc.

Why is #name comment NOT followed by name settings, but separated in sequence?

etracer65, 2019-05-31:
Is this really worth worrying about? There is no longer a custom "name" section/command. The parameter has been converted to a standard "set" type of parameter and is part of the master config so it's in that section. Technically the `name` command is still there for one more release but it warns that it's being removed. As I said before, the name is printed at the top as a comment (like the version, etc.) for reference to make it easier for people to keep track of their configs - that's all.

closed 2019-05-31 by RipperDrone