betaflight/betaflight #8362: OSD

rFlightFPV, 2019-05-31:
I have a few suggestions regarding the OSD:

1. could we please drop the decimals on the amp draw? Even with my low amp draw LR quad there is absolutely no point in showing the decimals. It‘s just nervously changing numbers without any information and therefore distracting.

2. Same applies to GPS coordinates, 5 decimals is enough by far, anything more is simply academic and not of any practical use.

3. Item reduction on a knob. An OSD from the „other world“ allows to add/reduce items in-flight with a knob on the radio (aux channel). Could we please have this too. User can also define order of items.

4. make logos of the items selectable (on/off)? I do not need the sat dish, RSSI logo, throttle % logo, etc.

etracer65, 2019-05-31:
Item 3 can already be done. For the rest, can't make everybody happy. More people NOT complaining about these items then are...

rFlightFPV, 2019-05-31:
Thanks. I can live with the logos although just unnecessary clutter imho, but the at least the second decimal on the amp draw is nonsense nobody needs, the measurement itself already is in error before the decimal point. Same for GPS: 5 decimals is resolution of approx. 1 meter. Having more makes no sense, to nobody.

etracer65, 2019-05-31:
In your opinion - that's correct. Unfortunately not everybody shares your opinion. But if it bothers you that much feel free to make changes to the display if you want.

https://github.com/betaflight/betaflight/blob/master/src/main/osd/osd_elements.c

RipperDrone, 2019-05-31:
@rFlightFPV 

- line 235: change format %d.%07d to %d.05d to make GPS coo display 5 decimals instead of 7
- line 625: change format "3d.%02d%c", abs(amperage) / 100, abs(amperage) % 100, SYM_AMP) to:
"3d.%01d%c", abs(amperage) / 10, abs(amperage) % 10, SYM_AMP)

[image]

RipperDrone, 2019-05-31:
...and redefine constant GPS_DEGREE_DIVIDER by factor of 100 smaller...

This should be it.

RipperDrone, 2019-05-31:
found it: 
[image]

Chage to: #define GPS_DEGREES_DIVIDER 100000L

Docteh, 2019-05-31:
If you were after removing just the icons my suggestion would be to make a custom font, where you blank out the icons you don't like.

RipperDrone, 2019-06-01:
@rFlightFPV - soo many hints now to make your wish come true :-) Now it's up to you - jump that line and make a pull request!

etracer65, 2019-06-01:
If you want to make a custom version for yourself feel free. But a PR that just changes the settings to match your preferences wouldn't be accepted.

RipperDrone, 2019-06-01:
Pity - thought we almost got a new contributor activated, driven by his urge to change the current code / representation. Anyway, I personally couldn't care less about GPS decimals and such, but @etracer65: Who defines which representation for OSD variables is preferrable for most BF users? Shouldn't there be a poll or something before you just digitally reject changing OSD number display formats and/or similar requests? Not fully motivaing for someone who wants to start contributing... <shrug>

etracer65, 2019-06-01:
Better to know now rather than after the work. The correct path would be to implement a flexible solution that works for people with different preferences - while ensuring that the solution is not excessive for the benefit.

rFlightFPV, 2019-06-01:
Thanks guys for your valuable input. I‘ll dive into this and see if I am capable of coding it. I did code some projects on an STM32 and various other things, so I‘m not a complete noob to coding. It‘s just don‘t have a lot of time to do such things.
@etracer65: would CLI commands be sufficient or do you require it being configurable in the OSD tab of the configurator as well?

etracer65, 2019-06-01:
@rFlightFPV Settings in the CLI is sufficient for an initial implementation. Configurator support wouldn't be allowed until later (we defer one firmware version before adding to the Configurator).

rFlightFPV, 2019-06-01:
@etracer65: thanks. I‘ll take a look at it.

closed 2019-07-08 by stale[bot]