betaflight/betaflight #8363: SPM4648 / F722-WING timing issues

dlasher, 2019-05-31:
**Describe the bug**
There appear to be timing issues between the Spektrum SPM4648 in SPECTRUM1048 mode, and the F722-WING, under all tested versions of firmware. Receiver and FC lose sync multiple times a second, (see screenshot). Does NOT happen under iNAV firmware with same hardware.

**To Reproduce**
F722-WING, with 4.0.2 firmware. SPM4648 on an RX/TX channel. Serial Receiver, set for SPECTRUM1024, paired with DX6i. Go to receiver, and watch channels (throttle most often) drop to min and back up (failsafe) - Setup screen will show "RX_FAILSAFE" several times a second

**Expected behavior**
Solid sync. normal (?!) behavior?

**Flight controller configuration**
```# diff show

# version
# Betaflight / MATEKF722SE (MF7S) 4.0.2 May  5 2019 / 12:04:09 (56bdc8d26) MSP API: 1.41

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id 

# name
name WING1

# resources

# timer

# dma

# mixer
mixer FLYING_WING

# servo

# servo mix

# feature
feature SOFTSERIAL
feature GPS
feature CHANNEL_FORWARDING

# beeper

# beacon

# map
map TAER1234

# serial
serial 3 2 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set mid_rc = 1499
set min_check = 1051
set rssi_channel = 12
set rc_interp_int = 50
set rc_smoothing_type = INTERPOLATION
set serialrx_provider = SPEK1024
set rx_min_usec = 750
set rx_max_usec = 2250
set motor_pwm_protocol = OFF
set failsafe_delay = 60
set failsafe_throttle_low_delay = 300
set failsafe_procedure = GPS-RESCUE
set align_board_yaw = 90
set gps_auto_baud = ON
set pid_process_denom = 16
set gyro_1_sensor_align = DEFAULT

# profile
profile 0

set throttle_boost = 0
set f_pitch = 0
set f_roll = 0

# rateprofile
rateprofile 0

# end the command batch
batch end
```

**Setup / Versions**
Matex F722-WING - amazon
Spectrum SPM4648 receiver - tx2
CIROCOMM GPS - tx4
Flycolor 30A ESC - esc1

**Additional context**
facebook group discussion about issue, with pictures : https://www.facebook.com/groups/betaflightgroup/permalink/324703988225484/?comment_tracking=%7B%22tn%22%3A%22O%22%7D

dlasher, 2019-05-31:
Screenshots / images here : https://imgur.com/a/AhK9kW3

etracer65, 2019-05-31:
Use the `SPEKTRUM2048/SRXL` provider.

dlasher, 2019-06-01:
Doesn't work. Stays in RX_FAILSAFE, and never moves channels. The DX6i, as far as I know, only speaks SPEKTRUM1024, not 2048

dlasher, 2019-06-01:
I think this : https://github.com/betaflight/betaflight/wiki/Spektrum-and-RC-Smoothing-Filter

might be the problem, but the suggestion there didn't work.

```# rc_smoothing_filter
###ERROR: UNKNOWN COMMAND, TRY 'HELP'###
# rc_smoothing_info
# RC Smoothing Type: FILTER
# Input filter type: PT1
# Active input cutoff: 0hz (manual)
# Derivative filter type: PT1
# Active derivative cutoff: 0hz (manual)

# set rc_smoothing_input_hz = 21
rc_smoothing_input_hz set to 21
# set rc_smoothing_derivative_hz = 21
rc_smoothing_derivative_hz set to 21
# rc_smoothing_info
# RC Smoothing Type: FILTER
# Input filter type: PT1
# Active input cutoff: 0hz (manual)
# Derivative filter type: PT1
# Active derivative cutoff: 0hz (manual)

# ```

dlasher, 2019-06-01:
https://dlasher.tinytake.com/tt/MzU2MDg2N18xMDczMjE0OA

Timeline:
Beginning - Idle, you can see the AUX channels "flapping"
00:08 - I move all sticks to max, they all "drop"
00:28 - all sticks/aux to max - nothing drops - weird
00:41 - hands off sticks

AndersHoglund, 2019-06-01:
Have you tried re-binding using `spektrum_sat_bind = 9` and `serialrx_provider = SPEK2048` ? Some later Dx6i should be capable of that. Also, SRXL is NOT supported by SPM4648, only by SPM4649T. 
Have you really connected it to UART2 TX pin? Why? Pointless since it can not do bidirectional halfduplex SRXL. Try using Rx pin and SPEKTRUM2048

dlasher, 2019-06-02:
> Have you tried re-binding using `spektrum_sat_bind = 9` and `serialrx_provider = SPEK2048` ? Some later Dx6i should be capable of that. Also, SRXL is NOT supported by SPM4648, only by SPM4649T.
> Have [...]

Sorry, clarified the notes above. The SPM4648 is on UART2, but RX only. Tried it with TX early on, saw that only the 4649T supported it, dropped to RX only.

Tried what you suggested, with two different dx6i's one very old, one newer, with a recent trip to Horizon for gimbals and firmware.

SPEKTRUM2048 - same behavior - both radios
SPEKTRUM1024 - same behavior - both radios

also tried under newly-released-today 4.0.3 - same behavior - both radios.

https://imgur.com/a/5v1DUjh

watageek, 2019-06-06:
Do you have the possibillity to power the RX with a known good power source and not rely on the 3.3v reg of the board.

Do you have access to another RX (exactly the same)

Just suggestions to rule out possible hardware fault.

watageek, 2019-06-06:
I power mine on a spracingf3 with an inline resistor on the 5v rail.

It was 30mA @3.3v maybe around 120ohms on top of my head

dlasher, 2019-06-06:
> Do you have the possibillity to power the RX with a known good power source and not rely on the 3.3v reg of the board.
> Do you have access to another RX (exactly the same)
> Just suggestions to rule out possible hardware fault.

Thanks, will give that a shot. I only have a single RX at this time, but I can power it off a different power source. Will try that tonight. (Wait, I have a Spektrum Sat around here that came with my dx6/rcv bundle - I might try that too)

For QA, I borrowed a friends DX7 (not S) which also only speaks DSM2/1024, and it exhibits the same behavior after binding.

watageek, 2019-06-08:
It is a 50ohms.

Wich make sense... 5v - 3.3v =1.7v
1.7v/0.03A=56.666ohms

etracer65, 2019-06-08:
Using a resistive dropper to supply 3.3v from 5v is a bad idea. The output voltage depends greatly on the current draw and slight variations (like the LED  turning on/off) can vary the voltage output significantly. You'd be better off running 3 stand silicon diodes in series for ~1.8v drop (0.6v each). This won't vary with current drawn by the receiver.

watageek, 2019-06-08:
Thank you. It's a clever idea that I never thought.

dlasher, 2019-07-08:
Seems to be another wrinkle of : https://github.com/betaflight/betaflight/wiki/Spektrum-and-RC-Smoothing-Filter -- and still needing to be fixed. Guess we just wait.

etracer65, 2019-07-08:
No, not related. Smoothing doesn't apply to Aux channels. And if you read the information you linked to then you'd understand that the only thing that needs to be "fixed" is the receiver firmware provided by Spektrum.

And yes, you can just wait. Betaflight is an open source project where people volunteer their time and generally work on things that they're interested in, and are related to the hardware they have. Since you are the only person seemingly having this issue, how about you contribute some development to resolve it. Otherwise your sense of entitlement that somebody should fix this for you is not likely to engage anybody else to help you out.

dlasher, 2019-07-09:
Bruce, thank you for responding so quickly, albeit with an existing chip firmly in place. I apologize if the concise summarization of the issue conveyed impatience or entitlement, it was not intended to. 

If you've read the entire thread, you understand two things:

1. It's happening on all channels

2. The firmware provided by Spectrum works with iNAV but fails to work with Betaflight, in any version.

Given that narrow scope, this doesn't seem like the kind of issue that Spectrum is likely to issue new firmware to resolve, but rather an issue that a reporter (me) would work with someone else in the community to help diagnose, troubleshoot, and resolve. Hence the posts above.

Wish your response had provided any of those.

etracer65, 2019-07-09:
For the last time, your problem is not related to smoothing or the wiki page you’ve linked to. As I said, smoothing doesn’t affect aux channels. You have reported that the problem is affecting ALL channels - which includes aux channels.

AndersHoglund, 2019-07-09:
I can not reproduce this problem with my SPM4648. Runs great. But I have different FC (SPRF3) and TXs (DX18/iX12) etc, not much in common with your setup. You need to play around with BF versions, settings and HW combinations to determine exactly what causes this behaviour.

dlasher, 2019-08-08:
Thanks @AndersHoglund for trying. I suspect that if you had access to a dx6i/dx7/DSM1024 it would show up.

In the mean time I'm using iNAV with my wings, because it's cheaper than buying a new transmitters. My quads not using the SPM4648/DSM1024 work fine with Betaflight.

closed 2019-09-14 by stale[bot]