betaflight/betaflight #8370: Iflight Succex TwinG F4 firmware

JustinBluntFPV, 2019-06-02:
When will firmware for the Iflight Succex Twin G F4 target be available? I see the F7(EXF722DUAL) version firmware up, but not the F4

mikeller, 2019-06-02:
@JustinBluntFPV: That's a question you'll have to ask your manufacturer / supplier, and not in here - unless they submit a target / arrange with somebody to do this for them it will not happen.

Docteh, 2019-06-03:
This listing: https://www.iflight-rc.com/index.php?route=product/product&product_id=809
Seems to suggest using the STM32F405 target, 

With this file: 
https://github.com/betaflight/betaflight/blob/master/unified_targets/configs/IFLIGHTF4_TWIN_G.config

If you're planning on using 4.0.x you might want the older version of the file, maybe: https://github.com/betaflight/betaflight/blob/f766f54b3584f458c2b70fb5da54219d08175eaf/unified_targets/configs/IFLIGHTF4_TWIN_G.config

Not sure if there is a conventional target.

mikeller, 2019-06-03:
@Docteh: Thanks for digging that up. Good to see that they are embracing Unified Targets - that's exactly what we are aiming for. From the discussion we had when this was added, this board seems to be an exact clone of the EXF722DUAL, so using this target might work as well (YMMV).

closed 2019-07-10 by stale[bot]