betaflight/betaflight #8397: SmartAudio doesnt work with Atlatl HV v2.2 BF4.0.3

JahVinz, 2019-06-08:
**Describe the bug**
VTX SA shows only stars ***** so impossible to use it.
tried to hook a TBS unify 5G8 on same pad : SA works flawlessly
tried 
- disabling softserial
- soldering VTX SA  to another dedicated PIN (TX2)
- flashed to BF 4.0.2
no luck.

**To Reproduce**
hook an Atlatl HV v2.2 to Aikon F4 FC with betaflight 4.0.2 or 4.03

**Expected behavior**
SA should work :)

**Flight controller configuration**
``` 
# diff

# version
# Betaflight / AIKONF4 (AIK4) 4.0.2 May  5 2019 / 11:34:11 (56bdc8d26) MSP API: 1.41

# start the command batch
batch start

board_name AIKONF4
manufacturer_id 

# name
name NEWB

# resources
resource MOTOR 1 C08
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C07
resource MOTOR 5 NONE
resource MOTOR 6 NONE

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST

# map
map TAER1234

# serial
serial 30 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1400 2100 0 0
aux 1 27 1 1700 2100 0 0
aux 2 13 1 1300 1700 0 0
aux 3 35 2 1300 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set acc_hardware = NONE
set acc_calibration = -641,-219,-152
set baro_hardware = NONE
set min_check = 1015
set max_check = 2000
set rssi_channel = 8
set blackbox_device = NONE
set dshot_idle_value = 490
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 1
set failsafe_throttle_low_delay = 1
set vbat_max_cell_voltage = 440
set beeper_dshot_beacon_tone = 5
set pid_process_denom = 1
set transient_throttle_limit = 10
set osd_rssi_pos = 14381
set osd_tim_2_pos = 14775
set osd_vtx_channel_pos = 14807
set osd_craft_name_pos = 6604
set osd_warnings_pos = 14761
set osd_avg_cell_voltage_pos = 14754
set vtx_band = 5
set vtx_channel = 3
set vtx_freq = 5732

# profile
profile 0

set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 150
set iterm_relax_cutoff = 35
set p_pitch = 30
set i_pitch = 90
set d_pitch = 27
set f_pitch = 90
set p_roll = 28
set i_roll = 84
set d_roll = 25
set f_roll = 84
set p_yaw = 30
set i_yaw = 90
set f_yaw = 30
set d_min_roll = 16
set d_min_pitch = 18
set d_min_advance = 0

# rateprofile
rateprofile 0

set roll_expo = 40
set pitch_expo = 40
set yaw_expo = 40
set tpa_rate = 75
set tpa_breakpoint = 1400

# end the command batch
batch end

# 
``` 

**Setup / Versions**
 - Flight controller [AIKON F4, betaflight 4.0.3 ];
 - Other components [Holybro ATLATL HV V2];
 - Atlatl’s	blue wire	to	the	TX1	pad
SOFTSERIAL enabled.

**Additional context**
doesnt seems to be related to a shared ressource issue 
``` 
# resource
resource BEEPER 1 B05
resource MOTOR 1 C08
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C07
resource PPM 1 B06
resource LED_STRIP 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 11 A09
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 12 A01
resource INVERTER 1 C00
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 B04
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B06
resource CAMERA_CONTROL 1 B03
resource ADC_BATT 1 C02
resource ADC_RSSI 1 C03
resource ADC_CURR 1 C01
resource FLASH_CS 1 B12
resource OSD_CS 1 A15
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource USB_DETECT 1 D02
``` 

Not an expert but found similar topics discussed by @fujin in #7603 who got the same FC and #7781

fujin, 2019-06-09:
@JahVinz try using the AKK RDQ Patch build from Betaflight Configurator, it may work. #7603 was specifically for SA2.0 Lite (TBS Unify Nano *not PRO32*) over SoftSerial. It broke everything else, though, so we nuked it (during RC phase).

I keep the patch offline for those 3 rigs and since moved away from SA2.0Lite over SoftSerial. You may need a combination of both the AKK/RDQ patch hack, *and* the SA2.0LiteOverSoftSerial garbage.

Aside from that, any compatibility with Holybro Atlatl line is purely coincidental/unintended. They do not interact with Betaflight or license SmartAudio protocol from TBS, and (as such) not part of our testing/support matrix.

If someone had close contact with Holybro they could suggest either supply of engineering samples (to be part of testing matrix - wouldn't imply support, but we'd at leats know) or licensing SA from TBS with associated changes, if any, to guarantee support. They are otherwise not a product I have used personally.

Hope this helps, and I'm sorry thing aren't working right for ya.

JahVinz, 2019-06-10:
thanks man :) i tried the AKK RDQ patch : still nothing on VTX SA.
what is "SA2.0LiteOverSoftSerial garbage" ? ;)

i just emailed holybro customer service and mentioned this thread. *cross fingers* i hope they will collaborate more closely with you guys.

RipperDrone, 2019-06-10:
@JahVinz Have u tried the (less preferrable) hardware mod? Solder a 10k resistor on VTX from SA line to GND and see if it helps. (hardware pulldown)

I think "SA2.0LiteOverSoftSerial garbage" was the attempt in the pre-RC 4.0 codebase to not only make SA lite work over hardware UART but also softserial. However, introducing forced 0x00 soft pulldown bytes in the data stream ended up being misinterpreted by other SA versions (1.0, 2.0) so the team decided to revert all soft pulldown attempts.

JahVinz, 2019-06-10:
@RipperDrone nope, never done that kind of hack but i  would like to try it :) could you point me to an url/video to explain more in details ?

thanks for the explanation. i tried also 4.0.0 rc4 and  @laz2wiringpi version he made to give a try (4.0.0_RC2 BUILD 1535 DNU fixed in 1542)  + AKK RDQ patch, still no luck.

going down to BF 3.5 or changing my VTX could also be an easy option as a TBS unify 5G8 on same pad worked flawlessly ;)

RipperDrone, 2019-06-10:
@JahVinz https://github.com/betaflight/betaflight/issues/5329 has the resistor mod photo in the thread :-)

I assume you have set vtx_halfduplex = ON set  in CLI...

JahVinz, 2019-06-10:
# get vtx_halfduplex
vtx_halfduplex = ON
Allowed values: OFF, ON

thanks man, will have a look :)

JahVinz, 2019-06-10:
tried Betaflight 3.5.7 same issue.
that leads me to try an hardware mod or switch from Atlatl HV to another VTX ;)

jamming, 2019-06-11:
Atlatl HV v2.2 don't support SmartAudio lite.  SmartAudio lite send extra byte 0x00 after a frame. Atlatl VTX is starting to send a response immediately after receiving the command frame, the response will be on the line while Betaflight is still busy sending the 0x00, thus causing the response to be corrupted. 

I try bf4.0.2 stable, it can work.

JahVinz, 2019-06-11:
@jamming on which FC ?

JahVinz, 2019-06-11:
@RipperDrone just tried the hackmod you suggested with a 10k ohm resistance. 
same issue : no vtx is being detected in OSD.
VTX SA just shows -------- :(

jamming, 2019-06-12:
I test it on both KakuteF4V2 and KakuteF7 , with BF4.0.2 stable firmware, flashed via Betaflight Configurator 10.5.1

fujin, 2019-06-12:
@jamming I believe this problem is specific to SmartAudio over SoftSerial -- the AKF4 only has spare SoftSerial 1 TX line for SmartAudio.

Both of those Kakute boards use hardware UARTs instead (so should be, and are reported to be, working).

JahVinz, 2019-06-14:
i disabled softserial and hooked SA cable to TX2 (tried that in the past it was not working) + i added hardware mod with a 10k ohm resistance...still no luck. VTX SA is just showing --------

running out of ideas now except getting another VTX.

JahVinz, 2019-06-16:
good news !!!! i was able to work around the issue and make it works :)
scrap my previous message, resistance was not properly soldered. :-D

so work around is : 
- disable SOFTSERIAL
- use TX2 pad for blue wiring (SmartAudio) of ATLATL HV
- solder a 10k ohm resistance between TX2 and a FC ground pad (used ESC unused ground pad in my case))
- set TBS smartaudio to UART2
- profit ;=)

i tried it on BF 4.0.3 and smartaudio works like a charm :)
thanks to @fujin for explanations and @RipperDrone for suggesting the hardware hack mod ;)

closed 2019-07-10 by JahVinz