betaflight/betaflight #8398: Possible bug in the GPS rescue mode about the arrow direction

christianv2m, 2019-06-08:
**Describe the bug**
Based on the official documentation of Betaflight rescue mode in this article: https://github.com/betaflight/betaflight/wiki/GPS-rescue-mode it says "VERY IMPORTANT: if your arrow does not point towards home, DO NOT activate GPS Rescue.". With this I assume that in all cases of a good performance, arrow always points back home.

This case that I explain, arrow does not point home (it does at the beginning but not after rescue mode is activated), drone does not fly back home in rescue mode, but the distance, when i come manually back home, it works properly. So it looks like there is no way to fly back home in rescue but that it has properly set the home point.

**To Reproduce**
I attach the video of the behaviour in the next URL: https://youtu.be/VufTKxEJjhI
My procedure:

1. Connect drone.
2. Wait for more than 8 satellites, which is super fast, like 30 secs.
3. Arm and leave.
4. Second 21 of video (armed time 2:01) I activate rescue mode.
5. Second 40 of video (armed time 2:21) I deactivate rescue to come back manually.
To consider:
-Minute of video 1:01 (armed time 2:41) that there is a stone white in the ground and distance it says 2m. That is the home point.
- At this point, the arrow works properly.
- It was very windy.

**Expected behavior**
I guess, come directly home in rescue mode.

**Flight controller configuration**
``` 
gps_provider = UBLOX
gps_sbas_mode = AUTO
gps_auto_config = ON
gps_auto_baud = OFF
gps_ublox_use_galileo = OFF
gps_set_home_point_once = OFF
gps_rescue_angle = 32
gps_rescue_initial_alt = 50
gps_rescue_descent_dist = 200
gps_rescue_ground_speed = 2000
gps_rescue_throttle_p = 150
gps_rescue_throttle_i = 20
gps_rescue_throttle_d = 50
gps_rescue_velocity_p = 80
gps_rescue_velocity_i = 20
gps_rescue_velocity_d = 15
gps_rescue_yaw_p = 40
gps_rescue_throttle_min = 1200
gps_rescue_throttle_max = 1600
gps_rescue_throttle_hover = 1280
gps_rescue_sanity_checks = RESCUE_SANITY_ON
gps_rescue_min_sats = 8
gps_rescue_min_dth = 100
gps_rescue_allow_arming_without_fix = OFF
gps_rescue_use_mag = ON
``` 

**Setup / Versions**
 - Flight controller Matek F722-SE AIO. Betaflight just flashed the 4.0.3.
 - Other components: crossfire nano receiver with diversity, gps Beitian 880-BN with compass, tbs unify pro v3 5V, runcam swift 2, vtx antenna tbs triumph, hellgate, 1F capacitor, 4S lipo 1500 100C, tmotor F40 1600kv.  
 - Not sure how to explain. About gps connected to uart 2 and scl and sda soldered to i2c proper paths.

**Additional context**
As I said, it was a windy day, specially up in the air. The area is wind milds area kind of.
I repeated 5 times the flight, always same behaving. 
People said arrow points to where drone has to fly to compensate wind, not necessary home point. But documentation is quite clear about it. At same time, dont understand why it has to face that far away to compensate wind, so that is why I open it as a possible bug.

Jetrell, 2019-06-10:
You state that your using a magnetometer.  Are you certain that it isn't receiving magnetic interference from the motors or power wires?
 This could be causing the magnetic heading to change. Which may lead to a conflict in the firmware, between the GPS and Compass return heading.

McGiverGim, 2019-06-10:
To see if it is a problem of the MAG, you can try `set gps_rescue_use_mag = OFF` in this way it will use only the GPS to bring the quad home.

Take into account too that the GPS Rescue is only that, a Rescue element. It must work more or less but it will not be perfect and can go wrong for some meters.

christianv2m, 2019-06-11:
Hi @McGiverGim  and @Jetrell , thanks for your answers.
As you recommended, next day I already tried disabling magnetometer and set the bauds of GPS to AUTO.
Behaviour is the same, the closest to the home point it works kind of OK, the farthest it does not. At the same time, the distance value is always working properly.
About rescue mode, I got to bring back the drone in rescue mode 3 times (even with a wrong arrow and the drone comming back looking backwards or in a side - where the arrow is pointing-) but it fail 2  other times.

closed 2019-07-18 by stale[bot]

ZupinM, 2021-02-13:
Today I have setup GPS rescue and I am having the same issue.  When flying manualy,  the arrow points home correctly, but when I activate rescue mode, the quad turns in the right direction towards home, but then home arrow stays pointing straight ahead as in video above, while quad keeps drifting in the wrong direction, until it detects flyaway and disarms.

Anyone found solution to this issue?