betaflight/betaflight #8405: User defined GPS filtering

cpihl, 2019-06-10:
Problem
The GPS filtering defined in io/gps.c/GPS_FILTER_VECTOR_LENGTH causes an incorrect compass display and home direction when no compass installed and instead based on gps track history. This in turn results in resque mode flying away from home. It may correct itself after some time but it is scary to see the resque mode flying away at high speed. It also reaches higher resque altitude than specified for fast drones as the delay makes it think it has not even started to move upwards when already reached target altitude.

When making rapid turns the GPS delay combined with gyroscope low latency results in erratic compass both in OSD and resque mode when no magnetometer installed.

Solution
User defined GPS_FILTER_VECTOR_LENGTH, where 0 results in no filtering. I tried with "#define GPS_FILTERING  0 " with excellent results, but this require recompiling. Maybe a setting field in Betaflight configurator GPS tab.

Tested with Gps BN-880 without connecting the magnetometer SDA/SDL wires, and Betaflight F4. With default GPS filtering the delay appear to be 5-10 seconds in OSD.

etracer65, 2019-06-10:
Heading using GPS data depends on distance traveled in a given direction to determine the heading vector. So if you're not moving (or moving slowly) or rapidly changing direction the GPS-based heading will not be very accurate. GPS has no way to determine heading other than through the change in position returned by the satellite fix. This is a limitation of GPS and not something that we can fix.

cpihl, 2019-06-11:
Yes, the GPS is a limitation. However the mentioned filter delay makes it a lot worse.

Example: A drone traveling at 100km/h in one direction for 5 seconds (5 samples filter), will travel 139m and then turn back all the way, will result in a compass with inverted direction visible in OSD. At rescue mode, programmed to altitude 100m (gps_rescue_initial_alt), the drone will first reach an altitude of 238m and then travel straight away from home for at least for 138m. 

Some drones can do 200km/h or more so then double the problem described above. The problem persists as long as any turn is made within the delay interval of 5 seconds, maybe during entire flight.

I have tested, and disabling the GPS filter eliminates the problem mentioned above.

atomiclama, 2019-06-11:
When I get my quad back in the air I'm going to try this. Cheers.
The issue I've had during rescue testing is that when activated the quad turns very rapidly, this knocks out the usually fairly accurate home arrow and the quad sets off for home in the wrong direction. It does recover at some point later but it's a bit disconcerting to watch the quad head off on it's own.

cpihl, 2019-06-11:
Make sure you test with GPS update speed of 1 sample per second, and fly as I described above. The default speed for Beitian BN-880 is 1Hz, even if it can be changed to 10Hz. Maybe the issue I describe is partly the source of issue #8409 https://github.com/betaflight/betaflight/issues/8409.

McGiverGim, 2019-06-11:
I don't know too much about the way the UBLOX messages work, but we have too a "reduced" speed obtaining messages, take a look to the comments on the right:
https://github.com/betaflight/betaflight/blob/a6ae050f9b1d9d904c551ca1acc1cd7d4859c97b/src/main/io/gps.c#L173-L182

We receive only one message of each five to reduce bandwidth. @cpihl I don't know if you can change this value too to see if it goes better.

atomiclama, 2019-06-11:
From memory the slow messages do not contain location or speed information. Think it is time and number  / signal strength of the satellites so just status info.

McGiverGim, 2019-06-11:
Thanks @atomiclama for the info!!

etracer65, 2019-06-11:
@cpihl Your testing and the resulting positive result is an example of confirmation bias. The code you're referring to isn't doing anything as it affects the GPS LAT/LON while in position hold. GPS position hold has been removed from Betaflight. So this is just legacy code that needs to be removed since it's doing nothing.

The real solution to your problem is to update your GPS module to provide data faster than 1Hz. 5Hz or 10Hz are preferred (along with the requisite baud rate increase).

etracer65, 2019-06-11:
Further information that you in fact didn't test any changes comes from:
> I tried with "#define GPS_FILTERING 0 " with excellent results

This change wouldn't have disabled the code. To disable the code you would need to either simply comment out the `#define GPS_FILTERING 1` line or add a `#undef GPS_FILTERING` after the original define.

And lastly if you had actually set `GPS_FILTER_VECTOR_LENGTH` to zero, then you would have created a division-by-zero exception in the code which would have caused unexpected bad results. But fortunately since the (now corrupted) value was never used it didn't cause anything bad to happen.

cpihl, 2019-06-11:
@etracer65  You are right. I made multiple mistakes.

When I changed the code I saw clear change in OSD including rescue arrow, however at the same time i merged in release 4.0.3 while starting from 4.0.0 and that must have been the source of the change, not the GPS_FILTERING define. Thank you for the info.

closed 2019-06-11 by cpihl