betaflight/betaflight #8409: Slow GPS fix/not accurate

methisto, 2019-06-10:
Hello: I have some Problems with my GPS on my 2nd Tyro99

Setup: 
Eachine Tyro99 (F4 Board)
Frsky RX4R Telemetry Receiver
BN-180 GPS

My Problem is that it takes really long to get the first GPS Fix when i start flying. After oen flight it goes faster, finds 7-10 satellites but next day the same issue.

The second one is that the GPS Position is sometimes wrong!
When i fly away 1-2 km the arrow sometimes starts to turn and point at the wrong direction!
When this happens and i fly back home, the distance from Home increases, and today even when i was 10m from my start point away it shows that im 70-80m away!

Should i try to mount the GPS more away from the Receiver and other stuff? Have it something to do with the Baud? Tried a second BN-180 but it didnt help...

I have a second Tyro99 with a XM+ non telemetry and an other GPS (not sure which one) and this one works flawless!

Thanks

etracer65, 2019-06-10:
Sounds like the GPS module is faulty or misconfigured. Have you used uCenter and configured the GPS module?

cpihl, 2019-06-11:
Maybe the turning arrow is the result of issue #8405 ?

aowi7280, 2019-06-12:
Mine was doing the same thing. Some GPS units have a very small backup battery. Mine only held the last known location for 3 hours. And they can be really slow to charge, several hours.  So I bought a 3V 5.5mAH Rechargeable Coin Cell Button Battery MS621FE FL11E and soldered it into the GPS module. It is big enough to hold the memory over night. You could go to a MS920 or 1220 battery for longer time.

aowi7280, 2019-06-12:
I have a Holybro micro GPS, M8N.

christopherwhull, 2019-06-13:
</soapbox> Got to remind myself it is an era where in general people have never used a non network attached GPS before.   My first Garmin in 1997 took 20 minutes to get a fix after 5 hours of air travel or replacing the batteries.  </soapbox>

Please review how a non network connected GPS gets initial fixes.  Up to a 12 minute initial fix per day is expected if the startup timing is unlucky.    The easy fix is power your FC with a battery bank under the sky as you lay out the tools of the trade for the day.    The GPS will keep that information in internal NVRAM until it becomes to old to use. 

The google keywords are "GPS Ephemeris Almanac."   The time to an intial fix will never match a network connected cell phone. 

https://en.wikipedia.org/wiki/GPS_signals#Almanac

atomiclama, 2019-06-13:
Beat me to it.
I just power up my quad via the USB on the way to the field and let the GPS sort it's self out. Can take 10-15 min for a first fix in the morning.

closed 2019-07-20 by stale[bot]