betaflight/betaflight #8410: AirMode

MBR595, 2019-06-10:
Hello,

I'm using a Lux f7 board with DALRC 40A engine. Oddly enough air mode is not working properly. If I turn it off the quad will flip and roll beautifully. But, when I turn it on the quad flips and rolls uncontrollably until it crashes. I'm not sure what to do. I've re-flashed, increased the min throt %, swapped the gyro. I'm not sure what else to do. Any help is appreciated. Also, I have a macbook and bl heli suite and black box logger don't play well with the mac. So there's that too.

etracer65, 2019-06-11:
Blackbox log viewer works just fine on Mac - I use it all the time. You need to provide the requested information when opening a new issue. Please read the following and provide the requested information - particularly the `diff` of your settings.

> **Describe the bug**
> A clear and concise description of what the bug is.
> 
> **To Reproduce**
> Steps to reproduce the behavior:
> 
> **Expected behavior**
> A clear and concise description of what you expected to happen.
> 
> **Flight controller configuration**
> Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
> 
> **Setup / Versions**
>  - Flight controller [what type is it, where was it bought from];
>  - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
>  - how are the different components wired up.
> 
> **Additional context**
> Add any other context about the problem here.

You'll also need to provide a blackbox log showing the problem occurring for anybody to be able to help.

RipperDrone, 2019-06-11:
Maybe you unintentionally have Stabilize/level mode active somehow. This typically results in limiting roll/pitch angles and doing weird tilt-back moves when you try to do fast manoeuvres...?

MBR595, 2019-06-11:
> Setup / Versions
>
> Flight controller— Lux F7 purchased from GET FPV
> Other components RX is a TBS Crossfire Nano.
> how are the different components wired up — all components are hard wired to the flight controller. DALRC 40A 4in1 is the ESC
> Additional context
> Blackbox may work fine with your set up. But it is not working properly with mine. I followed the instructions on set up. Please see screen shots. I’ve attached the files just in case it will work with your system.

> Describe the bug
> Whenever I try to use air mode the copter flips uncontrollably. Without air mode it flies fine. 
> 
> To Reproduce
> I’ve tried to use blackbox on my Mac. I’v used multiple SD cards that I know work and no file has been generated. 
> 
> Expected behavior
> I would expect air mode to work out the box
> 
> Flight controller configuration
> Create a diff and post it here in a code block. Put ``` (three backticks) at the start and end of the diff block (instructions on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/ <https://oscarliang.com/use-diff-not-dump-betaflight/>)
‘''# diff

# version
# Betaflight / MATEKF722SE (MF7S) 4.0.3 Jun  1 2019 / 11:38:08 (094cfc956) MSP API: 1.41

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL

# beeper

# beacon

# map
map TAER1234

# serial

# led

# color

# mode_color

# aux
aux 0 0 0 900 1300 0 0
aux 1 28 1 1900 2100 0 0
aux 2 35 2 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_to_use = SECOND
set max_check = 2000
set rssi_channel = 12
set serialrx_provider = CRSF
set dshot_idle_value = 0
set motor_pwm_protocol = DSHOT1200
set align_board_yaw = 180

# profile
profile 0

# rateprofile
rateprofile 0

set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30

# end the command batch
batch end

#

MBR595, 2019-06-11:
I’m not sure how I would have done that. The board is brand new and I only fly in rate mode.

etracer65, 2019-06-11:
`dshot_idle_value = 0` - Don't do that!

Set it back to the default: `set dshot_idle_value = 550`

MBR595, 2019-06-11:
I changed it back but I did that trying to get it to work. This problem started out the box with default settings. 
Martin Russell

RipperDrone, 2019-06-11:
Maybe it's idle value wrong/too low. Plus, using 2nd gyro only might still have a bug in the code since 722SE target is pretty new? Try to use first gyro for once, maybe. Just to double check...

MBR595, 2019-06-11:
I tried the first gyro and the second. Same issue.

Martin Russell

RipperDrone, 2019-06-11:
you might wanna try DShot 600 as well, and maybe also lower gyro/PID loop frequencies (4/4 to be on the safe side) - maybe CPU load gets too high? (FAST RAM optimizations of F7 are still a little in the works...)

RipperDrone, 2019-06-11:
have you tried both options to switch AM on - 1) permanently via feature in BFC and 2) on a switch?

MBR595, 2019-06-11:
I tried that as well. Same issue.

Martin Russell

MBR595, 2019-06-11:
Yes. Same issue exists. 

Martin Russell

RipperDrone, 2019-06-11:
Is Matek722SE the correct target for Lux F7?

MBR595, 2019-06-11:
That’s the target listed on GETfpv

Martin Russell

RipperDrone, 2019-06-11:
Since AM is just PID controller active on zero Throttle input, there seems to be something wrong with either your PID values or the rpm idle, it appears. From what you are describing, it resembles behavior of quad in Stabilize/Level mode, however you got it being active...? I'm out here, no more ideas... :-)

etracer65, 2019-06-11:
You likely have an ESC de-sync. Capture a blackbox log of the problem occurring and post it here.

MBR595, 2019-06-11:
Thank you for your help. This has been frustrating on all levels. 

Martin Russell

RipperDrone, 2019-06-11:
BTW, AFAIK AM gets enabled after surpassing a certain throttle value only (31% or something by default). I guess you did raise throttle beyond that threshold before trying the flips/rolls...

MBR595, 2019-06-11:
These are the logs. As I mentioned, I’ve had problems accessing it on my MacBook. Hopefully they work for you.

MBR595, 2019-06-11:
Yes, I raised it in BF and on the Taranis. When I raised the MIN throttle in the Taranis I could not get the quad to arm until I lowered it back to stock levels.

etracer65, 2019-06-11:
Where are the logs? You can't attach files using email replies.

MBR595, 2019-06-11:
How do you want me to send them to you if I can’t attach them?

etracer65, 2019-06-11:
Like most people - Use the actual GitHub site to reply instead of replying through email.

MBR595, 2019-06-11:
There’s no need to be rude.

etracer65, 2019-06-11:
Rude? Really? Answering your question so you can supply the information so we can help you solve your problem and that's somehow rude?

MBR595, 2019-06-11:
Yea really. You have a rude tone. I’ve never used Git Hub before. I’m just trying to get in the air and your condescending.

MBR595, 2019-06-11:
I mentioned before that blackbox hasn't been working for me. The files save as .BFL not TXT

etracer65, 2019-06-11:
.BFL is correct. Well good luck. We have better things to do then try and help people that apparently don't want to be helped.

closed 2019-06-11 by etracer65