betaflight/betaflight #8412: iFlight SucceX F7 TwinG (EXF722DUAL) - Motors 5 and 6 do not work with DSHOT enabled

FLDataTeK, 2019-06-11:
**Describe the bug**
When DSHOT is enabled motors 5 and 6 do not work. (this is one a X8 configuration) 

AJ Christensen has pinpointed the cause of the issue to be the LED strip is causing a conflict with DSHOT burst.  I disabled LED strip at his request and motor 5 and 6 now work fine. Details of the issue and resolution can be found at the link below.  This bug report is being submitted at his request for tracking purposes.

https://www.facebook.com/groups/betaflightgroup/permalink/330362470992969/

 # resource
resource BEEPER 1 C15
resource MOTOR 1 C08
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C07
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 B01
resource MOTOR 8 B00
resource PPM 1 A03
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 C10
resource SERIAL_TX 5 C12
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 C11
resource SERIAL_RX 5 D02
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11
resource LED 1 C04
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource CAMERA_CONTROL 1 A00
resource ADC_BATT 1 C01
resource ADC_RSSI 1 C00
resource ADC_CURR 1 C02
resource FLASH_CS 1 B09
resource OSD_CS 1 B12
resource GYRO_EXTI 1 A08
resource GYRO_EXTI 2 C13
resource GYRO_CS 1 A15
resource GYRO_CS 2 C03

# diff all
# version
# Betaflight / EXF722DUAL (EX7P) 4.0.3 Jun 1 2019 / 11:21:46 (094cfc956) MSP API: 1.41
# start the command batch
batch start
# reset configuration to default settings
defaults nosave
board_name EXF722DUAL
manufacturer_id 
mcu_id 002800463237510734363730
signature 
# name
# resources
# timer
# dma
# mixer
mixer OCTOX8
# servo
# servo mix
# feature
# beeper
# beacon
# map
# serial
# led
# color
# mode_color
# aux
# adjrange
# rxrange
# vtx
# rxfail
# display_name
# master
set motor_pwm_protocol = MULTISHOT
# profile
profile 0
# profile
profile 1
# profile
profile 2
# restore original profile selection
profile 0
# rateprofile
rateprofile 0
# rateprofile
rateprofile 1
# rateprofile
rateprofile 2
# rateprofile
rateprofile 3
# rateprofile
rateprofile 4
# rateprofile
rateprofile 5
# restore original rateprofile selection
rateprofile 0
# save configuration
save

fujin, 2019-06-12:
Was actually @mikeller who spotted the conflict, thanks for reporting back! 

I think we should just document the limitation in https://github.com/betaflight/betaflight/blob/master/docs/boards/Board%20-%20EXF722DUAL.md and perhaps ask iFlight to mention it in their product description.

closed 2019-07-19 by stale[bot]

roybeukers, 2021-09-03:
I had the same thing. Turn the led strip function off in the config tab