betaflight/betaflight #8417: BF 4.0.3 for Mamba F7 stack core heat 75 degrees

MoonKIE666, 2019-06-12:
I have installed the last stable version 4.0.3 in my Mamba F722 stack. And the flight controller gets very hot.  I think all the heat comes from the F7 chip.  It gets so hot that it produces a warning message on the OSD: CORE HEAT 75 Celsius degrees.

SchonRichtig, 2019-06-12:
While flying?
Or on the ground in the sun?
My Kakute F7 also gehts really hot on the ground and there is still 3.5.7 installd.

GhostMaster9, 2019-06-12:
F7 it's normal, but only when you are at the ground.

etracer65, 2019-06-12:
F7 in a tight stack without airflow, yes it will get hot. If you want to ignore the warning then either change the warning temp or turn the core temperature warning off. Otherwise improve the airflow around the processor. But the warning is working as designed and there are no problems with Betaflight.

closed 2019-06-12 by etracer65

GhostMaster9, 2019-06-12:
i would not recommend turning the warning off... 70 degress it's already very hot and will cause problems if the user ignore it.

pawelksi, 2019-06-24:
What then can be done, is there no way to lower the core temperature programmatically? I bought a new HGLRC Forward F7 with F745 chip, I have the same problem, without additional hardware, only connected by USB after less than a minute heats up above 70 degrees. This is dangerous, and actually excludes using it at competitions. It's not long before LiPo is connected to the start, and it may turn out that due to the temperature the drone will not fly.

etracer65, 2019-06-24:
Complain to the hardware manufacturer and/or use a different board.

pawelksi, 2019-06-24:
I filed a complaint, I got a refund of cash, but I would like to understand why this is happening, why the temperature is growing so drastically. It may be possible to lower the core clock speed and thereby reduce the temperature, it is known that F 7 warms up much more than F4

etracer65, 2019-06-24:
Just a poor hardware design without thought to thermal considerations.

pawelksi, 2019-06-24:
How can you take care of thermal issues, what do you mean? See my video, MCU is on the top.
https://photos.app.goo.gl/QEMArFs2JPHgQeC59

etracer65, 2019-06-24:
> Just a poor hardware design without thought to thermal considerations.

By the board designer.

In your case you could probably improve the situation slightly by removing the (apparently multiple) stickers on top of the MCU so that it can actually radiate heat...

pawelksi, 2019-06-24:
It's no diference. Genuine MCU just stopped working one day. I replace for a new one, but the temperature is the same. I thought the processors were from a bad party, but the new ones work identically

mikeller, 2019-06-25:
@pawelksi: The STM32F745 is known to heat up more than other MCUs when running Betaflight, and require a board design that allows good dissipation of heat - other boards using this MCU have the same problem.