betaflight/betaflight #8423: The rx_spi can't bind to Jumper T16

zhangzq71, 2019-06-13:
**The rx_spi can't bind to Jumper T16**
I tried to add a CC2500 module to a Taulabs Quanton board. That CC2500 module had been verified that it could be bind to Jumper T16 in D8 mode by using OpenSky project. And I am sure the rx_spi works because I checked this by using a logic analyzer, and I compared the spi data while in binding mode with the OpenSky project and found 

1. BF keep sending (0x0C, 0x96), (0x0C, 0x9B), (0x0C, 0xA0) with the second value increase 6 for each.
2. OpenSky keep sending (0xF5, 0xFF) 

I am trying the BF in version 'commit 9d5017103484f1889a2a3922c72193f5f573fe2d'

mikeller, 2019-06-13:
@zhangzq71: Can you please supply the information you were asked to supply when opening this issue?

zhangzq71, 2019-06-13:
@mikeller Sorry, this is my first time to posted the issue, I don't know what are the necessary information?
The board I am using is Quanton board, using the BF version a6c07b0. What are the other information do you want?

jflyper, 2019-06-15:
The information were stated in the pre-formatted text you have removed when you opened this issue ticket. You can view it again by pressing a new issue button (just view it, pls don’t open a new issue)

zhangzq71, 2019-06-15:
@jflyper thanks for your information.

zhangzq71, 2019-06-15:
@mikeller I think I've provided the necessary information, is it OK to remove the 'Missing Information'

mikeller, 2019-06-16:
@zhangzq71: The COLIBRI target that you are using is built without support for SPI RX - this will never work.

zhangzq71, 2019-06-16:
@mikeller Sorry, I forgot to mention that I've modified the COLIBRI configuration, here it is,
```
/*
 * This file is part of Cleanflight and Betaflight.
 *
 * Cleanflight and Betaflight are free software. You can redistribute
 * this software and/or modify this software under the terms of the
 * GNU General Public License as published by the Free Software
 * Foundation, either version 3 of the License, or (at your option)
 * any later version.
 *
 * Cleanflight and Betaflight are distributed in the hope that they
 * will be useful, but WITHOUT ANY WARRANTY; without even the implied
 * warranty of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
 * See the GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this software.
 *
 * If not, see <http://www.gnu.org/licenses/>.
 */

#pragma once

#define TARGET_BOARD_IDENTIFIER "COLI"

#define USBD_PRODUCT_STRING "Colibri"
#ifdef OPBL
#define USBD_SERIALNUMBER_STRING "0x8020000"
#endif

#define TARGET_XTAL_MHZ         16

#define LED0_PIN                PC14
#define LED1_PIN                PC13

#define USE_BEEPER
#define BEEPER_PIN              PC5

#define INVERTER_PIN_UART2      PB2 // PB2 used as inverter select GPIO

#define GYRO_1_CS_PIN           PC4
#define GYRO_1_SPI_INSTANCE     SPI1

#define USE_ACC
#define USE_ACC_SPI_MPU6000

#define USE_GYRO
#define USE_GYRO_SPI_MPU6000
#define GYRO_1_ALIGN            CW270_DEG_FLIP

// MPU6000 interrupts
#define USE_EXTI
#define USE_GYRO_EXTI
#define GYRO_1_EXTI_PIN         PC0
#define USE_MPU_DATA_READY_SIGNAL

#define USE_MAG
#define USE_MAG_HMC5883
#define USE_MAG_QMC5883
#define MAG_HMC5883_ALIGN       CW270_DEG_FLIP

#define MAG_INT_EXTI            PC1
#define USE_MAG_DATA_READY_SIGNAL
#define ENSURE_MAG_DATA_READY_IS_HIGH

#define USE_BARO
#define USE_BARO_MS5611

#define FLASH_CS_PIN            PB12
#define FLASH_SPI_INSTANCE      SPI2

#define USE_FLASHFS
#define USE_FLASH_M25P16

#define USE_VCP
#define USE_USB_DETECT
#define USB_DETECT_PIN          PA9

#define USE_UART1
#define UART1_RX_PIN            PB7
#define UART1_TX_PIN            PB6

// #define USE_UART2
#define UART2_RX_PIN            PA3
#define UART2_TX_PIN            PA2

// #define USE_UART3
#define UART3_RX_PIN            PB11
#define UART3_TX_PIN            PB10

//#define USE_UART4
#define UART4_RX_PIN PC11
#define UART4_TX_PIN PC10

//#define USE_UART5
#define UART5_RX_PIN            PD2
#define UART5_TX_PIN            PC12

// #define USE_SOFTSERIAL1
// #define USE_SOFTSERIAL2

#define SERIAL_PORT_COUNT       2 //VCP, UART1, UART2, UART3, SOFTSERIAL x 2

#define USE_ESCSERIAL
#define ESCSERIAL_TIMER_TX_PIN  PA10  // (HARDARE=0,PPM)

#define USE_SPI

#define USE_SPI_DEVICE_1
#define SPI1_NSS_PIN            PC4
#define SPI1_SCK_PIN            PA5
#define SPI1_MISO_PIN           PA6
#define SPI1_MOSI_PIN           PA7

#define USE_SPI_DEVICE_2
#define SPI2_NSS_PIN            PB12
#define SPI2_SCK_PIN            PB13
#define SPI2_MISO_PIN           PC2
#define SPI2_MOSI_PIN           PC3

#define USE_SPI_DEVICE_3
#define SPI3_NSS_PIN            PD2
#define SPI3_SCK_PIN            PC10
#define SPI3_MISO_PIN           PC11
#define SPI3_MOSI_PIN           PC12

#define USE_I2C
#define USE_I2C_DEVICE_3
#define I2C_DEVICE              (I2CDEV_3)
#define I2C3_SCL                PA8
#define I2C3_SDA                PC9

// alternative defaults for Colibri/Gemini target
#define USE_TARGET_CONFIG

#define ENABLE_BLACKBOX_LOGGING_ON_SPIFLASH_BY_DEFAULT

// #define DEFAULT_RX_FEATURE      FEATURE_RX_SERIAL

#define TARGET_IO_PORTA         0xffff
#define TARGET_IO_PORTB         0xffff
#define TARGET_IO_PORTC         0xffff
#define TARGET_IO_PORTD         0xffff

#define USABLE_TIMER_CHANNEL_COUNT 6
// #define USED_TIMERS             ( TIM_N(1) | TIM_N(2) | TIM_N(3) | TIM_N(4) | TIM_N(5) | TIM_N(12) | TIM_N(8) | TIM_N(10) | TIM_N(11))
#define USED_TIMERS             ( TIM_N(1) | TIM_N(3) | TIM_N(8)

#define USE_RX_SPI
#define RX_SPI_INSTANCE         SPI3

#define USE_RX_FRSKY_SPI_D
#define USE_RX_FRSKY_SPI_X
#define USE_RX_SFHSS_SPI
#define DEFAULT_RX_FEATURE      FEATURE_RX_SPI
#define RX_SPI_DEFAULT_PROTOCOL RX_SPI_FRSKY_D
#define USE_RX_FRSKY_SPI_TELEMETRY

#define RX_NSS_PIN              PD2

#define RX_SPI_EXTI_PIN         PB10

#define USE_RX_CC2500_SPI_PA_LNA

#define RX_CC2500_SPI_TX_EN_PIN   PB14
#define RX_CC2500_SPI_LNA_EN_PIN  PB15

#define BINDPLUG_PIN             NONE

#define RX_SPI_LED_PIN          PA2
```

And the SPI connection between STM32F405RG and CC2500 module works, and here is what I got from the Logic Analyzer that probed that SPI connection, which is the data sent when in binding mode, and have a comparison to OpenSky project which can be successfully bind to Jumper T16, 
1. BF keep sending (0x0C, 0x96), (0x0C, 0x9B), (0x0C, 0xA0) with the second value increase 5 for each.
2. OpenSky keep sending (0xF5, 0xFF)

mikeller, 2019-06-16:
If you are using custom hardware with custom firmware it is really hard to diagnose what is not working. RX_SPI has been proven to work for non-custom hardware with FrSky or Multimodule TX for a lot of users.
In order to narrow down what it is that is not working, can I suggest that you first try to bind your custom hardware / firmware to a FrSky (XJT) or Multimodule TX, and only then move on to an unproven TX?
Also, you report that you managed to bind OpenSky in FrSky D mode (D8), and now you are trying to bind Betaflight in FrSky X mode (D16) - this is making your observations about data being sent pretty much pointless - they are different protocols.

zhangzq71, 2019-06-16:
The Jumper T16 used the DIY-Multiprotocol-TX module which can successfully bind to OpenSky in D8 mode while OpenSky has only D8 mode support. Actually I tried to bind (Quanton+CC2500) hardware to Jumper T16 in both D8 and D16 mode, and D16(8 ch) mode, but all failed, so I don't know what is the wrong, I've not found the difference in the source code yet.

mikeller, 2019-06-16:
Then I suspect something with your hardware configuration is different to what Betaflight expects - both FrSky D and FrSky X work just fine with the FrSky SPI RX and a Multiprotocol module.

zhangzq71, 2019-06-16:
Yes, that is why I posted this issue, what is most possible cause do you think?

mikeller, 2019-06-17:
Not really sure where the problem lies, hard to tell without having the hardware. Does the RX LED blink, then go solid when binding, and then blink again when bound?

Also, can I recommend you move to the standard STM32F405 Unified Target, and then configure the SPI RX through runtime configuration (see #8440 for how to do this)? This way at least we only have to consider the configuration and not also the target definition.

zhangzq71, 2019-06-17:
The RX LED blinks if the SPI connection to CC2500 module is OK, if I removed one of the SPI line the RX LED does not blink, and it is always on after run the 'bind_rx_spi' command in CLI.

I've inspected the data between by adding a logic analyzer in between the SPI connection to CC2500 module, and found the data send to and read from CC2500 module are OK, so I concerned about if something wrong in the firmware. However I will try to move to the standard STM32F405 Unified Target.

zhangzq71, 2019-06-18:
@mikeller I tried to move to STM32F405 unified target, and then put the content from 'dump hardware' command, it has some error
```
# resource BARO_CS 1 NONE
Resource is freed

# resource BARO_EOC 1 NONE
###ERROR: INVALID RESOURCE NAME: 'BARO_EOC'###

# resource BARO_XCLR 1 NONE
###ERROR: INVALID RESOURCE NAME: 'BARO_XCLR'###

# set mco2_on_pc9 = OFF
###ERROR: INVALID NAME###
```
then I comment those statement and run again, no error. then 'save'.

After reconnected I found the model in the setup keep rotating randomly. and try to 'bind_rx_spi', but it has the same problem as before. here it is the 'dump hardware' output
```
# version
# Betaflight / COLIBRI (COLI) 4.1.0 Jun 13 2019 / 22:37:19 (ee564432f) MSP API: 1.42

# start the command batch
batch start

board_name COLIBRI
manufacturer_id 

# resources
resource BEEPER 1 C05
resource MOTOR 1 B00
resource MOTOR 2 B04
resource MOTOR 3 B01
resource MOTOR 4 B05
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A10
resource PWM 1 A10
resource PWM 2 C06
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource INVERTER 1 NONE
resource INVERTER 2 NONE
resource INVERTER 3 NONE
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
resource INVERTER 8 NONE
resource INVERTER 9 NONE
resource INVERTER 10 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 NONE
resource I2C_SCL 3 A08
resource I2C_SDA 1 NONE
resource I2C_SDA 2 NONE
resource I2C_SDA 3 C09
resource LED 1 C14
resource LED 2 C13
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 C02
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 C03
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 A10
resource CAMERA_CONTROL 1 NONE
resource ADC_BATT 1 NONE
resource ADC_RSSI 1 NONE
resource ADC_CURR 1 NONE
resource ADC_EXT 1 NONE
resource BARO_CS 1 NONE
resource BARO_EOC 1 NONE
resource BARO_XCLR 1 NONE
resource COMPASS_CS 1 NONE
resource COMPASS_EXTI 1 C01
resource PINIO 1 NONE
resource PINIO 2 NONE
resource PINIO 3 NONE
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 B12
resource RX_SPI_CS 1 D02
resource RX_SPI_EXTI 1 B10
resource RX_SPI_BIND 1 NONE
resource RX_SPI_LED 1 A02
resource RX_SPI_CC2500_TX_EN 1 B14
resource RX_SPI_CC2500_LNA_EN 1 B15
resource GYRO_EXTI 1 C00
resource GYRO_EXTI 2 NONE
resource GYRO_CS 1 C04
resource GYRO_CS 2 NONE
resource USB_DETECT 1 A09

# timer
timer A10 AF1
# pin A10: TIM1 CH3 (AF1)
timer C06 AF3
# pin C06: TIM8 CH1 (AF3)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B04 AF2
# pin B04: TIM3 CH1 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B05 AF2
# pin B05: TIM3 CH2 (AF2)

# dma
dma SPI_TX 1 NONE
dma SPI_TX 2 NONE
dma SPI_TX 3 NONE
dma SPI_RX 1 NONE
dma SPI_RX 2 NONE
dma SPI_RX 3 NONE
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
dma ADC 2 NONE
dma ADC 3 NONE
dma UART_TX 1 NONE
dma UART_TX 2 NONE
dma UART_TX 3 NONE
dma UART_TX 4 NONE
dma UART_TX 5 NONE
dma UART_TX 6 NONE
dma UART_TX 7 NONE
dma UART_TX 8 NONE
dma UART_RX 1 NONE
dma UART_RX 2 NONE
dma UART_RX 3 NONE
dma UART_RX 4 NONE
dma UART_RX 5 NONE
dma UART_RX 6 NONE
dma UART_RX 7 NONE
dma UART_RX 8 NONE
dma pin A10 0
# pin A10: DMA2 Stream 6 Channel 0
dma pin C06 0
# pin C06: DMA2 Stream 2 Channel 0
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B04 0
# pin B04: DMA1 Stream 4 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin B05 0
# pin B05: DMA1 Stream 5 Channel 5

# master
set gyro_to_use = FIRST
set align_mag = DEFAULT
set mag_bustype = I2C
set mag_i2c_device = 3
set mag_i2c_address = 0
set mag_spi_device = 0
set baro_bustype = I2C
set baro_spi_device = 0
set baro_i2c_device = 3
set baro_i2c_address = 0
set rx_spi_bus = 3
set rx_spi_led_inversion = OFF
set adc_device = 1
set blackbox_device = SPIFLASH
set dshot_burst = OFF
set current_meter = VIRTUAL
set battery_meter = NONE
set ibata_scale = 400
set beeper_inversion = OFF
set beeper_od = ON
set beeper_frequency = 0
set system_hse_mhz = 16
set cc2500_spi_chip_detect = ON
set led_inversion = 0
set dashboard_i2c_bus = 3
set dashboard_i2c_addr = 60
set usb_msc_pin_pullup = ON
set flash_spi_bus = 2
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_i2cBus = 0
set gyro_1_i2c_address = 0
set gyro_1_sensor_align = CW270FLIP
set gyro_2_bustype = SPI
set gyro_2_spibus = 1
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = DEFAULT
set mco2_on_pc9 = OFF

# end the command batch
batch end
```

mikeller, 2019-06-20:
@zhangzq71:

> # resource BARO_EOC 1 NONE
> ###ERROR: INVALID RESOURCE NAME: 'BARO_EOC'###

Looks like you are trying to apply a dump from 4.1 to a 4.0 unified target - this won't work, either create the `dump hardware` with a 4.0 target, or use a 4.1 Unified Target to apply it to.

zhangzq71, 2019-06-20:
@mikeller OK, I will try it tonight.

zhangzq71, 2019-06-21:
It is OK now, because I supposed the CC2500 data received interrupt pin should be GDO2 which as OpenSky does, but after reading the code and found BF uses GDO0. Thanks mikeller.

closed 2019-06-21 by zhangzq71

mikeller, 2019-06-23:
@zhangzq71: Interesting. As far as I know they can be used interchangeably, dependent on how the CC2500 is initialised. Do you know if there is any OpenSky hardware out there that can run Betaflight? If so, it might make sense to make this configurable.

zhangzq71, 2019-06-23:
Yes, GDO0 and GDO2 are interchangeable. OpenSky is nothing but a receiver how can it run Betaflight? I have no idea on how to make selection of GDO0 or GDO2 can be configurable in firmware, because which GDO0/GDO2 pin does connect to STM32F405 had been determined  in hardware design. Only one way is to connect both GDO0/GDO2 to 2 of the STM32F405 pin, but I think is better to use a jumper in the board to let user choose which pin he like to be the CC2500 data received output.

mikeller, 2019-06-24:
@zhangzq71: If there aren't any boards that can run Betaflight and come with GDO2 pre-wired for the data received interrupt then my remark is moot, and there is no need to make this configurable, and Betaflight will just use GDO0 in all cases.