betaflight/betaflight #8425: Is any support for the ST LPS22HB barometer being developed or requested yet?

wingunder, 2019-06-13:
Is someone planning, or maybe already busy implementing the LPS22HB barometer for BF?

Possible software candidates that could be modified/wrapped and used, would be:

- https://github.com/simondlevy/LPS22HB
(Copyright: GPL 3)

- https://github.com/aws/amazon-freertos/tree/master/lib/third_party/mcu_vendor/st/stm32l475_discovery/BSP/Components/lps22hb
(Copyright: STMicroelectronics)

- https://github.com/intel/zephyr/tree/master/drivers/sensor/lps22hb
(Copyright: Apache 2.0)

- https://github.com/stm32duino/LPS22HB/tree/master/src
(Copyright: STMicroelectronics)

The following off-the-shelf break-out boards could be used for testing:
- [STEVAL-MET001V1](https://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mems-motion-sensor-eval-boards/steval-met001v1.html) evaluation break-out board from STMicroelectronics.
- [Ultimate Sensor Fusion Solution - LSM6DSM + LIS2MD](https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-lsm6dsm-lis2md/), in 'pass-through' mode.

mikeller, 2019-06-13:
@wingunder: Nothing planned as far as I know. Are there any flight controllers out there / planned that use this chip?

wingunder, 2019-06-13:
@mikeller: No, not that I know of, but I have an LS22HB and I'm planning to attach it to a NERO.

Before I start hacking on something, I just wanted to know if this is not already work in progress.

Furthermore, any pointers or suggestions to libraries and license preferences for such an integration will be welcome.

mikeller, 2019-06-13:
@wingunder: Sounds good, go for it. As for how to implement it, it's probably easiest to see which one of the existing barometer drivers in `src/main/drivers/barometer/barometer` matches this one closest, and use that driver as a template.

etracer65, 2019-06-13:
Since that sensor supports I2C and SPI it would probably be best to base the driver on a sensor that does likewise - like BMP280 or MS5611.

wingunder, 2019-06-13:
@mikeller: OK thanks, I already peeked into the other barometer implementations and I'll follow suit.

wingunder, 2019-06-13:
> Since that sensor supports I2C and SPI it would probably be best to base the driver on a sensor that does likewise - like BMP280 or MS5611.
@etracer65: Thanks for the tip.

jflyper, 2019-06-13:
@wingunder If you meant LPS22HB, we already have a working driver (`barometer_lps.c`), and there are at least one market available target that uses it. The option is `USE_BARO_SPI_LPS`.

wingunder, 2019-06-13:
@jflyper Thanks a lot. That's exactly what I was looking for. Sorry for my typo and the confusion about LS22HB. It's indeed the LPS22HB. Closing.

closed 2019-06-13 by wingunder

jflyper, 2019-06-14:
@wingunder If you are interested in newer parts, then there will be a support for BMP388 shortly. Adafruit has a bob.

wingunder, 2019-06-14:
@jflyper Thanks for the tip. I'll see that I get hold of a BMP388 bob in the meantime.