betaflight/betaflight #8428: OMNIBUSF4SD (OBSD) 3.5.0 Aug 14 2018 / 03:49:56 (1d710afd7) MSP API: 1.40

dannygarcia78, 2019-06-14:
**Describe the bug**
A clear and concise description of what the bug is.
i connect to the config and main tab where it displays the gyro it doesnt move at all , and in config tab it says no gyro detected,

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.
on modes tab there is no angle mode , horizon mode, or acro trainer modes ,

etracer65, 2019-06-14:
Please provide the information requested when you opened the issue.

dannygarcia78, 2019-06-15:
Never mind my mistake yall, I was flashing the wrong target

dannygarcia78, 2019-06-15:
This is to confusing how do I delete this whole thing

closed 2019-06-15 by dannygarcia78