betaflight/betaflight #8430: wrong pull up/down usage with halfduplex serial and pushpull setting

ronlix, 2019-06-15:
the halfduplex pull up/down setup on serial IO config workes wrong when using pushpull and half duplex (SERIAL_BIDIR | SERIAL_BIDIR_PP).

see:
https://github.com/betaflight/betaflight/blob/master/src/main/drivers/serial_uart_stm32f7xx.c#L439
```
    if ((options & SERIAL_BIDIR) && txIO) {
        ioConfig_t ioCfg = IO_CONFIG(
            ((options & SERIAL_INVERTED) || (options & SERIAL_BIDIR_PP)) ? GPIO_MODE_AF_PP : GPIO_MODE_AF_OD,
            GPIO_SPEED_FREQ_HIGH,
            ((options & SERIAL_INVERTED) || (options & SERIAL_BIDIR_PP)) ? GPIO_PULLDOWN : GPIO_PULLUP
        );

        IOInit(txIO, OWNER_SERIAL_TX, RESOURCE_INDEX(device));
        IOConfigGPIOAF(txIO, ioCfg, uartdev->tx.af);
    }
```

if SERIAL_BIDIR_PP is set, the pull down will be active which should not be like that.

i will do a pull request with my esc sensor passthrough very soon anyway where i fixed this for the F7. i think on the H7 it will be the same, but i cant test it there. the F4 seems to work fine.
my branch where i fixed it: https://github.com/ronlix/betaflight/tree/FETtec_esc_passthrough

ronlix, 2019-06-16:
i would sugguest to leave:
` ((options & SERIAL_INVERTED) || (options & SERIAL_BIDIR_PP)) ? GPIO_MODE_AF_PP : GPIO_MODE_AF_OD,`
as it is, as it makes sense. with _PP its clear and inverted dont workes in OD mode.

but change:
`((options & SERIAL_INVERTED) || (options & SERIAL_BIDIR_PP)) ? GPIO_PULLDOWN : GPIO_PULLUP`

to:
`(options & SERIAL_INVERTED) ? GPIO_PULLDOWN : GPIO_PULLUP`
as just the fact that the active output is done with pushpull dont means it should be pulled down.

mikeller, 2019-07-17:
@ronlix: Has this been addressed by the changes to serial passthrough?

ronlix, 2019-07-17:
Hi, 

i dont think so, at last not from my side.

mikeller, 2019-07-17:
@blckmn: Fyi, something to be looked at in the UART revamp.

klutvott123, 2022-04-21:
Fixed in https://github.com/betaflight/betaflight/issues/10220

closed 2022-04-21 by klutvott123