betaflight/betaflight #8431: Kakute F4 v2 AIO, Smartaudio and OSD bug

NikoN-FPV, 2019-06-15:
hey devs ;) i have a kakute f4 v2 aio and i got 2 bugs to report, sry i´m new to github :P 

first, the core temp in the osd works on #1609 but not on #1610 till the change --> #1610 (05.05.2019 13:18:06)
Made temperature OSD display consistent. (commit: 05f5449)
only the C is shown in the OSD but no temp..

and the second, vtx sa (tbs unify hv pro) is working on #1657 but not on #1658 cause of --> #1658 (25.05.2019 00:33:08)
SmartAudio conversion to VTX table and V2.1 implementation (commit: e976ea1)
want to change vtx channel/power via osd or luascript but nothing happens.. 

thx in advance :)

etracer65, 2019-06-15:
Please provide the information requested when you opened the issue.

etracer65, 2019-06-15:
Also don't combine multiple unrelated items into one issue. I believe that I already replied to you on the Facebook group that core temp display in the OSD is working without problems.

NikoN-FPV, 2019-06-15:
hey bruce :) yeah it was me, thx for answering ;) ok sry my failure, first time @ github ^^ 
but what to do now? i also tested newer versions until #1699 but the core temp is not here anymore, only the C..  and also the smartaudio does not work anymore.. in the osd and luascript the data is shown but can´t change anything..

etracer65, 2019-06-15:
I just tested the core temperature appearing in the OSD using the latest build 1699 (again) and it works perfectly. Maybe you simply have it placed on the right side of your screen and the numbers are off the edge. Otherwise do a clean, full-chip erase whenever trying to use nightly builds. Don't use a saved `diff`. Set up the minimum manually (only add core temperature to the middle of the OSD) and test.

VTX control now requires you to configure the the `vtxtable` - you really need to read the details of the commits that go into them if you intend to use nightly builds. Read: https://github.com/betaflight/betaflight/pull/8276

NikoN-FPV, 2019-06-15:
i did full chip erase, no diffs or dumps, i placed core temp good readable in the middle and only the C is on the screen..  until #1609 its fine, builds above have only the C.. i can send you a pic, dump or something if it helps.. 

and please forgive me,  my english is not the best, so its hard for me to read all the "coders slang" :P

etracer65, 2019-06-15:
If you are going to use nightly builds it's your responsibility to read and understand the changes. These are provided in the pull requests. If you're not going to do this then stick to only released versions.

As requested when you opened the issue, provide a `diff` or your settings for the non-working configuration. Also provide the output of the `status` command.

McGiverGim, 2019-06-15:
Have you updated to latest fonts of the nightly Configurator? Test it.

NikoN-FPV, 2019-06-15:
ok, sry bruce ^^ my mistake.. i checked all over and the temp is here but around 10 spaces to the right from the C and was overlaying..  

and smartaudio is working again with the vtxtable :) 

so thx for help and sry for stealing your time ^^

closed 2019-06-15 by NikoN-FPV