betaflight/betaflight #8434: Auto enable crash flip mode if disarmed upside down

pkruger, 2019-06-16:
**Is your feature request related to a problem? Please describe.**
During a race, it is time consuming to disarm, set crash flip mode, arm flip, disarm, disable crash flip and arm again.  

**Describe the solution you'd like**
After the first arm, if the quad is upside down when it is disarmed, crash flip (may be if enabled, feature?) should be enabled automatically.  Once the quad is right side up, a disarm and re-arm disables crash flip again.

**Describe alternatives you've considered**
Currently enabling crash flip via a radio aux channel works, but is cumbersome.  Maybe there is a way of enabling crash flip from the radio by using logic switches/telemetry, but that seems complicated to setup (and relies on telemetry)

**Additional context**
None.

theSkonk, 2019-06-16:
it could lead to burnt motors and damage... also a hell  of a lot of racers don't use the accelerometer so it wouldn't work any way.. i don't see this idea talking off as it presents dangers to gear and your surroundings

pkruger, 2019-06-16:
@theSkonk 
Why burnt motors?  You still have to disarm and re-arm, only difference is you don't have to enable crash flip from your radio.  (It may mean that there must be another way to enable/disable crash flip, I mean as a feature in case you don't want it.)  For this I (and they) will enable the accelerometer, don't see why that is an issue.

etracer65, 2019-06-16:
This will never happen. The choice to use crash-flip will always need to be mad by the user. We won't select a arming mode automatically.

pkruger, 2019-06-17:
@etracer65 I understand about arming automatically - that's not a good idea - ever.  This only enables crash flip, you still have to arm it.  So if you crash and disarm and the quad is upside down, crash flip is enabled until the quad is upright again - then you disarm, crash flip is deactivated, you arm again and take off.

mikeller, 2019-06-17:
Potentially this could be combined with `small_angle`: If this mode was enabled, whenever the orientation of the craft is detected as being outside of `small_angle`, then, instead of refusing to arm, simply arm in crash flip mode. This way there is no change to the fact that the arming behaviour changes dependent on the craft's orientation.

etracer65, 2019-06-17:
> Potentially this could be combined with small_angle: If this mode was enabled, whenever the orientation of the craft is detected as being outside of small_angle, then, instead of refusing to arm, [...]

I think this is a bad idea. Arming in crash-flip is still armed. Motors can spin. Arming disabled because of small angle is not armed so it is a change in the arming behavior.

The behavior in which the motors armed needs to be completely decided by the user. The attitude estimate (particularly after a crash) takes a bit to reorient. If the user tries to rearm quickly the estimate may not consider the craft "upside down" and thus arm in normal mode and spin all the motors unexpectedly. Any time I arm I want to know exactly what's going to happen. Relying on the attitude adds an element of unknown behavior.

mikeller, 2019-06-17:
Not advocating anything here, but:

> If the user tries to rearm quickly the estimate may not consider the craft "upside down" and thus arm in normal mode and spin all the motors unexpectedly. 

This is the case already now in default configuration. Only that now the expectation (the craft will not spin anything at all) is even further from the reality (all motors spin) than the expectation after the proposed change is (motors may spin on RPY deflection).

> Any time I arm I want to know exactly what's going to happen.

The current default for `small_angle` already causes this to not hold true.

So if this is decided to be a safety issue, then I think we **must** change what we do with `small_angle`.

etracer65, 2019-06-17:
I disagree. In the current state the choice I make (normal or crash-flip) is what will happen when I arm. There's no ambiguity. Small-angle may ***prevent** arming in normal mode, but it won't change the expected arming type. Crash-flip already overrides the small-angle arming disable so If I choose to arm in crash-flip I know what's going to happen. If I choose to arm with crash-flip disabled, then the motors are going to spin unless some arming disabled reason (like small-angle) prevents it.

pkruger, 2019-06-21:
@etracer65 If crash flip is on it is displayed in the OSD.  Also when arming/armed don't we all assume that the motors are spinning, i.e. don't touch?  I don't quite understand what the issue is, you still have to arm manually, if it is upside down it just means that crash flip is enabled when you arm.

etracer65, 2019-06-21:
@pkruger Not everybody has or uses an OSD. And if they do there's no guarantee that they have the warnings element enabled. Even if they do and can see the warning element it's too late because it displays after arming. Further the indicator doesn't help if the firmware "decides" to arm in a mode contrary to what the pilot wants as they can't do anything to correct the problem and make it arm as desired.

The issue is simply that the mode in which the quad arms must be under the pilot's control.

Additionally an example on why the pilot wouldn't want automatic selection of crash-flip based on orientation. Assume the quad got stuck in a tree but was right-side up. The pilot would want to use crash-flip as a common way to "wiggle" the quad safely out of the tree. But he couldn't because the quad would arm in normal mode spinning the motors and likely burning them up.

pkruger, 2019-06-23:
@etracer65 I understand.  How about keeping crash flip as is now, with an additional option to turn on crash flip automatically if upside down?  That way everyone is happy, i.e. you can still use crash flip on a switch as before and if you're a racer you can enable this auto crash flip feature in the CLI.

nikolaswidell, 2019-06-23:
Good idea, maybe a separate setting to enable this feature.

One downside is that sometimes when upside down, you are not able to flip the quad with turtle mode. Sometimes I personally forgot to turn off the turtle mode when going to retrieve the quad, this basically means that if I forgot to turn off the turtle mode and go pick it up, just a little tap on the gimbal would spin the props.

Because of this I have set the beeper to the same AUX than crash flip, so when flipping the crash flip switch the quad also starts to beep (and ofc stops when actually arming the quad to flip over).

etracer65, 2019-06-23:
> with an additional option to turn on crash flip **automatically** if upside down?

For the 3rd time:

**The mode in which the quad arms must be under the pilot's control.**

The pilot must make the decision if it's appropriate to arm in crash-flip or normal mode. Period.

> That way everyone is happy, i.e. you can still use crash flip on a switch as before and if you're a racer you can enable this auto crash flip feature in the CLI.

How do you address the case where "you're a racer and enabled the auto crash flip feature" and the incorrect mode gets selected? Maybe it's not quite "upside down" enough to trigger the settings. Maybe the attitude estimate is incorrect due to the crash that caused you to be upside down? So the quad arms in normal mode unexpectedly and damages the motors by digging it - or worst case completely flips out (because the PID controller is active) causing more damage or possible injury.

Or the opposite - the quad thinks it's upside-down but really not because of that same attitude estimate issue after a crash. Then the quad always arms in crash-flip and the pilot can't take off even though they're already right-side up.

And overall, what's the point? Is enabling a single switch before you arm to enable crash-flip really that "cumbersome"? Why? Don't you know whether you want to use crash-flip mode?

So to answer the next question about "why the attitude estimate is wrong"... It's because the estimate of the quad's orientation is based on a fusion of gyro and accelerometer data. Under high motion the gyro data is used exclusively since the ACC data can't be trusted (can tell if acceleration is from motion or gravity). In a crash the forces (both gyro rotation and ACC measurement) can easily exceed the sensor limits (2000dps and 16G) and the values are limited. So the mathematical model of the attitude becomes corrupted - often quite significantly. Once the quad has come to a complete stop on the ground it can start "trusting" the accelerometer and the attitude estimate can begin to converge - but this takes some time. So it the pilot tries to rearm quickly after a crash the attitude estimate is likely to be wrong. Just ask tiny whoop pilots who mostly fly in angle mode - this is a big problem for them.

pkruger, 2019-06-24:
@etracer65 Chill out mate, I just want to understand why you think this is a bad idea. :o)  I think there are some corner cases you mentioned above I haven't thought of, not a perfect world.  Thanks for your input.  Good news is if this ever comes up again, you can point them to this issue and be done with it - so not all in vain... :o)

closed 2019-06-24 by pkruger