betaflight/betaflight #8436: bf 4.1 vtx tramp doesnt change power level, always sets 25mw

m013, 2019-06-16:
**Describe the bug**
when i change power level - nothing happens. it worked on earlier builds (even when i had to set vtx table manually, but it was without single digit power levels) and release version of 4.0.x

vtx - pandrc 5804v2
levels 25 100 200 400 600

on my micro with a mini version of this vtx, which has 25/100/200 power levels switching works correctly 

channel /band selection and pit mode have no issues

**To Reproduce**
try to set it via osd or lua. cover your antennas and see the video doesnt get cleaner.
no effect with low power on disarm on and off

[dump_cli_Leopard_20190616_185455.txt](https://github.com/betaflight/betaflight/files/3294275/dump_cli_Leopard_20190616_185455.txt)

functionpointer, 2019-06-18:
Your problem is probably in your vtxtable:
`vtxtable powervalues  0 1 2`
This is for some smartaudio devices and won't work for tramp.

Try the example shown in https://github.com/betaflight/betaflight/pull/7251

m013, 2019-06-18:
@functionpointer , initially i tried this table right after the previous one stopped to work (it didnt even show a power level, afaik)

vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues  0 1 2 3 4
vtxtable powerlabels 25 100 200 400 600

still, it works for my micro quad as i mentioned above.

now i added this one from your link and in works again, thanks

vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

functionpointer, 2019-06-18:
Are you sure your micro quad has a tramp compatible vtx and not a smartaudio one?
Because the powervalues sure are wrong.
But hey if it works...
Maybe the pandarc guys also accept these powervalues in addition to the correct ones. That would violate the tramp protocol but maybe they dont care.

m013, 2019-06-18:
@functionpointer , i am sure. because in bf vtx protocol is set to tramp.
i clearly see difference in fpv feed when i switch from 25 to 200mw (in case of low power on disarm)

closed 2019-06-18 by etracer65