betaflight/betaflight #8442: Easy and practical debugging procedure for initialisation code.

wingunder, 2019-06-17:
Using backbox for debugging post-initialisation code (everything in the run() function) works pretty well. However, I would like to know how developers debug the initialisation code easily, as I do not exactly understand how backbox could be used effectively for doing this.

I guess the straight forward answer to the above question would be to just use an SWD debugger, but this is not always practical/possible. One could also just listen to the buzzer or look at the OSD, but these are very crude ways of debugging.

I think it would be quite nice to log the startup sequences. Correct me if I'm wrong, but the initialisation is not time critical, so the real-time efficient blackbox-way of logging would be an overkill. A simple plain text log-file with timestamped, human readable log lines, in plain text, will do the trick. In order to avoid the misuse of this logging mechanism, that most probably would cause real-time problems, one could make sure that this initialisation file handle is only available during the initialisation phase, in other words, _before_ run() is called.

Hopefully something similar to the above described, already exists. If so, please kindly point me to it.

Thanks in advance.

Docteh, 2019-06-18:
For debugging methods I think it always depends on what you want to debug. It sounds like you have in mind a situation where you need to gather initialization, but initialization on a bench is no good?

etracer65, 2019-06-18:
When not using hardware debugging I will sometime just blink the status LED as a simple debugging indicator. This is fine if I'm just looking for an indication that the code is reaching a given point.

wingunder, 2019-06-18:
@Docteh You're right. It's trivial to plug a USB cable in, set it into [msc](https://github.com/betaflight/betaflight/wiki/Mass-Storage-Device-(MSC)-Support) mode and read a file from the mass storage device, as opposed to hooking up the debugger, etc, etc, until you find the point in the code where things start going wrong. I don't say a debugger is not needed, but it might well be that you could find problems faster, even on non-technical people's hardware, if a startup log is available.

@etracer65 Thanks for your tip. One could also use an unused serial port and spit out some debug data, if you are in the initialisation phase. This, as well as the LED toggling, however requires code changes and firmware uploads.

Switching to [msc](https://github.com/betaflight/betaflight/wiki/Mass-Storage-Device-(MSC)-Support) mode and reading a file is an easy task, even for non-technical people. As the startup log will always be the same for a certain platform and firmware version, a simple diff (the GNU one with 2 files, not the CLI one) could possibly point out some issues very quickly.

The startup code could be compiled into the firmware or not, depending on the hardware resources. Enabling/disabling the startup debug log could be done from the CLI.

One major downside might be that the startup log statements might clutter the code.

etracer65, 2019-06-18:
Where exactly would we store this "startup log"? Microcontrollers don't have a lot of memory and allocating a big chunk for an arbitrarily large "log" that will be rarely used doesn't seem like a good idea. Entering a fully functional mass storage state is quite far along in the initialization sequence so useless to debug many startup problems.

ledvinap, 2019-06-18:
@etracer65 : One possibility is put debug into on MSP port, wrapped as msp commands (but not actually using MSP plumbing to be able to log ASAP). And startup phase before opening serial port can be absorbed by relatively small buffer (used as debug queue after serial port is available).

Docteh, 2019-06-18:
@wingunder making things human readable means you're allocating space for those human readable strings

I think the unified build for STM32F722 is in the worst position as far as code space goes. AXIM_FLASH1 looks like where the code is put. 
```
Building STM32F7X2
Memory region         Used Size  Region Size  %age Used
        ITCM_RAM:       15856 B        16 KB     96.78%
      ITCM_FLASH:          0 GB        16 KB      0.00%
ITCM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     ITCM_FLASH1:          0 GB       480 KB      0.00%
      AXIM_FLASH:        8886 B        16 KB     54.24%
AXIM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     AXIM_FLASH1:      444256 B       480 KB     90.38%
        DTCM_RAM:       27864 B        64 KB     42.52%
           SRAM1:       62508 B       176 KB     34.68%
           SRAM2:          0 GB        16 KB      0.00%
       MEMORY_B1:          0 GB         0 GB      -nan%
   text	   data	    bss	    dec	    hex	filename
 446838	   6304	  84064	 537206	  83276	./obj/main/betaflight_STM32F7X2.elf
Building STM32F7X2 succeeded.
```

I guess with this issue post, I am wondering about what you are looking to debug.

wingunder, 2019-06-18:
@etracer65 Thanks for your feedback. 
> Where exactly would we store this "startup log"? 

On the sdcard, if available.

> Microcontrollers don't have a lot of memory and allocating a big chunk for an arbitrarily large "log" that will be rarely used doesn't seem like a good idea.

This was not really what I had in mind.  

> Entering a fully functional mass storage state is quite far along in the initialization sequence so useless to debug many startup problems.

Thanks, I saw that in the init() function, but I managed to move the blackbox initialization, which includes the sdcard initialization, up to just after the SPI initialization. This is quite a way up in the initialization and works on my NERO. So I think one could quite easily get the sdcard up and running early enough, at least before the bulk of the peripherals get initialized.

wingunder, 2019-06-18:
@Docteh Thanks, you raised a good point about the memory drawback that human readability will cause. To counter this, one could use an enum and map messages to a uint16_t. This would however require post processing.

In this case, I'm back to square one: Just use backbox and map it back into human readable format on a pc. I'll try this, as I'm now pretty convinced that the backbox initialization can be moved up far enough in init(), in order to be usable for logging most peripheral initialization.

fujin, 2019-06-19:
blackmagicprobe & gdb?

wingunder, 2019-06-19:
> blackmagicprobe & gdb?

@fujin Point taken. Thanks.

closed 2019-06-21 by wingunder