betaflight/betaflight #8444: DYS ESC Telemetry

venii-ugol, 2019-06-19:
ESC DYS ARIA 40A (not 4 in 1 ESC). Connect TE to OMNIBUS AIO F4V6, tune, see in OSD telemetry for each ESC (4pcs), but only temperature and RPM.. See in configurator : Voltage =0, Current = 0, but source - ESC and all powered on.. Wrote to DYS, they answered : "BF software only show out one voltge and current, can't show 4 esc, so, you only need connect one CU from 4 escs, but, KISS FC can show four voltage and current " - **is true??** Why if connect 4 in 1 ESC for exemple ESC holybro tekko32 all work, we see current from all ESC and average, but whet we connect 4 ESC to 1 RX on FC isn`t work..

etracer65, 2019-06-19:
Sounds like the manufacturer is saying they didn't implement current and voltage telemetry. Since they provide a separate current connection on the ESC (which is extremely unusual) this is probably the case.

mikeller, 2019-06-20:
They probably saved a few cents by only adding one shunt resistor instead of four individual ones, in order to be able to sell a few cents cheaper...

venii-ugol, 2019-06-20:
As I understand it, they say that BF cannot accept four values of V or I from four ESC. Probably, I don’t know if ESC is 4 in 1, then the connection goes with one ESC controller which gives 4 values to the port, but if there are separate ESC, then there should be 4 connections for each of which gives its V and I values. That's it, how they write, BF cannot, but KISS can ... probably

mikeller, 2019-06-20:
> how they write, BF cannot, but KISS can

I would put that down to technical incompetence of the person writing the documentation.

venii-ugol, 2019-06-20:
[image]
[image]

venii-ugol, 2019-06-20:
[image]

mikeller, 2019-06-20:
@venii-ugol: This is not the right place to complain about the technical incompetence of dys staff - please open this complaint with dys.

closed 2019-06-20 by mikeller

venii-ugol, 2019-06-20:
ok, sorry . I just wanted to clarify the situation, maybe I did not translate correctly, since English is not my native language, and I wanted to make sure that from the BF side everything is fine and we need to continue to correspond with DYS..

mikeller, 2019-06-20:
@venii-ugol: As every other BLHeli_S / BLHeli32 4-in-1 ESC that is on the market proves, voltage and current monitoring over ESC telemetry works fine in Betaflight, and this is just a case of a DYS product not implementing these features.

venii-ugol, 2019-06-20:
Thanks, question is closed