betaflight/betaflight #8447: spektrum 2048 omnibusf4 not recognized

NoCake, 2019-06-19:
1. DSMX not communicating with omnibusf4 on emax minimag v2 for fw 4.0.2, 4.0.3. Unsure about previous 4.0.X versions.  Works on fw 3.5.7.

2. reproduce:  With fresh install of 4.0.2 (I tried this on 4.0.3 as well but the diff is 4.0.2) choose serial dsmx 2048, uart1 (for emax mini magnum v2).  Cannot bind using CLI steps.  However if I roll back to 3.5.7 and bind, then re-install 4.0.X the rx is not recognized, but it works on 3.5.7.

3. emax mini magnum v2 omnibusf4.  DSMX diversity satellite using the DSM pins on FC.

4. fw 4.0.2, 4.0.3.  Configurator 10.5.1.  fw 3.5.7 works.

5. DIFF below.  Note I have set spektrum_sat_bind = 9 because I was trying to bind.  As mentioned, if I bind using 3.5.7, verify rx working, then install 4.0.X, then no rx response with spektrum_sat_bind = 0.

```
# diff

# version
# Betaflight / OMNIBUSF4 (OBF4) 4.0.2 May  7 2019 / 19:40:31 (56bdc8d26) MSP API: 1.41

# start the command batch
batch start

board_name OMNIBUSF4
manufacturer_id AIRB

# resources
resource MOTOR 1 B01
resource MOTOR 2 A02
resource MOTOR 3 B00
resource MOTOR 4 A03

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -AIRMODE
feature -RX_SPI
feature RX_SERIAL
feature TELEMETRY
feature LED_STRIP

# beeper
beeper -ON_USB
beeper -CRASH_FLIP

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 64 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200

# led
led 0 7,7:SU:CTOW:10
led 1 8,7:SU:CTOW:10

# color

# mode_color

# aux
aux 0 0 2 1525 2050 0 0
aux 1 2 1 1300 1700 0 0
aux 2 13 0 1550 2075 0 0
aux 3 35 1 1725 2100 0 0
aux 4 49 1 900 1300 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 4 s 1900
rxfail 5 s 1500
rxfail 7 s 1100

# display_name

# master
set gyro_lowpass_hz = 0
set gyro_lowpass2_type = BIQUAD
set gyro_lowpass2_hz = 0
set dyn_lpf_gyro_min_hz = 0
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set serialrx_provider = SPEK2048
set spektrum_sat_bind = 9
set spektrum_sat_bind_autoreset = OFF
set dshot_idle_value = 360
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 10
set align_board_yaw = 90
set current_meter = VIRTUAL
set ibatv_scale = 155
set yaw_motors_reversed = ON
set small_angle = 180
set auto_disarm_delay = 60
set yaw_deadband = 2
set pid_process_denom = 1
set ledstrip_visual_beeper = ON
set osd_units = IMPERIAL
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 60
set osd_vbat_pos = 385
set osd_rssi_pos = 2061
set osd_tim_1_pos = 2455
set osd_tim_2_pos = 2423
set osd_flymode_pos = 1
set osd_throttle_pos = 129
set osd_vtx_channel_pos = 353
set osd_ah_pos = 200
set osd_current_pos = 2070
set osd_mah_drawn_pos = 2049
set osd_craft_name_pos = 2442
set osd_gps_speed_pos = 66
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 386
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 1
set osd_pid_roll_pos = 322
set osd_pid_pitch_pos = 354
set osd_pid_yaw_pos = 386
set osd_debug_pos = 1
set osd_power_pos = 47
set osd_pidrate_profile_pos = 418
set osd_warnings_pos = 2185
set osd_avg_cell_voltage_pos = 2433
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 360
set osd_disarmed_pos = 2314
set osd_nheading_pos = 311
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set vtx_band = 5
set vtx_power = 2
set vtx_freq = 5658
set vcd_video_system = NTSC

# profile
profile 0

set dyn_lpf_dterm_min_hz = 60
set dyn_lpf_dterm_max_hz = 175
set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 6000
set crash_delay = 150
set crash_recovery = ON
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 10
set throttle_boost = 2
set p_pitch = 32
set i_pitch = 90
set d_pitch = 70
set f_pitch = 92
set p_roll = 28
set i_roll = 84
set d_roll = 65
set f_roll = 90
set p_yaw = 30
set i_yaw = 90
set f_yaw = 80
set d_min_roll = 30
set d_min_pitch = 35
set d_min_boost_gain = 55
set d_min_advance = 0

# rateprofile
rateprofile 0

set thr_mid = 0
set thr_expo = 25
set roll_rc_rate = 110
set pitch_rc_rate = 110
set yaw_rc_rate = 130
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 40
set tpa_rate = 75
set tpa_breakpoint = 1400

# end the command batch
batch end
```

etracer65, 2019-06-19:
You have a clone that has incorrectly implemented hardware that's affecting UART1. Try using a different UART. Alternately you can try this workaround for UART1.
```
resource pinio 1 b14
set pinio_config = 1,1,1,1
save
```

closed 2019-06-19 by etracer65

mikeller, 2019-06-20:
Duplicate of #7849.