betaflight/betaflight #8453: FlySky Flit10 Rx - Betaflight Receiver Tab Not Working

simtcrom, 2019-06-21:
FC - [Airbot Omnibus F4 Nano V6](https://www.banggood.in/Original-Airbot-Omnibus-F4-Nano-V6-Flight-Controller-OSD-5V-2A-BEC-LC-Filter-2-6S-20x20mm-p-1320256.html?rmmds=myorder&cur_warehouse=CN)
Rx - FlySky [Flit10](https://www.banggood.in/Original-Airbot-Omnibus-F4-Nano-V6-Flight-Controller-OSD-5V-2A-BEC-LC-Filter-2-6S-20x20mm-p-1320256.html?rmmds=myorder&cur_warehouse=CN)
Tx - FlySky FS-i6 Modded with FlyPlus Firmware

Connection,
[image]

Diff,
[diff.txt](https://github.com/betaflight/betaflight/files/3315049/diff.txt)

I bound the Rx to Tx fine. I can see solid green light on Rx. Also Tx is showing SNR, RSSI etc fine. If I turn off Tx, the solid green light on Rx becomes blinking state (another confirmation for binding is fine).
Configured Rx as Serial and provider as IBUS in BF. Also configured UART6 as Serial Rx.
**Issue is Receiver tab in BF does not respond.**
Also tried wiring IBUS wire to UART1 and no success.

The very same setup using my old Flysky FS-RX2A Pro V1 works fine.

I tried the Rx on another F4 FC (Omnibus F4 pro v2) Running INAV and all works fine.
Flashed INAV to Airbot Omnibus F4 Nano V6 and I can get Receiver tab working.
Does this mean its not a hardware issue but betaflight issue?

Can some one please assist?

etracer65, 2019-06-21:
Try using RX3 (UART 3).

simtcrom, 2019-06-21:
> Try using RX3 (UART 3).

Thank you. Worked.

simtcrom, 2019-06-21:
Used UART3 and all set.

closed 2019-06-21 by simtcrom

jpedryc, 2023-02-14:
For people who might land here - my setup:
* Omnibus F4 SD
* Flit10 RX
* FlySky FS-i6
* BetaFlight Configurator 10.9.0

Flashed 4.4.0 and tried resolving similar issue. Everything was connected fine (UART1 - IBUS) - receiver tab not working :sweat: 
Flashed 4.3.2 and with the same set up - everything started working as expected.

Reminder: Don't trust the freshest updates, try older versions :stuck_out_tongue_winking_eye: