betaflight/betaflight #8458: Support CRSF binding from CLI

dzenbot, 2019-06-22:
**Problem**
When typically updating your TBS Crossfire TX module's firmware, this causes all previously bound Crossfire RX to be invalidated, forcing you to manually bind each one of them using a cumbersome and hard-to-reach physical button on the rx (specially on the Nano RX). This forces you, most of the time, to open up every single quad you have with Crossfire and consumes a lot of time. The user experience is really bad in that sense.

**Suggested Solution**
Imagine a CLI command, similar to `bl` for bootload. Let's call it `bling` for fun. When entering on `bling` mode, your RX is programmatically set to bind mode. I am unfamiliar with the CSFR protocol APIs, so take my suggestion more as a consumer than a developer's idea.

KnuckleUpFPV, 2019-06-22:
New crossfire firmware binds without the button and updates the rx. There is no need for this. I swap between two modules. I never touch the receiver. Just power up tx and put it in bind. Done!

kamik2, 2019-06-24:
this has been removed as it was a bug.
from now on if you swap the tx modul you need to push the button.
this is not needed if the rx was bound before (thats how it should be)

KnuckleUpFPV, 2019-06-24:
Im on 2.95 and can swap modules without the button. If it was a bug that was fixed its still working.

dzenbot, 2019-06-24:
@KnuckleUpFPV are you really on `2.95`? That's quite a few releases away, is it not?

KnuckleUpFPV, 2019-06-24:
Yeah im on 2.95. Been on it for a few weeks. Ran 2.93 before that for a bit until the dropped telemetry bug from switching from 150hz to 50hz showed its face. Then went back to 2.42. Until 2.95 dropped and was supposed to fix the telemetry bug. Which it didn't. If you put it into force telemetry mode its fine. In normal mode it still drops telemetry after switching to 50hz mode from time to time. Are you still using the old configurator or agent x? If you are not on agent x you wont be able to get 2.95. Also get the early access code from the tbs lounge and put it in the settings after your device code. Seperate it with a comma.

kamik2, 2019-06-25:
> Im on 2.95 and can swap modules without the button. If it was a bug that was fixed its still working.

you need to reload the fw (got to a other fw than 2.95, update the rx and roll back to 2.95)
the files for 2.95 have been replaced a while ago.

btw - the lounge code is not used atm. there is no extra fw avalible with it.

mikeller, 2019-06-25:
To add one more bit of information here: While the CRSF specification hints at CRSF commands for binding, the specification that was used to add CRSF support to Betaflight does not cover any details for such commands, and I am not sure if they were ever actually implemented.

KnuckleUpFPV, 2019-06-25:
My 2.95 is a few days old. There is beta firmware for other devices. Like the smart audio fix. I went all the way back to 2.16 to test an issue for a friend. Couldnt take my lite back that far. But my full got rolled back then back to 2.95. Updated both modules at the same time to be sure they were on the exact same iteration of the firmware.

closed 2019-08-01 by stale[bot]

PaulFPV, 2022-04-12:
I think binding from CLI is a good idea (for example the button on your Nano diversity doesn't work, like mine in this moment).

Adding this feature is something already done in the TBS side or is a work for both TBS and Betaflight developers?

haslinghuis, 2022-04-12:
Don't know if there is any status on this but do agree.

Having one nano missing the boot button. Have to replace it as I need to do a recovery. Another nano is buried in the quad and the sixty9 has the boot button at the underside of the board, not using the casing on a 20x20 it's hard to press the button - using tweezers it just like eating Japanese food :grin:

tomav, 2023-04-30:
Missing button too on my Crossfire Nano RX. If anyone knows how to put it in bind mode... Thanks!

haslinghuis, 2023-04-30:
Finally replaced the nano receiver.

Did you try to bind using the button on TX instead:
[image]
https://www.youtube.com/watch?v=8SYiMctDex8

tomav, 2023-05-10:
I did. This is true for a new receiver I suppose. Mine is bound to a TX and I want to bind it to another TX.

PR #13119 (2023-12-04): Allow CRSF RX binding via cli
This Pull Request enables CRSF RX binding via CLI. 

The change has been tested with `OMNIBUSF4SD` and TBS Tracer Nano RX.

**Rationale**
There are conditions under which access to the binding function is necessary but limited or not possible using a button.
Examples:
- Access to the receiver is limited
- The pairing button is physically damaged

**Steps to test a feature:**
1) Configure CRSF receiver
2) Go to `CLI`
3) Execute `bind_rx`

**Result:** 
- CLI outputs `Binding...`
- RX starts blinking a green LED indicating binding mode

Resolves #8458