betaflight/betaflight #8459: ESC Throttle limiter

Healwave, 2019-06-22:
**Is your feature request related to a problem? Please describe.**
Not a software problem, but a power inconsistency problem

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
I came across this thread https://github.com/ButterFlight/butterflight-configurator/issues/37 because I was looking exactly the same. There is too much inconsistency on throttle across the lipo cycle.. I notice a lot harder to go through my small gates when the lipo is full rather than when the lipo cycle it at 70%.

I'm not aware of the implementation of betaflight.. But maybe it can simpler than what this thread is requesting: Instead of having a throttle level calculation based on vbat (VBAT Throttle compensation), we could just assign a cap to the max throttle value that FC sends to the ESCs. I'm not sure if this is possible nor I have enough knowledge about BF, or if the firmware even have access to this kind of information. But anyway, let me try to explain:

Input values from TX are between 1000-2000. The FC will send a throttle value to the ESCs between 0-1000 (I made up this part, don't know if this works this way)

So, at the current state of the BF/Butterflight, a 2000 input value would be translated to an ESC throttle value of 1000.

But, I would suggest to add an option that let user set which value/relative value that he wants when throttle input is 2000. 
So, let's say that the user has set the max value of ESC throttle of 700 (70%). What is supposed to happen, is that the input value sent by TX of 2000, would be translated to ESC throttle value of 700. And the magic happens here: If the battery is low, but it still can reach that limited throttle level of 70%, it will. But if the battery is low and cannot reach that level, it doesn't matter at all, it should use the max it can (still limited but physically incapable) and if the user wants even more consistency at lower levels of the battery, he/she should lower the **_ESC Throttle Limiter_** value in order to accomplish that.

[ESC Throttle Limit.xlsx](https://github.com/betaflight/betaflight/files/3316704/ESC.Throttle.Limit.xlsx)

**Additional context**
https://github.com/betaflight/betaflight/issues/5567, https://github.com/betaflight/betaflight/pull/5528

Healwave, 2019-06-22:
Issue will be closed to specify this better. I will open when I I've wrote enough detail and some samples

Healwave, 2019-06-22:
I think it's enough detailed now. Reopening the request.

etracer65, 2019-06-22:
This issue has been discussed and abandoned as unworkable. Battery voltage varies to much during flight to be useful in this regard. Also it creates a feedback loop spins out of control since by increasing throttle in response to low battery you put more load on the battery. Which then causes the battery to sag even more, causing more throttle. And so on.

Healwave, 2019-06-22:
Closing this request because throttle limit does the same thing.

closed 2019-06-22 by Healwave