betaflight/betaflight #8460: MSC on mac

Beerotor, 2019-06-22:
I am trying to use the MSC on the beerotor f4 chip, but when I use the MSC in the CLI or even select it in betaflight, I never get to see the storage device in my mac devices, is there something I am doing wrong or is it not possible on mac? thank you

mikeller, 2019-06-22:
Please supply the following information:

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

etracer65, 2019-06-22:
I use it every day - works perfectly. It can take about 15 seconds for the SD card to show up so make sure you're not impatient. Also try reformatting your SD card and make sure it works when directly connected to your Mac (using a card reader).

closed 2019-06-23 by no-response[bot]