betaflight/betaflight #8464: Crashflip Inconsistencies

louis-test, 2019-06-23:
**Describe the bug**
Crashflip seems "weak" with newer firmware versions of BF prompting a code review.

**To Reproduce**
Place quad upside down in grass, try to turtle.

**Expected behavior**
Quad turtles with a high power burst from the motors

**Flight controller configuration**
BF Default with accelerometer off

**Setup / Versions**
Setup: FC are always the Pyrodone F4. ESC is various (Pyrodrone, Hobbywing, Airbot, etc.) Motors are 2205.5 Hyperlite 1722kv or 2205 RDQ 1800kv running on 6S.
Issue observed from multiple pilots with "fleets" of the above setups.

Versions:
FC FW are 3.5.3 or 4.0 (nightly w/ RPM filtering enabled)
ESC are BLHeli_S or BLHeli_32 (latest fw)

Some people feel that the issue is worst with newer BF firmware and with BLHeli_S ESC.

**Additional context**
Code review of mixer.c prompted due to report of BF firmware inconsistencies in turtle efficiency. Line 734 reads to me that the motor output cannot reach 100% due to subtracting CRASH_FLIP_DEADBAND. However, AFAIK this code has been present since introduction of turtle. 

Can someone take a look at applyFlipOverAfterCrashModeToMotors() and ensure we are giving turtle our best effort?

            motorOutput = (motorOutput < motorOutputMin + CRASH_FLIP_DEADBAND) ? disarmMotorOutput : (motorOutput - CRASH_FLIP_DEADBAND);

etracer65, 2019-06-24:
Try without RPM filtering enabled on 4.0 and use a non-beta BLHeli32 firmware. Changes were made to the BLHeli32 beta firmware that slightly delays motor starting. This has the effect of reducing the effect of crash-flip when you just bump the sticks.

The crash-flip mixer code is unchanged from 3.5 to 4.0.

louis-test, 2019-06-24:
I tried that just an hour ago. I'm trying to get some clarity on the user reported differences and eliminate any confirmation bias.

2 Quads - both using 1800kv 2205 RDQ motors - both using 6s on same frame
Quad #1 = BLHeli_S 16.7 and Pyro FC
Quad #2 = BLHeli_32 32.6 and Pyro FC
Both on 3.5.3 of Betaflight - Both on stable releases of BLHeli - Both have identical settings PIDs, Rates, Features, etc.

Without a doubt, the BLHeli_32 quad struggles more to ramp up the motors from a full stall state. I found the PR that references runaway takeoff prevention and tried disabling that (no effect). I also attempted to set crashflip_motor_percent to get all motors spinning some minimum amount. This does allow the motors to spin some (more than before), but I was unsuccessful in getting them to accelerate and flip in a very short grass.

Based on the above I think the issue boils down to two things, Motor Size and BLHeli_32. Friends with 2207 do not seem to have these issues.

However, while I acknowledge that the code hasn't changed since 3.5, doesn't that conditional rob "CRASH_FLIP_DEADBAND" from the max motor output? I would rewrite that line as:
        motorOutput = (motorOutput < motorOutputMin + CRASH_FLIP_DEADBAND) ? disarmMotorOutput : motorOutput;

Please correct me if i'm misunderstanding that.

etracer65, 2019-06-24:
The code is correct. With your example the throttle would "jump" to a higher value when leaving the deadband rather than start smoothly.

And why are you concerned about this? Surely you don't need full throttle to flip your quad.

And if you find differences in BLHeli_S and BLHeli32 I would take that up with the appropriate developer. They handle motor startup and power ramp differently so it's not really surprising.

But that being said, nobody else has complained about this ever so I'm finding it difficult to perceive that there's a problem. The props you use have a bigger impact on reverse thrust than anything else.

louis-test, 2019-06-24:
Thanks for explaining that to me. I think we can close this as I'll take it
up with blheli.

I beleive the root cause of the user reports that the functionality changed
with betaflight was due to the way the additional stickband was added. Open
to your thoughts on this.

Agree that 100% throttle isn't needed but I assumed that the startup torque
might be proportional to the commanded input. The issue is that the motors
can't *start* spinning. Once they get going there is no issue flipping.

I know that in my team specifically we have alot of pilots complaining.
However it is with noting were mostly all using the same frame which is a
top mount that has all the motors in the grass when upside down.

If you're interested in statistics please let me know and I'll get a poll
of who is having issues and what configuration they're using.

etracer65, 2019-06-24:
The changes you mention involving `CRASH_FLIP_STICK_MINF` were added in BF3.4 (2 major releases and nearly a year ago) and have remained unchanged since.

louis-test, 2019-06-24:
Thanks etracer. Alot of us skipped 3.4 and only picked up these changes in 3.5. I hope I am not giving the wrong impression that I "blame" you or your team for any of the work you did. I am trying to take a scientific approach and go find why things aren't working well any more.

The fact that I can now explain to pilots why they *think* turtle changed with BF Firmware is great. I can  communicate with confidence that it did not change how much power the motors received but only the stick range when the maneuver is initiated.

I don't know where to communicate with the BLHeli_32 devs. Do you have a recommendation? I reached out on the RCGroups thread but I don't know if that's where they're looking.

Again, many many thanks to you and the other BF devs.

etracer65, 2019-06-24:
Either rcgroups or open an issue in their GitHub. https://github.com/bitdump/BLHeli

mikeller, 2019-06-24:
@louis-test:

> I don't know where to communicate with the BLHeli_32 devs.

@sskaug (mostly in https://github.com/bitdump/BLHeli) is the person to talk to.

closed 2019-08-01 by stale[bot]